wireviz/WireViz #310: Internal Jumper

alorman, 2023-03-12:
Is there a way to create an internal jumper within a plug?
I've mocked up a very quick demonstration below.

[image]

I've seen this need in things like Harting connectors and other things that may use the plug presence as a state change. 

Thank you!

avenjamin, 2023-03-12:
See #286

JonMcLin, 2023-03-13:
[image]

`connectors:
  X1: &bar
    type: Molex Micro-Fit
    subtype: male
    pinlabels: [A, B, C, D]

  X2: &baz
    type: Molex Micro-Fit
    subtype: male
    pinlabels: [Clk, Gnd]

cables:
  W1:
    category: bundle # bundles are routed together, but more loosely than normal cables
    wirecount: 3
    colors: [BK] # if number of items in color list is less than wirecount, loop colors
    gauge: 0.25 mm2
    show_equiv: true
    length: 0.2

connections:
  -
    - X1: [1,2,3,4]
    - W1: [1,1,2,3]

  -
    - W1: [2,3]
    - X2: [1,2]

`

alorman, 2023-03-13:
Awesome! Thanks all, closing issue

closed 2023-03-13 by alorman

kvid, 2023-03-26:
@alorman , @avenjamin , @JonMcLin - thank you for contributing with a well described use case, link to an earlier relevant issue, and an alternative solution. This is a great example of the power of a cooperating user community.

In case you haven"t already been notified due to subscribing/watching the relevant issues, I just want to inform you about the recent development in #286 with [a new alternative](https://github.com/formatc1702/WireViz/issues/286#issuecomment-1483701234_) you can try out. Feel free to contribute in such earlier issues aswell.

kvid, 2024-05-04:
@alorman, @avenjamin, @JonMcLin, and others with similar use cases - I added today a suggested work-around for over-sized loops in https://github.com/wireviz/WireViz/issues/286#issuecomment-2094309143