wireviz/WireViz #319: Cable with numbered cores

zfarkas88, 2023-06-23:
Hi,

would it be possible to visualize somehow cables with numbered cores?

Thank you in advance!

[image]

Colour code: VDE 0293-334

kvid, 2023-07-04:
As long as all wires (cores?) in the cable are numbered 1, 2, 3, 4, etc. - then this is already supported, but custom numbering that differs from the sequence of natural numbers (similar to custom pin numbering of a connector) is not currently supported.

zfarkas88, 2023-07-04:
by visualisation I mean something like this
[image]

```
connectors:
  X1:
    type: Molex MX150L
    subtype: female
    pinlabels: [Pin 1, Pin 2, Pin 3, Pin 4, Pin 5, Pin 6]
    notes: Connector 1
  X2:
    type: Molex MX150L
    subtype: female
    pinlabels: [Pin 1, Pin 2, Pin 3, Pin 4, Pin 5, Pin 6]
    notes: Connector 2  

cables:
  K43:
    gauge: 0.75 mm2
    length: 8
    colors: [BK,BK,BK,BK,BK,YEGN]
    wirecount: 6
    shield: false
    wirelabels: ["1","2","3","4","5","6"]
  
connections:
    - #K43
        - X1: [1,2,3,4,5,6]
        - K43: [1,2,3,4,5,6]
        - X2: [1,2,3,4,5,6]
```

kvid, 2023-07-16:
As far as I know, Graphviz doesn't support printing text on top of spline edges, and anyway I think it would've been tricky to obtain readable text on top of any wire, as they normally are either very thin or multicolored. However, if anyone find a way to implement this, please share your ideas here or create a PR.

Currently, I recommend using the allocated wire numbers (if your core numbers are a sequence of natural numbers), or `wirelabels` otherwise. Both are visualized as text above each wire in the central part of the cable box.

martinrieder, 2024-05-21:
This would pose a solution to #8, though technically not feasible directly in Graphviz. It may be needed to add the numbering by post-processing the SVG output.