wireviz/WireViz #323: Overriding rankdir using tweaks does not work with dev build

alialsheikh, 2023-08-12:
For some reason, after installing the dev version, output rankdir has changed from LR to TB and is stuck that way. With v0.4 I cannot find a .dv file and so had to resort to using tweaks. The below works on v0.32 but does not work well on v0.4.

```
tweak:
  override:
    graph:
      rankdir: 'LR'
```

Any idea why? I need to use v0.4 due to the connector-to-connector support, but now my entire output is destroyed. Not presentable. Below are two screenshots for comparison.

With v0.32
[image]

With v0.4
[image]

kvid, 2023-08-14:
This seems to be a duplicate of #258 that was fixed by PR #261 and then merged into master in [v0.3.2](https://github.com/formatc1702/WireViz/releases/tag/v0.3.2), but not currently included in `dev`, `latest`, or any of the PR branches. If you need this fix together with such a branch, then I suggest you create your own local branch from the `dev` or PR branch and cherry-pick the fix https://github.com/formatc1702/WireViz/commit/9af0cb8ab0314bc6754b153769f7562b5b38ec10 on top of it.

The same issue has also been reported in #277, #280, and #299.

kvid, 2023-08-28:
I cherry-picked the fix for this issue into `dev` yesterday (see https://github.com/wireviz/WireViz/issues/316#issuecomment-1694685896 for details) and when the PR branches get rebased on top of `dev` again, then they also will contain it.

@alialsheikh - if you are able to use the updated `dev` code and verify that your reported problem is solved, then I suggest closing this issue. If you depend on some PR branch not yet updated, then describe that here and also in that PR to ask for a rebase.