wireviz/WireViz #326: [feature] Hide Cable Blocks

heinduplessis, 2023-10-04:
Sometimes I want to publish a simple wire diagram and the intermediary "wire block" adds clutter and confusion. Could someone add a "hidden_cable_block" property in the Yaml definition to indicate not to draw the intermediary cable block?

So for example [image] would render with just the X1 and X2 blocks and NOT the W1 block. The white green and brown wires would connect directly to the two connectors (X1 and X2). The shield wire would obviously then be omitted in this example, but would show up again if the W1 hidden_cable_block is set to false or omitted again.

kvid, 2023-10-04:
This sems to be a duplicate of #212 - where a work-around is described. (#278 was another duplicate.)

heinduplessis, 2023-10-05:
Thank you, I implemented the work-around.

closed 2023-10-05 by heinduplessis