wireviz/WireViz #330: Shield/Screen individual cores

scottbouch, 2024-03-20:
Hi, I need to draw a cable where only two cores of a 4 core cable are screened. Presently I have just added a note to the cable properties, but it would be nice to be able to show it with oval shaped rings around the screened cores as you would see in diagrams.

Is there a way to graphically show screens on individual cores?

My example: https://scottbouch.com/tmp/UHF2.html

(Many thanks for this wonderful solution to drawing cables, I love it!)

Cheers, Scott

kvid, 2024-05-04:
@scottbouch, thank you for sharing your use case. I don't know how to better accomplish your request with the current versions, but maybe a future version can support this. See also #56 for other discussions about complicated cable structures. Before starting a PR adding such a feature, it might be a good idea to discuss here:
1. How flexible should such a new feature be?
      - Should we support more than one group of shielded wires?
      - Should we support outer levels shielding containing other shield groups and wires?
      - Should we support other shield properties, like e.g. foil, braid, etc.?
      - Should we also support other features that are shared in subsets of wires, like e.g. twisted pairs (until #3 is resolved), wrapped binder groups, etc.?
2. How can we easily specify it in the YAML input syntax without making it too complicated for the simple cases?
3. How can we visualize it clearly in the diagram?
4. Do we need to add any cable shield details in the BOM description?
5. How much of these features might also be supported for connectors? (See #285)

As suggestions to points 2 and 3 above, see 2 examples below that still allow the original syntax for the simple cases:
```yaml
cables:
  w:
    colors: [RD, BU, YE, GN]
    shield: [YE, GN]
```
[image]
```yaml
cables:
  w:
    colors: [RD, BU, YE, GN]
    shield:
      ShieldP1: [RD, BU]
      ShieldP2: [YE, GN]
      ShieldAll: [ShieldP1, ShieldP2]
```
[image]

scottbouch, 2024-05-05:
Hi @kvid, thank you so much for responding to this request!

1. All of your points are valid, and likely to be experienced in real life. When I worked in industrial control systems, we would use long steel wire armoured cross-site cables with an overall foil screen, containing several individually screened twisted pairs or twisted triples (thesenses may be electrically isolated from each other and the overall screen). The triple conductor circuits were used where you needed to compensate for volt drop across a long cable length (ie: a sense wire). We didn't mark the armourinng on diagrams, it was an assumed connection of 'safety earth' between pieces of equipment and junction boxes. Some people may wish to use construction details here in the BOM but some may wish to reference a datasheet instead (see point 4). Ialso have some aircraft multicore cabling where theremay be say 12 conductors, each individually screened. So yes, some flexibility to this would be good to suit all eventualities.
2. As I am only a novice with yaml, I shall leave this question to the devs expeexperience!
3. Your suggested diagrams look good! It would be good if the loop touched the screen 'wire' to show an electrical connection.
4. The cable datasheet would include screen/shield details, the BOM would only normally include the cable type, reference number, etc.. From which you can find the datasheet. The data sheet would specify foil/braid, drain wire etc... But for a hobbyist with a cable with no part number or datasheet, the ability to manually add some basic construction details may be helpful (also including insulation material foreach layer (for example, PTFE requires different stripping tools to PVC, so these details help to plan the actual making process)
5. Yes, connectors shells often provide a means of terminating a screen, this could be shown as an 's' terminal as suggested in the other post to land the screen 'wire' on.

Many thanks again for the interest in this request, hopefully I can support further development from my own practical wiring experiences.

scottbouch, 2024-05-06:
In this example, red and blue are not screened, but green and yellow are. The screens are wrapped in a plastic films, so are keps isolated from each other across the length of the cable.

Some versions ofthis cable type allow the screens to touch, sooption for individdual screening of cores would be of benefit.

Thanks, Scott

[image]
[image]

kvid, 2024-05-06:
@scottbouch, thank you for all the real-world examples that show my suggested examples might be useful and not not too complex.

A consequence of implementing my suggestions above, is that the order of wires within the cable node might need to be reordered from the specified order (and numbering) to be able to draw a loop around consecutive wires.

About point 3, where you wrote "_It would be good if the loop touched the screen 'wire' to show an electrical connection._": I agree with you, but I've not yet found a way to obtain that. I used a table with rounded border inside a multi-row cell in the table of wires, and if I reduced the space around the rounded border to make it touch the shield wire below, it would also touch the wire above and the cells on each side. Maybe someone with Graphviz experience might suggest a trick for such a thing?

scottbouch, 2024-05-06:
Hi @kvid , thanks for the work on this.

The order that they appear in the cable node doesn't really matter in this instance as its the colours that matter, but actually this is a DEF-STAN cable, so the green and yellow would end up together anyway. However, I suppose there may be occasions though where a user builds up their own harness including screened and non-screened wires, they may wish to lay them out in a mixed combination to suit theirons needs.

Regarding the loop around the wires, what you displayed before was good, maybe go with that then in the future see if the gap could be closed up? I.e.: its still a significant improvement!

Thanks again, Scott

kvid, 2024-05-07:
@scottbouch wrote:
> [...]
> Regarding the loop around the wires, what you displayed before was good, maybe go with that then in the future see if the gap could be closed up? I.e.: its still a significant improvement!

I agree - we can probably live with the small gap until someone finds a way to improve it.

> 
> Worth noting the alternative approach from this above document of dashed parallel lines instead of a loop (dated 1964). But probably best see how it's done in ISO standards, which may well be a loop as discussed.

The dashed parallel lines could be an alternative visualization for individual screened wires, but I'm not sure how it can be used to show that multiple wires have a common shield.

I've not found "_how it's done in ISO standards_", but I found [examples](https://www.electrical-symbols.com/electric-electronic-symbols/lines-wires-conductors-symbols.htm) using dashed loops around a single wire or group of wires and [one with references to IEC 60617-2019 and IEEE Std 315-1993](https://www.asutpp.com/symbols-for-wires-and-connections.html). I also have seen a copy of the latter and can confirm it uses dashed loops around the wires. It seems IEC uses a dashed circle around a single wire, and a dashed circle outside a group of wires with arrows into each of the shielded wires. The latter alternative allows unshielded wires in-between shielded wires in the same group (see note A5 at the last linked page), but it might be a bit hard to read. IEEE allows separate dashed loops to be connected with a dashed line when not all wires in the common shield are adjacent in the diagram, but the IEC alternatives are also allowed.
[image]
Both arrows into each wire and connected loops (the left variations in the figure above) can be used to avoid reordering wires in the cable node, but they both might also be challenging to create with the limited Graphviz syntax.

I've tried using dashed loops in one of my examples, and I made one of the loops with a thicker line to see if that is better:
[image]
<details>
  <summary>Click to expand Graphviz <code>.gv</code> file source code</summary>
  
 ```dot
graph {
// Graph generated by WireViz 0.4-dev (and manually modified to add multiple wire shields)

// https://github.com/formatc1702/WireViz

	graph [bgcolor="#FFFFFF" fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
	node [fillcolor="#FFFFFF" fontname=arial height=0 margin=0 shape=none style=filled width=0]
	edge [fontname=arial style=bold]
	x1 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">x1</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">5-pin</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td port="p1r">1</td>
   </tr>
   <tr>
    <td port="p2r">2</td>
   </tr>
   <tr>
    <td port="p3r">3</td>
   </tr>
   <tr>
    <td port="p4r">4</td>
   </tr>
   <tr>
    <td port="p5r">5</td>
   </tr>
  </table>
 </td></tr>
</table>
> fillcolor="#FFFFFF" shape=box style=filled]
	x2 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">x2</td>
  </tr></table>
 </td></tr>
 <tr><td>
[... 142 lines cut ...]
    </td>
   </tr>
   <tr><td>&nbsp;</td></tr>
   <tr>
    <td>x1:5</td>
    <td>ShieldP2</td>
    <td></td>
   </tr>
   <tr><td colspan="5" cellpadding="0" height="2" bgcolor="#000000" border="0" port="ws2"></td></tr>
   <tr><td>&nbsp;</td></tr>
   <tr>
    <td rowspan="1">&nbsp;</td>
    <td></td>
    <td>ShieldAll</td>
    <td>x2:5</td>
   </tr>
   <tr><td colspan="5" cellpadding="0" height="2" bgcolor="#000000" border="0" port="ws"></td></tr>
   <tr><td>&nbsp;</td></tr>
  </table>
 </td></tr>
</table>
> fillcolor="#FFFFFF" shape=box style=filled]
}
```
</details>

scottbouch, 2024-05-08:
Hi @kvid, that is a very good website, thanks.

Overall, it would be good to implement whats presently easy or possble, with a 'TODO' list for future improvements.

Another take on it, in this example, the screens are shown as single circles connected by a line, even if they are drawn next to each other. These mic/tel wires are the exact wiring type I shred an image of, for aircraft communication circuits.

[image]
Source: https://scottbouch.com/mcfs/lightningt5t55/docs/ap101b-1005-10-sh55-fig1-al3-nov81-vhf-uhf-intercommunication-telebriefing.jpg

Maybe it would be good if possible to ideally give the user the option to connect circles with lines, or to draw ovals enveloping several wires, this gives freedom.

Also worth noting that the multiple arrows are used in this drawing, but not for screens, instead are used to provide wiring harness identification.

Cheers, Scott

scottbouch, 2024-05-08:
I have just been reading IEEE 315-1975, turns out the multiple headed arrows can be used for a variety of reasons, such as the application of screening or nomenclature as shown in these posts.

It also defines long dashes for screening/shielding:
[image]

Here the arrows area used to show which conductors are shielded, plus we are into the circles and ovals:
[image]

[image]

The next section of the standard  is onto coaxial cables, but that can wait for a future date!

Cheers, Scott.

chloepilonv, 2025-02-24:
Hello!
I was just wondering if this feature had been implemented or is planning to be (I don't see it so officially in the syntax doc, whatsoever). I tried the following code, but get no results : 
`cables:
  Main cable:
    type: signal cable 
    colors: [BN, WH, RD, BU, GN, PK] #[BN, WH, BN, WH, BN, WH]
    shield:
      shield1: [BN, WH]`
Thank you a lot !

chloepilonv, 2025-02-24:
> Nobody has yet created any PR that implements any of this.
> We discussed above some examples on how it could look like, but not all
> details are sorted out, e.g. how should it look like when non-adjacent
> wires have a common shield. And some thinking needs to be done about how to
> combine it with e.g. #439 (comment)

Hi! Ok good to know, thanks for the fast reply. We’ll check it out.

*Chloé Pilon Vaillancourt*
*Engineering Physics*
c: ***@***.***
t: +33 771398758