wireviz/WireViz #335: AWG to mm2 Conversion Error

cjohnsonoem, 2024-04-30:
Here is some simplified code to generate a cable harness using 20 AWG wire. When the output is generated, it converts 20 AWG to 0.75mm2, which is 18 AWG. Correct value should be 0.5mm2.

```
connectors:
  X1:
    type: JAE Electronics IL-AG5-6P-S3C1
    subtype: male
    pinlabels: [ACC, SUPPLY, IGN, SEC_SUPPLY, SEC_SIGNAL]

  X2:
    type: JAE Electronics IL-AG5-6S-S3C1
    subtype: female
    pinlabels: [ACC, SUPPLY, IGN, SEC_SUPPLY, SEC_SIGNAL]

cables:
  W1:
    gauge: 20 AWG
    show_equiv: true
    length: 0.125
    colors: [RD,BK,GN,BU,YE]
    wirecount: 5
    shield: false
  
connections:
  - # connect input to output
    - X1: [1-5]
    - W1: [1-5]
    - X2: [1-5]
```

[image]

It appears that the table in wv_helper.py is incorrect:

```
awg_equiv_table = {
    '0.09': '28',
    '0.14': '26',
    '0.25': '24',
    '0.34': '22',
    '0.5': '21',
    '0.75': '20',
    '1': '18',
    '1.5': '16',
    '2.5': '14',
    '4': '12',
    '6': '10',
    '10': '8',
    '16': '6',
    '25': '4',
    '35': '2',
    '50': '1',
}
```

cjohnsonoem, 2024-04-30:
Sorry, it appears this issue has already been hashed in #282

kvid, 2024-05-01:
@cjohnsonoem - Please contribute to the original issue you already found yourself! Read the comments and find out if any suggested solution would resolve your use case. Add also your own alternative thoughts. Even though it's many months since the last comment, this issue is not dead.