wireviz/WireViz #337: [feature] Add option to use DEF-STAN cable color code

scottbouch, 2024-05-05:
Hi all, 

It would be nice to be able to call up the DEF-STAN (defence standard) color codes if possible? Clearly this ispossiblength by setting cusom colours, but to access a default colour library would be good.

See reference: 
https://www.fscables.com/footer/links/colour_codes/def-stan-colour-codes.html

I'm not sure which DEF-STAN standard number defines the colours and numbering/order, be good to find this specific reference to tile the colour code.

Cheers, Scott

kvid, 2024-05-08:
It seems the 36 color codes listed at FS Cables are equal to Table 4 in UK Defence Standard 61-12 _Wires, Cords, and Cables, Electrical_ Part 4: _Cables, Special Purpose, Electrical (Sub-Miniature Electric Cables)_ Issue 4 (2005-12-19) Reprinted 2008-06-13 Incorporating amendment 1, labeled _DEF STAN 61-12 Part 4 Issue 4_ at each page.

The "_Note: The colour code of 25 core cables has core colours 23 & 24 replaced with colours 26 & 28_" at FS Cables is described a bit differently in the standard document: A 25 core cable has cores 1-3 in the center, 4-11 in first layer, and 12-22,25,26,28 in second layer, i.e. cores 23, 24, and 27 are skipped. Both descriptions end up with the same set of colors, but the order of the last 3 cores will not be the same if I understand it correctly. Please correct me and explain why if I'm wrong!

Larger cables with 50, 60, 85, 108, and 162 cores are also described, but they have only one red and one blue core adjacent to each other in each layer with all remaining cores being white.

scottbouch, 2024-05-09:
Hi @kvid ,

Thanks for that research; I have now found from Scribd a copy of DEF-STAN 61-12 Part 4 Issue 3, but I imagine the colour tables will be the same as your copy at Issue 4.

For comparison, I had a look at Eland Cables datasheets, but these don't go into this level of detail about the two rogue colours.

I think that FS Cables wording  is slightly misleading, making it sound like a swap over of 23 & 24 for 26 & 28. 

However, DEF-STAN 61-12 Part 4 states the cores of a 25 way cable listed out in the following order:
Centre: (1 to 3)
First layer: (4 to 11)
Second layer: (12 to 22, 25, 26, 28) in numerical order but with omissions of 23 and 24

FS Cables choice of wording makes it sound like the order is: (12 to 22, 26, 28, 25), but DEF-STAN 61-12 Part 4 appears much clearer about the listed order. I'm tempted to say the DEF-STAN order is likely to be the most correct, and worth following.

Cheers, Scott

scottbouch, 2024-05-09:
PS, here is an example of a 12 way DEF-STAN 61-12 Part 4 cable.

[image]

Interesting to note that Plessey Mk4 plugs and sockets were made with 2, 3, 4, 6, 12, 18, 25 way variants. These connectors first appeared circa 1950. And these standardised numbers of cores listed in DEF-STAN 61-12 Part 4 may have coincided.
See: https://scottbouch.com/mcfs/lightningt5t55/connector-104-pattern-plessey-mk4.htm