wireviz/WireViz #341: Different wire length in same cable harness?

VIPQualityPost, 2024-05-07:
We connect several power supplies into a single harness, but they are located in different places so different color wire has a few inches difference in the BOM to make everything tidy. Is it possible to do this with wireviz? I would like to move over all of our internal wiring diagrams in Excel into something nicer like wireviz but need this feature.

kvid, 2024-05-07:
You might like my earlier https://github.com/wireviz/WireViz/issues/268#issuecomment-1024807332 where I suggest how such a new feature might be. If that older issue covers your feature request in full, then please continue this discussion there.

However, when limited by existing features in the current version, you might be able to obtain your goal by creating "extension bundle(s)" for the longer wires and connect the wires to the main bundle via "virtual splices" as in my #270 work-around.

martinrieder, 2024-05-13:
The "virtual splice" method is nice, but will not calculate the overall length, since the wire "extension" is added as a separate BOM item. It could be achieved with #273 being implemented to "unite" a wire along with its virtual splices.

martinrieder, 2024-05-13:
That means the total wire length will be summed and the cable is joined into a single BOM entry when their attributes are matching. It seems that #273 would require calculating the length and qty attribute accordingly, in order to "unite" wires that run through these virtual splices. 

On the other hand, how would a true splice be handled in this calculation? What if it splits up into multiple strands of equal length?

kvid, 2024-05-13:
@martinrieder wrote:
> That means the total wire length will be summed and the cable is joined into a single BOM entry when their attributes are matching. 

Yes, as all matching wires in the harness.

> It seems that #273 would require calculating the length and qty attribute accordingly, in order to "unite" wires that run through these virtual splices.

#273 is only a suggested feature helping the user to detect unintentional differences between connected wires. The `length` attribute would not normally be included in such a difference checking.

> 
> On the other hand, how would a true splice be handled in this calculation? How do you distinguish between both if they only differ by a simple tweak to the `.gv` output? Would you imply that simply by the number of wires connected?

The lengths of matching wires will be added up the same way with virtual splices as with actual splices.

martinrieder, 2024-05-21:
The current BOM functionality only calculates an overall length for the wire. I would like to enhance that length and wire gauge calculation by its resulting resistance. This would allow checking for current carrying capacity of wires. Be aware that the latter requires to distinguish between different cable categories.

martinrieder, 2024-05-21:
I realize that the above is far from what @VIPQualityPost requested. Sorry for hijacking this issue. I shall create another one to have a feature request for the advanced wiring calculation.

To simplify the task of extending wires, I appended some other suggestion to https://github.com/wireviz/WireViz/issues/270#issuecomment-2122153851

> Okay, coming back to the original request. I suggest to have wire-to-wire connections generated in the same fashion as it was implemented for connectors. Instead of using "==", I would simply use "--" to denote this in the connection list:
> 
> ```
> # single wire to ferrule
> -
>  - WIRE
>  - "--"
> # single wire-to-wire as a virtual splice
> -
>   - WIRE1
>   - "--"
>   - WIRE2
> # multiple wires in an actual splice
> -
>   - WIRE1
>   - "--"
>   - WIRE2
> -
>   - WIRE1
>   - "--"
>   - WIRE3
> ```

@VIPQualityPost please comment if you find this useful. How would you do the splicing on your ground line, assuming that your power supplies have a common ground?