wireviz/WireViz #342: Windows build failing

VIPQualityPost, 2024-05-08:
Hi,

I installed on macos, it works fine. I tried to install on my work windows machine today, and am unable to generate even demo examples from repo: 
```
metadata = Metadata(**yaml_data.get('metadata', {})),
AttributeError: 'NoneType' object has no attribute 'get'
```

VIPQualityPost, 2024-05-08:
This was with demo02.yml from the repo.

kvid, 2024-05-10:
@VIPQualityPost - thank's for reporting this error, but please provide information about your set of versions as described in [CONTRIBUTING.md](https://github.com/wireviz/WireViz/blob/master/docs/CONTRIBUTING.md) and how you installed WireViz and dependencies. Otherwise, it might be hard to help you when nobody can recreate the error you report.

kvid, 2024-06-02:
@VIPQualityPost wrote:
> [...]
> ```
> metadata = Metadata(**yaml_data.get('metadata', {})),
> AttributeError: 'NoneType' object has no attribute 'get'
> ```

The error message you report tells me that you have no `yaml_data`, which is highly unusual. Either your YAML input is very weird (which isn't the case if you use the original "demo02.yml from the repo"), or maybe something's wrong with your `PyYAML` library installation, but I need more information from your installation setup to know.

Please try my suggestion in https://github.com/wireviz/WireViz/issues/366#issuecomment-2135646682 and report here the full output from all these commands.

VIPQualityPost, 2024-06-02:
Sorry about disappearing off the issue- have been moving and not had access to my windows machine in the meantime. I will grab that info and get you more detail in a week or two!

miroc99, 2024-06-03:
I also have the same problem, install wireviz 0.4 from pip and use demo01.yml to test but fail
os: Windows10 22H2
python: 3.12.2(venv)
wireviz: 0.4

`WireViz 0.4
Input file:   demo01.yml
Output file:  demo01.[html|png|svg|tsv]
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "d:\Dev\wireviz\.venv\Scripts\wireviz.exe\__main__.py", line 7, in <module>
  File "d:\Dev\wireviz\.venv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Dev\wireviz\.venv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "d:\Dev\wireviz\.venv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Dev\wireviz\.venv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Dev\wireviz\.venv\Lib\site-packages\wireviz\wv_cli.py", line 141, in wireviz
    wv.parse(
  File "d:\Dev\wireviz\.venv\Lib\site-packages\wireviz\wireviz.py", line 88, in parse
    yaml_data, yaml_file = _get_yaml_data_and_path(inp)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "d:\Dev\wireviz\.venv\Lib\site-packages\wireviz\wireviz.py", line 416, in _get_yaml_data_and_path
    raise e
  File "d:\Dev\wireviz\.venv\Lib\site-packages\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path
    yaml_path = Path(inp).expanduser().resolve(strict=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\miroc\AppData\Local\Programs\Python\Python312\Lib\pathlib.py", line 1240, in resolve
    s = self._flavour.realpath(self, strict=strict)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen ntpath>", line 722, in realpath
OSError: [WinError 123] 檔案名稱、目錄名稱或磁碟區標籤語法錯誤。: 'D:\\Dev\\wireviz\\connectors:\n  X1:\n    type: D-Sub\n    subtype: female\n    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]\n  X2:\n    type: Molex KK 254\n    subtype: female\n    pinlabels: [GND, RX, TX]\n\ncables:\n  W1:\n    gauge: 0.25 mm2\n    length: 0.2\n    color_code: DIN\n    wirecount: 3\n    shield: true\n\nconnections:\n  -\n    - X1: [5,2,3]\n    - W1: [1,2,3]\n    - X2: [1,3,2]\n  -\n    - X1: 5\n    - W1: s\n'`