wireviz/WireViz #343: [request] BOM quantities and units

scottbouch, 2024-05-13:
Hi, one further request...

I came across a little frustration: quantities in the BOM is limited to integers, ie, for counting items.

There are some circumstances where a measurement is needed, such as length or volume where a quantity of a material resource is cut for use.

For example:

**_Lengths:_**
- A cut length of heat-shrink sleeving to be applied over the end of a cable where a connector cable clamp or gland is applied, there is a need to specify the length.
- Cut lengths of sleeves to cover the soldered joints to connector contacts.
- Where a short wire is soldered to the cut off screen/shield of a cable for termination, it would be useful to add the length of this short wire to the connector BoM. 
- I also have some connectors requiring turns of lashing cord to provide strain-relief around the wires, it would be good to specify the length of lashing cord/thread to cut for use.

**_Volumes:_**
- Some connectors and glands are "potted" in resin or an RTV compound, so a volume quantity would be helpful too.
- Glues and thread-lock also are sometimes used, but maybe the quantity here could be "AR" for As Required.

**_Units_**
It would be great if the unit can be specified by the user for the given quantity.

Hope all of this makes sense! :)

Cheers, Scott;

kvid, 2024-05-13:
Hi Scott,

What version of WireViz are you using?

Quantities in the BOM are NOT limited to integers. Several [examples](https://github.com/wireviz/WireViz/blob/master/examples/readme.md) use decimal numbers as lengths.

> Maybe its easiest to add a text variable for unit, and allow the quantity to be a float instead of an integer.

This is exactly how it is implemented (and documented in [syntax.md](https://github.com/wireviz/WireViz/blob/master/docs/syntax.md)).

The `cable.length` has a simplified syntax accepting unit as a string after the number, separated by a space (even though it's currently implemented with an internal non-documented `length_unit` attribute behind the scene). For other quantities, an extra `unit` attribute must be used, but maybe these also will be simplified in future versions.

Default unit for `cable.length` is `m` when not specified. For other quantities, an empty unit is default (i.e. a count).

You can use any text string as unit, but note that only entries with exact same units are added up in the BOM. No unit conversion is currently implemented.

Try including something like this in your harness:
```yaml
cables:
  Long cable:
    colors: [BK, WH]
    length: 0.1 km
  Test cable:
    colors: [RD, BK]
    length: 2.5 yard
    additional_components:
      - type: Heat-shrink
        qty: 1.5
        unit: inch
      - type: Resin
        qty: 0.2
        unit: dl
additional_bom_items:
  - description: Item not in diagram
    qty: 11.5
    unit: mm
connections:
  - - Test cable
  - - Long cable
```

scottbouch, 2024-05-13:
Sorry, as you were posting this I just updated my original post...

It is in relation to connector parts, my edit was to make it specific to the connectors. I see that cables can have a length specified, but for connectors where a cut length of something is needed, I can only set an integer and no unit. This approach makes sense for back-shells, screws, cable ties, and so on; but for lashing cord, potting compound etc.. it would help if there was a float and a unit allowed for connector parts.

Thanks, Scott

kvid, 2024-05-13:
Have you tried added my `additional_components` section above to a connector?

Please provide a simple example YAML input, the output you get, and explain what you expect instead.

scottbouch, 2024-05-13:
Hi, no I hadn't - thank you, I shall give it a go, it may be the answer to my problem!!

scottbouch, 2024-05-13:
Thank you again for this support, I have edited the yaml and now have a corrected BOM, please ignore this request! You already have an ace system! Just user trouble...

Many thanks again, Scott

scottbouch, 2024-05-13:
Closed, not an issue, just was user incompetence on my part! 

Thanks for the tech support and guidance!

closed 2024-05-13 by scottbouch