wireviz/WireViz #344: [bug?] OSError: [WinError 123] with version 0.4

KlopDesign, 2024-05-13:
Thanks for the great tool! However, with version WireViz 0.4 I cannot get a good output. 

With `wireviz demo01.yml` and `wireviz tutorial01.yml` I get: 
`OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: ...`
Python version 3.7.0 using [pyenv](https://github.com/pyenv-win/pyenv-win). 

I installed WireViz 0.3.2 and that one works just fine. However, demo02.yml does not work as in #304: Exception: First item not found anywhere.

kvid, 2024-05-13:
@KlopDesign - Thank's for trying out our new release, and reporting your problem here! Can you please provide the information specified in [CONTRIBUTING.md](https://github.com/wireviz/WireViz/blob/master/docs/CONTRIBUTING.md) to enable others to re-create your error situation. Please also tell us how you installed WireViz and dependencies. I also hope you can provide a dump of your CMD window where the error message appear (hopefully also showing your latest commands, exception trace, and your current directory.

dmason1992, 2024-05-13:
I have experience a similar issue. All problems go away when reverting to version 0.3

I get the error below when trying to create any of the example diagrams

```
File "", line 198, in _run_module_as_main
File "", line 88, in _run_code
File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\Scripts\wireviz.exe\__main__.py", line 7, in

File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\click\core.py", line 1157, in __call__

return self.main(*args, **kwargs)

^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\click\core.py", line 1078, in main

rv = self.invoke(ctx)

^^^^^^^^^^^^^^^^

File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\click\core.py", line 1434, in invoke

return ctx.invoke(self.callback, **ctx.params)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\click\core.py", line 783, in invoke

return __callback(*args, **kwargs)

^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\wireviz\wv_cli.py", line 141, in wireviz

wv.parse(
File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\wireviz\wireviz.py", line 88, in parse

yaml_data, yaml_file = _get_yaml_data_and_path(inp)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\wireviz\wireviz.py", line 416, in _get_yaml_data_and_path

raise e

File "C:\Users\dmaso\AppData\Roaming\Python\Python312-32\site-packages\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path

yaml_path = Path(inp).expanduser().resolve(strict=True)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Program Files (x86)\Python312-32\Lib\pathlib.py", line 1240, in resolve

s = self._flavour.realpath(self, strict=strict)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "", line 722, in realpath
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 'C:\\Users\\dmaso\\AppData\\Local\\Programs\\Microsoft VS Code\\connectors:\n X1: &boo\n type: Molex Micro-Fit\n subtype: male\n pinlabels: [GND, VCC]\n X2: &con_power_f # define template\n type: Molex Micro-Fit\n subtype: female\n pinlabels: [GND, VCC]\n X3:\n <<: *con_power_f # create from template\n X4:\n <<: *con_power_f # create from template\n\ncables:\n W1: &wire_power # define template\n colors: [BK, RD] # number of wires implicit in color list\n gauge: 0.25 # assume mm2 if no gauge unit is specified\n show_equiv: true\n length: 0.2\n W2:\n <<: *wire_power # create from template\n W3:\n <<: *wire_power # create from template\n gauge: 20 awg\n\nconnections:\n -\n - X1: [1-2]\n - W1: [1-2]\n - X2: [1-2]\n -\n - X1: [1-2]\n - W2: [1-2]\n - X3: [1-2]\n -\n - X1: [1-2]\n - W3: [1-2]\n - X4: [1-2]'

WireViz 0.4
Input file: c:\Users\dmaso\Documents\Github\Casa-enclosure\Wiring\LDOextension.yml
Output file: c:\Users\dmaso\Documents\Github\Casa-enclosure\Wiring\output\LDOextension\LDOextension.[html|png|svg|tsv]
```

freiadam, 2024-05-14:
Same issue here. I have installed the new version with "pip install --upgrade wireviz"

alecraymond, 2024-05-14:
I also have this issue when trying to call `wireviz` directly. `WireViz\src\wireviz\build_examples.py` works fine.

WireViz 0.4
dot - graphviz version 11.0.0 (20240428.1522)
Python 3.12.3
Microsoft Windows 10 Pro 10.0.19045 N/A Build 19045

kvid, 2024-05-14:
It seems you have found a bug that isn't triggered by the tests we have done. How do you execute WireViz? Please include a copy of the command line including arguments. If you call it from another code or script, please include that, but please strip it down to a minimum if it's large.

freiadam, 2024-05-14:
I cd into the directory of the.yml file, then simply "wireviz u1.yml" like before.

alecraymond, 2024-05-14:
Windows Command Prompt:
```
Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ARaymond>cd C:\Repos\WireViz\examples

C:\Repos\WireViz\examples>git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

C:\Repos\WireViz\examples>wireviz demo01.yml

WireViz 0.4
Input file:   demo01.yml
Output file:  demo01.[html|png|svg|tsv]
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Scripts\wireviz.exe\__main__.py", line 7, in <module>
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Repos\WireViz\src\wireviz\wv_cli.py", line 141, in wireviz
    wv.parse(
  File "C:\Repos\WireViz\src\wireviz\wireviz.py", line 88, in parse
    yaml_data, yaml_file = _get_yaml_data_and_path(inp)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Repos\WireViz\src\wireviz\wireviz.py", line 416, in _get_yaml_data_and_path
    raise e
  File "C:\Repos\WireViz\src\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path
    yaml_path = Path(inp).expanduser().resolve(strict=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\Lib\pathlib.py", line 1240, in resolve
    s = self._flavour.realpath(self, strict=strict)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen ntpath>", line 722, in realpath
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 'C:\\Repos\\WireViz\\examples\\connectors:\n  X1:\n    type: D-Sub\n    subtype: female\n    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]\n  X2:\n    type: Molex KK 254\n    subtype: female\n    pinlabels: [GND, RX, TX]\n\ncables:\n  W1:\n    gauge: 0.25 mm2\n    length: 0.2\n    color_code: DIN\n    wirecount: 3\n    shield: true\n\nconnections:\n  -\n    - X1: [5,2,3]\n    - W1: [1,2,3]\n    - X2: [1,3,2]\n  -\n    - X1: 5\n    - W1: s\n'

C:\Repos\WireViz\examples>
```

kvid, 2024-05-14:
I'm not able to reproduce the reported OSError here, but I notice my Python version is older than what @alecraymond uses, but newer than what @KlopDesign uses. My Graphviz versions is also older than what @alecraymond uses, but as far as I know, the Graphviz `dot` command has not yet been called at the place in the code where the OSError is raised. It's actually before parsing the input YAML, so I assume the Graphviz version should not matter. The log below is my test from a fresh installation similar to what @alecraymond reported.

I hope more users (both those who see this error and those who don't) will report their version numbers, because that can help finding a pattern about what is working and what is not.
```
Microsoft Windows [Version 10.0.19045.4170]
(c) Microsoft Corporation. All rights reserved.

C:\Users\kvid>cd \

C:\>mkdir Repos

C:\>cd Repos

C:\Repos>git clone git@github.com:wireviz/WireViz.git
Cloning into 'WireViz'...
remote: Enumerating objects: 4581, done.
remote: Counting objects: 100% (1530/1530), done.
remote: Compressing objects: 100% (389/389), done.
remote: Total 4581 (delta 1108), reused 1348 (delta 1026), pack-reused 3051
Receiving objects: 100% (4581/4581), 17.71 MiB | 7.58 MiB/s, done.
Resolving deltas: 100% (2944/2944), done.

C:\Repos>cd WireViz

C:\Repos\WireViz>python3 -m venv .venv

C:\Repos\WireViz>.venv\Scripts\activate.bat

(.venv) C:\Repos\WireViz>pip install --editable=.
Obtaining file:///C:/Repos/WireViz
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting pillow
  Downloading pillow-10.3.0-cp39-cp39-win_amd64.whl (2.5 MB)
     ---------------------------------------- 2.5/2.5 MB 9.5 MB/s eta 0:00:00
Collecting pyyaml
  Using cached PyYAML-6.0.1-cp39-cp39-win_amd64.whl (152 kB)
Collecting graphviz
  Downloading graphviz-0.20.3-py3-none-any.whl (47 kB)
     ---------------------------------------- 47.1/47.1 KB 2.5 MB/s eta 0:00:00
Collecting click
  Using cached click-8.1.7-py3-none-any.whl (97 kB)
Collecting colorama
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Building wheels for collected packages: wireviz
  Building editable for wireviz (pyproject.toml) ... done
  Created wheel for wireviz: filename=wireviz-0.4-0.editable-py3-none-any.whl size=16533 sha256=79972fe007caaf9a32a0a766f08ddc4c0ead4739c059206211a7c5ccb75c7fe7
  Stored in directory: C:\Users\kvid\AppData\Local\Temp\pip-ephem-wheel-cache-k1ko3huk\wheels\6a\8a\6d\0113f8d91acb38ed269ee904f50f80ae4f95e69925a8385f69
Successfully built wireviz
Installing collected packages: pyyaml, pillow, graphviz, colorama, click, wireviz
Successfully installed click-8.1.7 colorama-0.4.6 graphviz-0.20.3 pillow-10.3.0 pyyaml-6.0.1 wireviz-0.4
WARNING: You are using pip version 22.0.4; however, version 24.0 is available.
You should consider upgrading via the 'C:\Repos\WireViz\.venv\Scripts\python.exe -m pip install --upgrade pip' command.

(.venv) C:\Repos\WireViz>C:\Repos\WireViz\.venv\Scripts\python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:\repos\wireviz\.venv\lib\site-packages (22.0.4)
Collecting pip
  Downloading pip-24.0-py3-none-any.whl (2.1 MB)
     ---------------------------------------- 2.1/2.1 MB 8.4 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-24.0

(.venv) C:\Repos\WireViz>cd examples

(.venv) C:\Repos\WireViz\examples>wireviz demo01.yml

WireViz 0.4
Input file:   demo01.yml
Output file:  demo01.[html|png|svg|tsv]

(.venv) C:\Repos\WireViz\examples>wireviz -V

WireViz 0.4

(.venv) C:\Repos\WireViz\examples>dot -V
dot - graphviz version 2.44.1 (20200629.0846)

(.venv) C:\Repos\WireViz\examples>python -V
Python 3.9.13

(.venv) C:\Repos\WireViz\examples>pip list
Package    Version Editable project location
---------- ------- -------------------------
click      8.1.7
colorama   0.4.6
graphviz   0.20.3
pillow     10.3.0
pip        24.0
PyYAML     6.0.1
setuptools 58.1.0
wireviz    0.4     C:\Repos\WireViz

(.venv) C:\Repos\WireViz\examples>ver

Microsoft Windows [Version 10.0.19045.4170]

(.venv) C:\Repos\WireViz\examples>
```

alecraymond, 2024-05-14:
Trying to replicate your results, I noticed some differences:
1. My `Repos` directory is not otherwise empty.
2. `python3` required me to install python through the Microsoft Store. The same version was installed, but there may still be an issue with python installations and how my PATH is set up.
3. `setuptools` was not listed among my pip packages.

I will try again using an empty starting directory, python 3.9.13, and `setuptools` installed.
```
Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ARaymond>cd \

C:\>mkdir Repos
A subdirectory or file Repos already exists.

C:\>cd Repos

C:\Repos>git clone git@github.com:wireviz/WireViz.git
Cloning into 'WireViz'...
remote: Enumerating objects: 4581, done.
remote: Counting objects: 100% (1520/1520), done.
remote: Compressing objects: 100% (390/390), done.
Receiving objects: 100% (4581/4581), 15.72 MiB | 5.10 MiB/spack-reused 3061
Receiving objects: 100% (4581/4581), 17.71 MiB | 4.98 MiB/s, done.
Resolving deltas: 100% (2942/2942), done.

C:\Repos>cd WireViz

C:\Repos\WireViz>python3 -m venv .venv
Python was not found; run without arguments to install from the Microsoft Store, or disable this shortcut from Settings > Manage App Execution Aliases.

C:\Repos\WireViz>python3

C:\Repos\WireViz>python3 -m venv .venv

C:\Repos\WireViz>.venv\Scripts\activate.bat

(.venv) C:\Repos\WireViz>pip install --editable=.
Obtaining file:///C:/Repos/WireViz
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting click (from wireviz==0.4)
  Using cached click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
Collecting pyyaml (from wireviz==0.4)
  Using cached PyYAML-6.0.1-cp312-cp312-win_amd64.whl.metadata (2.1 kB)
Collecting pillow (from wireviz==0.4)
  Using cached pillow-10.3.0-cp312-cp312-win_amd64.whl.metadata (9.4 kB)
Collecting graphviz (from wireviz==0.4)
  Using cached graphviz-0.20.3-py3-none-any.whl.metadata (12 kB)
Collecting colorama (from click->wireviz==0.4)
  Using cached colorama-0.4.6-py2.py3-none-any.whl.metadata (17 kB)
Using cached click-8.1.7-py3-none-any.whl (97 kB)
Using cached graphviz-0.20.3-py3-none-any.whl (47 kB)
Using cached pillow-10.3.0-cp312-cp312-win_amd64.whl (2.5 MB)
Using cached PyYAML-6.0.1-cp312-cp312-win_amd64.whl (138 kB)
Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Building wheels for collected packages: wireviz
  Building editable for wireviz (pyproject.toml) ... done
  Created wheel for wireviz: filename=wireviz-0.4-0.editable-py3-none-any.whl size=16533 sha256=d49cee2a1bfc9830a485ad3cf242355cc9d16b69573bc80de9fd6a27e93596ca
  Stored in directory: C:\Users\ARaymond\AppData\Local\Temp\pip-ephem-wheel-cache-n6t7byeg\wheels\73\67\6f\41bdb497ce48eb774f69cc82c3d47e9400fe15f003ebafe267
Successfully built wireviz
Installing collected packages: pyyaml, pillow, graphviz, colorama, click, wireviz
Successfully installed click-8.1.7 colorama-0.4.6 graphviz-0.20.3 pillow-10.3.0 pyyaml-6.0.1 wireviz-0.4

(.venv) C:\Repos\WireViz>cd examples

(.venv) C:\Repos\WireViz\examples>wireviz demo01.yml
[... 46 lines cut ...]
click    8.1.7
colorama 0.4.6
graphviz 0.20.3
pillow   10.3.0
pip      24.0
PyYAML   6.0.1
wireviz  0.4     C:\Repos\WireViz

(.venv) C:\Repos\WireViz\examples>ver

Microsoft Windows [Version 10.0.19045.4291]

(.venv) C:\Repos\WireViz\examples>python3 -V
Python 3.12.3

(.venv) C:\Repos\WireViz\examples>where python
C:\Repos\WireViz\.venv\Scripts\python.exe
C:\Users\ARaymond\AppData\Local\Programs\Python\Python312\python.exe
C:\Users\ARaymond\AppData\Local\Microsoft\WindowsApps\python.exe

(.venv) C:\Repos\WireViz\examples>where python3
C:\Users\ARaymond\AppData\Local\Microsoft\WindowsApps\python3.exe

(.venv) C:\Repos\WireViz\examples>
```

kvid, 2024-05-14:
I have updated my log above to also include the extra commands @alecraymond just used, in case it might help to compare...

alecraymond, 2024-05-14:
Another attempt with all other versions of python removed. There may be an issue with my graphviz installation this time, since `dot -V` was not recognized. I found the old source for `dot - graphviz version 2.44.1 (20200629.0846)` but I'll have to build it to more closely match @kvid's setup. I don't have any more time for the next couple of days, but I'll keep looking if the issue is still unresolved by then.

```
Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ARaymond>cd \

C:\>mkdir Test

C:\>cd Test

C:\Test>git clone git@github.com:wireviz/WireViz.git
Cloning into 'WireViz'...
remote: Enumerating objects: 4581, done.
remote: Counting objects: 100% (1530/1530), done.
remote: Compressing objects: 100% (389/389), done.
Receiving objects: 100% (4581/4581), 17.71 MiB | 1.84 MiB/s, done.used 3051

Resolving deltas: 100% (2944/2944), done.

C:\Test>cd WireViz

C:\Test\WireViz>python -m venv .venv

C:\Test\WireViz>.venv\Scripts\activate.bat

(.venv) C:\Test\WireViz>pip install --editable=.
Obtaining file:///C:/Test/WireViz
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting click
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ---------------------------------------- 97.9/97.9 KB 329.7 kB/s eta 0:00:00
Collecting pillow
  Downloading pillow-10.3.0-cp39-cp39-win_amd64.whl (2.5 MB)
     ---------------------------------------- 2.5/2.5 MB 1.5 MB/s eta 0:00:00
Collecting graphviz
  Downloading graphviz-0.20.3-py3-none-any.whl (47 kB)
     ---------------------------------------- 47.1/47.1 KB 784.8 kB/s eta 0:00:00
Collecting pyyaml
  Downloading PyYAML-6.0.1-cp39-cp39-win_amd64.whl (152 kB)
     ---------------------------------------- 152.8/152.8 KB 2.3 MB/s eta 0:00:00
Collecting colorama
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Building wheels for collected packages: wireviz
  Building editable for wireviz (pyproject.toml) ... done
  Created wheel for wireviz: filename=wireviz-0.4-0.editable-py3-none-any.whl size=16529 sha256=de753171b408b7fc404211ee2fe2f6efc0737df67194ee1b6b8679976a88ccfe
  Stored in directory: C:\Users\ARaymond\AppData\Local\Temp\pip-ephem-wheel-cache-9p2m7_c_\wheels\f3\72\16\3640e3c047432e747545c4c8b43a910247d9dc7d843774f947
Successfully built wireviz
Installing collected packages: pyyaml, pillow, graphviz, colorama, click, wireviz
Successfully installed click-8.1.7 colorama-0.4.6 graphviz-0.20.3 pillow-10.3.0 pyyaml-6.0.1 wireviz-0.4
WARNING: You are using pip version 22.0.4; however, version 24.0 is available.
You should consider upgrading via the 'C:\Test\WireViz\.venv\Scripts\python.exe -m pip install --upgrade pip' command.

(.venv) C:\Test\WireViz>C:\Test\WireViz\.venv\Scripts\python.exe -m pip install --upgrade pip
Requirement already satisfied: pip in c:\test\wireviz\.venv\lib\site-packages (22.0.4)
Collecting pip
  Downloading pip-24.0-py3-none-any.whl (2.1 MB)
     ---------------------------------------- 2.1/2.1 MB 4.1 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 22.0.4
    Uninstalling pip-22.0.4:
      Successfully uninstalled pip-22.0.4
Successfully installed pip-24.0

(.venv) C:\Test\WireViz>cd examples
[... 45 lines cut ...]
(.venv) C:\Test\WireViz\examples>pip list
Package    Version Editable project location
---------- ------- -------------------------
click      8.1.7
colorama   0.4.6
graphviz   0.20.3
pillow     10.3.0
pip        24.0
PyYAML     6.0.1
setuptools 58.1.0
wireviz    0.4     C:\Test\WireViz

(.venv) C:\Test\WireViz\examples>python3 -V
'python3' is not recognized as an internal or external command,
operable program or batch file.

(.venv) C:\Test\WireViz\examples>where python
C:\Test\WireViz\.venv\Scripts\python.exe
C:\Users\ARaymond\AppData\Local\Programs\Python\Python39\python.exe

(.venv) C:\Test\WireViz\examples>where python3
INFO: Could not find files for the given pattern(s).

(.venv) C:\Test\WireViz\examples>
```

JarrettR, 2024-05-15:
I'm getting the same issue.

Whatever the root cause is, the issue seems to be that the error type raised isn't `ENAMETOOLONG`, it's `EINVAL`.

My fix, without understanding any of the underlying code:

At https://github.com/wireviz/WireViz/blob/master/src/wireviz/wireviz.py#L413

Change the next few lines to:
```
            from errno import ENAMETOOLONG, EINVAL

            if type(e) is OSError and e.errno != ENAMETOOLONG and e.errno != EINVAL:
                raise e
```

kvid, 2024-05-15:
@JarrettR - Thank's a lot for finding the errno value of the raised OSError and suggesting the fix! I've not yet been able to reproduce this error myself. Since you posted the fix here instead of creating a PR, I took the liberty to create PR #346 and adding you as an author. Some users find it easier to test such a fix from an existing branch instead of modifying the code themselves.

If you disagree how I did this, please explain how you rather want it, and we can look into it.

@KlopDesign @dmason1992 @freiadam @alecraymond @JarrettR and others experiencing this error, please test the fix in the branch of PR #346 and report back the outcome together with your OS and versions.

alecraymond, 2024-05-17:
The fix works for me:
```
Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.

C:\Repos\WireViz\examples>wireviz demo01.yml

WireViz 0.4
Input file:   demo01.yml
Output file:  demo01.[html|png|svg|tsv]

C:\Repos\WireViz\examples>ver

Microsoft Windows [Version 10.0.19045.4291]

C:\Repos\WireViz\examples>python -V
Python 3.12.3

C:\Repos\WireViz\examples>wireviz -V

WireViz 0.4

C:\Repos\WireViz\examples>dot -V
dot - graphviz version 11.0.0 (20240428.1522)

C:\Repos\WireViz\examples>pip list
Package  Version Editable project location
-------- ------- -------------------------
click    8.1.7
colorama 0.4.6
graphviz 0.20.3
pillow   10.3.0
pip      24.0
PyYAML   6.0.1
wireviz  0.4     C:\Repos\WireViz

C:\Repos\WireViz\examples>git diff
diff --git a/src/wireviz/wireviz.py b/src/wireviz/wireviz.py
index db69241..5fc3ee1 100755
--- a/src/wireviz/wireviz.py
+++ b/src/wireviz/wireviz.py
@@ -408,11 +408,12 @@ def _get_yaml_data_and_path(inp: Union[str, Path, Dict]) -> (Dict, Path):
             yaml_str = open_file_read(yaml_path).read()
         except (FileNotFoundError, OSError) as e:
             # if inp is a long YAML string, Pathlib will raise OSError: [errno.ENAMETOOLONG]
+            # (in Windows, it seems OSError [errno.EINVAL] might be raised in some cases)
             # when trying to expand and resolve it as a path.
             # Catch this error, but raise any others
-            from errno import ENAMETOOLONG
+            from errno import EINVAL, ENAMETOOLONG

-            if type(e) is OSError and e.errno != ENAMETOOLONG:
+            if type(e) is OSError and e.errno not in (EINVAL, ENAMETOOLONG):
                 raise e
             # file does not exist; assume inp is a YAML string
             yaml_str = inp

C:\Repos\WireViz\examples>git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   ../src/wireviz/wireviz.py

no changes added to commit (use "git add" and/or "git commit -a")
```

Basilisvirus, 2024-07-05:
I can confirm the issue.
After I updated from version `wireviz 0.3.2` (which worked as expected):
`PS C:\Users\Vasilis> pip3 install --upgrade wireviz`
my new version is: `WireViz 0.4` and when I try to compile a .yml file I get:

```
PS C:\Program Files\Graphviz\bin> wireviz source.yml

WireViz 0.4
Input file:   source.yml
Output file:  source.[html|png|svg|tsv]
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python312\Scripts\wireviz.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wv_cli.py", line 141, in wireviz
    wv.parse(
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 88, in parse
    yaml_data, yaml_file = _get_yaml_data_and_path(inp)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 416, in _get_yaml_data_and_path
    raise e
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path
    yaml_path = Path(inp).expanduser().resolve(strict=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\pathlib.py", line 1240, in resolve
    s = self._flavour.realpath(self, strict=strict)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen ntpath>", line 722, in realpath
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 'C:\\Program Files\\Graphviz\\bin\\connectors:\n  X1:\n    type: D-Sub\n    subtype: female\n    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]\n  X2:\n    type: Molex KK 254\n    subtype: female\n    pinlabels: [GND, RX, TX]\n\ncables:\n  W1:\n    gauge: 0.25 mm2\n    length: 0.2\n    color_code: DIN\n    wirecount: 3\n    shield: true\n\nconnections:\n  -\n    - X1: [5,2,3]\n    - W1: [1,2,3]\n    - X2: [1,3,2]\n  -\n    - X1: 5\n    - W1: s'
```

Windows version:

[image]

Python version: 

```
PS C:\Program Files\Graphviz\bin> python --version
Python 3.12.3

```
After trying the fix, I still get the error, but it changed a little bit:

```
PS C:\Program Files\Graphviz\bin> wireviz source.yml

WireViz 0.4
Input file:   source.yml
Output file:  source.[html|png|svg|tsv]
Traceback (most recent call last):
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path
    yaml_path = Path(inp).expanduser().resolve(strict=True)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\pathlib.py", line 1240, in resolve
    s = self._flavour.realpath(self, strict=strict)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen ntpath>", line 722, in realpath
OSError: [WinError 123] The filename, directory name, or volume label syntax is incorrect: 'C:\\Program Files\\Graphviz\\bin\\connectors:\n  X1:\n    type: D-Sub\n    subtype: female\n    pinlabels: [DCD, RX, TX, DTR, GND, DSR, RTS, CTS, RI]\n  X2:\n    type: Molex KK 254\n    subtype: female\n    pinlabels: [GND, RX, TX]\n\ncables:\n  W1:\n    gauge: 0.25 mm2\n    length: 0.2\n    color_code: DIN\n    wirecount: 3\n    shield: true\n\nconnections:\n  -\n    - X1: [5,2,3]\n    - W1: [1,2,3]\n    - X2: [1,3,2]\n  -\n    - X1: 5\n    - W1: s'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Program Files\Python312\Scripts\wireviz.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wv_cli.py", line 141, in wireviz
    wv.parse(
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 88, in parse
    yaml_data, yaml_file = _get_yaml_data_and_path(inp)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python312\Lib\site-packages\wireviz\wireviz.py", line 415, in _get_yaml_data_and_path
    if type(e) is OSError and e.errno != ENAMETOOLONG and e.errno != EINVAL:
                                                                     ^^^^^^
NameError: name 'EINVAL' is not defined
```
Saying that the EINVAL is not defined.

kvid, 2024-07-05:
The relevant fixes are merged into the `release/v0.4.1-rc` branch of PR #365 that is not yet released, but also describes how you can install the collection of fixes before the release.

kvid, 2024-07-13:
Closed after releasing v0.4.1.

PR #346 (2024-06-04): Handle OSError [errno.EINVAL] that might be raised in Windows
Suggested fix posted in
https://github.com/wireviz/WireViz/issues/344#issuecomment-2113476151

Please verify that this fixes #344 and report back.

~~The base branch of this PR is `master` for the time beeing, but it should be changed to a new release/v0.4.1-rc before any merge-in. #345 should also be fixed and updating the changelog before releasing the hotfix.~~
closed 2024-07-13 by kvid