wireviz/WireViz #348: [feature] Version compatibility checks

martinrieder, 2024-05-17:
With respect to the latest  backwards-incompatible changes, input files may require some compatibility check. It is quite common that drawings are revised after many years. May I suggest adding an (optional) "syntax version" tag? How else could it be assured that future versions of WireViz support the syntax of an older one?

Note that following issue was raised by @nanangp for adding versioning checks _on the CLI syntax_ as well:
https://github.com/nanangp/vscode-wireviz-preview/issues/6

Going a bit further, some of the settings given as a command line argument, could also be listed in a header of the YAML file. This could either be a comment or incorporated in the WireViz syntax, like the technical drawing changes require adding some general info for the title block as well.

martinrieder, 2024-05-21:
I changed the title because I believe that schema validation provides just what we need here. It allows to check your YAML even before running WireViz.

martinrieder, 2024-05-28:
Found this as a starting point: https://github.com/cooked/vscode-wireviz/blob/master/wireviz-yaml-schema.json

martinrieder, 2024-06-09:
Just providing a quick update: 
- I have been fiddling with JSON Schema. It is a powerful tool, but things quickly get complex. 
- I am planning to evaluate Yamale and compare both solutions. It seems easier to define the schema in YAML.

Any comments on this are welcome. The main question that arises: Do we need Schema validation for live editing, i.e. in VS Code.

martinrieder, 2024-06-10:
I have attached the YAML Schema for Yamale. GH would not accept the file extension, so I had to rename it to .TXT.

[schema.yaml.txt](https://github.com/user-attachments/files/15774839/schema.yaml.txt)

Running Yamale with `schema.yaml` in the WireViz folder will validate all YAML files from there. It even found that the GitHub Actions would not match and that there is an issue with `ex14.yml`...

```
C:\Git\martinrieder\WireViz>yamale --no-strict -n 1 -s schema.yaml
Finding yaml files...
Found 26 yaml files.
Validating...
Validation failed!
Error validating data 'C:\Git\martinrieder\WireViz\.github\workflows\main.yml' with schema 'schema.yaml'
        connectors: Required field missing
        cables: Required field missing
        connections: Required field missing
----
Error validating data 'C:\Git\martinrieder\WireViz\examples\ex14.yml' with schema 'schema.yaml'
        connections.0.7: '<=>' is not a designator.
        connections.0.7: '<=>' is not a list.
        connections.0.7: '<=>' is not a map.
        connections.0.7: '<=>' not in ('--', '<--', '<-->', '-->', '==', '<==', '<==>', '==>')

```

kvid, 2024-06-10:
@martinrieder wrote:
> [...]
> ```
> C:\Git\martinrieder\WireViz>yamale --no-strict -n 1 -s schema.yaml
> Finding yaml files...
> Found 26 yaml files.
> Validating...
> Validation failed!
> Error validating data 'C:\Git\martinrieder\WireViz\.github\workflows\main.yml' with schema 'schema.yaml'

`.github\workflows\main.yml` is not a WireViz input file, and uses another syntax.

>         connectors: Required field missing
>         cables: Required field missing
>         connections: Required field missing

None of these are required to be valid WireViz input. All documented top-level keys are optional, and any other top-level keys are just ignored. A random non-documented top-level key is often used to make templates to inherit later in the YAML input, e.g. in `demo02.yml`, `ex05.yml`, and `ex06.yml`. The syntax documentation is currently not very clear about this.

WireViz will happily process YAML input including only a single key, and it doesn't have to be one of the documented ones. Any non-dict as top-level will fail because `yaml_data` is assumed to be a dict.

Do you want to validate what WireViz really accept, or a stricter, more limited syntax interpreted from a syntax description not written precisely enough for validation?

With simple use cases it makes sense to drop top-level sections without contents. With v0.4, `connections` is required to obtain a non-empty diagram, but e.g. #267 is a use case where only `additional_bom_items` is needed.

> ----
> Error validating data 'C:\Git\martinrieder\WireViz\examples\ex14.yml' with schema 'schema.yaml'
>         connections.0.7: '<=>' is not a designator.
>         connections.0.7: '<=>' is not a [...]

Your validation rules are stricter than what WireViz accepts. The syntax documentation doesn't list all legal variations. See https://github.com/wireviz/WireViz/blob/v0.4/src/wireviz/wv_helper.py#L128-L140

martinrieder, 2024-06-10:
@kvid wrote:
> Do you want to validate what WireViz really accept, or a stricter, more limited syntax interpreted from a syntax description not written precisely enough for validation?

Neither of both. This is just an investigation about finding a tool that would suit the following purpose:

- Validation at runtime: simple compatibility check for old YAML files
- Validation while editing: syntax highlighting and code completion 

> Your validation rules are stricter than what WireViz accepts. The syntax documentation doesn't list all legal variations. See https://github.com/wireviz/WireViz/blob/v0.4/src/wireviz/wv_helper.py#L128-L140

Thanks for the hints. Copying the syntax description into the schema format that Yamale expects was quite easy. You proved that this manual process is not ideal though. Yet, the resulting file is still  human-readable, kind of self-explaining... 

The point that @liambeguin is making in https://github.com/wireviz/WireViz/issues/220#issuecomment-2158760443 is absolutely valid. If the schema could be generated from code, that would make it much easier to maintain.

martinrieder, 2024-06-14:
Uploading my DRAFT of the JSON Schema file here. Please validate your files with this and let me know if I missed something.

[jsonschema.json](https://github.com/user-attachments/files/15838370/jsonschema.json)

The file structure is based on the Yamale schema, which in turn is an interpretation of the manually written syntax description.
I did some adaptions after looking at DataClasses.py and thereby noticed that `image` was missing the attributes `scale` and `fixedsize`.

martinrieder, 2026-10-23:
Updated JSON Schema here: https://gist.github.com/martinrieder/e0807864c7a3fec3a312d5b6afba00f8