wireviz/WireViz #350: Way to implement Jumpers/Interconnects

tobiasfalk, 2024-05-18:
in #286 I suggested a way to implement Jumpers/Interconnects, this is a summery/spinoff on the request of @kvid.

The Idea how it could look (very similar to https://github.com/wireviz/WireViz/issues/48#issuecomment-656802355):
[image]

# Black Circles ⬤

The Black circles can be implemented by adding the following to the table in the corresponding Pin/Way row:
```
<td  port="p1J"><FONT FACE="Sans Not-Rotated 14" POINT-SIZE="13.0" COLOR="#0066ff">⬤</FONT></td>
```
or
```
<td  port="p1J"><FONT FACE="Sans Not-Rotated 14" POINT-SIZE="13.0" COLOR="#0066ff">&#11044;</FONT></td>
```
- The Port is later used to draw the vertical dashed lines.
- The Font definition is because the PNG generator's Arial font does not know the ⬤ symbol.
- The Point size change is because the Circle is clipped at the bottom, if it is not done.
- The Color is to make differentiation easier, the lines should be the same color as the corresponding  ⬤

Doing this will lead to the following:
[image]

# Vertical dashed lines
This is more complicated and would require a change in how the output is generated, and I do not know how to implement this in Python.
The way I was told to do it(asked about it in the [GrapghViz Forum](https://forum.graphviz.org/t/straitening-one-line-throu-a-table/2196/6) ) would require the use of gvpr and neato, with are post processors for the dot output.
I is currently not working, but I am working on it. 
Some examples of my current Output:
[image]

[image]

tobiasfalk, 2024-05-18:
So the wertical lines also work:
# Vertical dashed lines
they work by the following:
## the .gv file line definition
```
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    X2:p3J:c -- X2:p4J:c[straight=true]

    edge [color="#0066ff" headclip=false, tailclip=false, style="dashed", constraint=false]
	X2:p1J:c -- X2:p5J:c[straight=true]
    X2:p5J:c -- X2:p9J:c[straight=true]

    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    X2:p2J:c -- X2:p6J:c[straight=true]
```

- `headclip=false` & `tailclip=false` to make the lince start and stop in the center of the "port"/circle
- `style="dashed"` to make the line dashed(duh)
- `constraint=false` just to prevent any problems with ranking, do not know if it realy is needed but better safe than sorry
- `[straight=true]` adds the atribute `straight` to a line, this is later needed by gvpr to straiten out all the lines that have this attribut, setting it false does nothing so maybe doing it by `[straight=straight]` would be better

## the gvpr post prossecing
for this one needs to create a "filter"/process file(do not know the thermology) with the following in it(`pin2pin.gvpr`):
```
/*******************************************************************

  see https://forum.graphviz.org/t/straitening-one-line-throu-a-table/2196
  input must include pos values (must be output from one of the engines w/ -Tdot)
  
*******************************************************************/
BEG_G{
  string tok[int];
  int cnt;
}
/********************************************************
  the weakness is in the head==tail test
  it would be better to mark every target edge with a new attribute and test for that attribute
*********************************************************/
E[straight] {
  cnt=tokens($.pos,tok," ");
  $.oldpos=$.pos;
  pos=tok[0]+" " + tok[0]+" "+tok[cnt-1]+" "+tok[cnt-1];
}
```
this is some magic I do not understand but I think it removes all the cureved point in a line that has the attribute `straight`.

## Generating the Output
this is done by the folloing Comand: `dot .\test.gv | gvpr -cf pin2pin.gvpr | neato -n2 -Tpng -o file.png`

- `dot .\test.gv` your typical dot comand
- `gvpr -cf pin2pin.gvpr` takes the output of the dot comand and process it with the `pin2pin.gvpr` file
- `neato -n2 -Tpng -o file.png` takes the output of gvpr and converts it in to the wished output format (in this case png)

## Remarks
As mentioned I do not know how to Implement this in python but it works manualy at least.

## Result
[image]

the input gb file:
```
graph {
// Graph generated by WireViz 0.3.2
// https://github.com/formatc1702/WireViz
	graph [bgcolor="#FFFFFF" fontname=arial nodesep=0.33 rankdir=LR ranksep=2]
	node [fillcolor="#FFFFFF" fontname=arial height=0 margin=0 shape=none style=filled width=0]
	edge [fontname=arial style=bold label=" "]

	X2 [label=<
<table border="0" cellspacing="0" cellpadding="0">
[... 179 lines cut ...]
    </td>
   </tr>
   <tr>
    <td><!-- 5_in --></td>
    <td>
     5:GN
    </td>
    <td>X2:9:RS-485 Shield</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w5" height="6">
     <table cellspacing="0" cellborder="0" border="0">
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#00ff00" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
     </table>
    </td>
   </tr>
   <tr><td>&nbsp;</td></tr>
  </table>
 </td></tr>
</table>
> fillcolor="#FFFFFF" shape=box style=filled]
}
```

kvid, 2024-05-18:
I'm curious - why do you need to specify font face, point-size, and color? It seems to work without in my test.

tobiasfalk, 2024-05-18:
> I'm curious - when do you need to specify font face, point-size, and color? Is it for `neato` only? It seems to work without in my simple test to PNG with dot - not adding your filter commands.

When I not specified the Font and point size the bloke circle was cut of at the bottom. 
The color would be if one wants to habe the different jumpers in different colors, I would leave the standard black, this was just for when the feature is needed/requested/wanted
I am not at home but when I am, I will send a picture

tobiasfalk, 2024-05-18:
The first (Pin 3) is `<FONT FACE="Sans Not-Rotated 14" POINT-SIZE="13.0">&#11044;</FONT>`
The second (Pin 4) is `<FONT>&#11044;</FONT>`
The third (Pin 5) is `><FONT FACE="Sans Not-Rotated 14">&#11044;</FONT>`
Done with `dot .\test.gv | gvpr -cf pin2pin.gvpr | neato -n2 -Tpng -o file.png`
[image]

tobiasfalk, 2024-05-18:
Ok, the `FACE="Sans Not-Rotated 14"` can be changed to `FACE="Sans"`

tobiasfalk, 2024-05-20:
Updated the code of `pin2pin.gvpr` above, to prevent creation of white/not used space

olmari, 2024-05-27:
Very new user of WireViz, and I already love it. This is exactly one big thing I miss at the moment too, so +1 from me <3

The very ongoing documentation happens https://github.com/vaasahacklab/gatekeeper/tree/new_hardware_docs/docs and indeed all the current "jumpers" would be really jumpers, so this would be so spot on :)

kvid, 2024-05-28:
@olmari - Be aware that both #286 and #48 might be worth visiting. Please try the work-around I suggested in https://github.com/wireviz/WireViz/issues/286#issuecomment-2094309143 that might help you while waiting for better solutions.

tobiasfalk, 2024-05-29:
https://gitlab.com/graphviz/graphviz/-/issues/2547

kvid, 2024-06-01:
@tobiasfalk - I'm impressed by your effort with this issue, and obtaining a great result in such a short time!

I'm sorry that I have not yet had time to play with your solutions. I help maintaining this on my spare time (which in some periods is very limited), and have lately given priority to all the bugs after releasing v0.4.

I didn't want to give you any critical comments in your creative process, because the work you have done here is highly valuable for this project - also beyond this issue. I guess #3, #120, #330, and maybe others can benefit from your work on post processing the dot output.

However, now when you have a complete solution, I must mention a small concern I have about the dashed lines. During my work on #330, I realized dashed lines are used a lot for shielding in the relevant IEEE and IEC standards, and I wonder if a solid line would be better for your straight jumpers - maybe a bit thicker than the table border lines? I know it's simple to implement such a change, but can you first find references to how such jumpers are drawn in other diagrams, so we can compare this to any established practice?

tobiasfalk, 2024-06-01:
> However, now when you have a complete solution, I must mention a small concern I have about the dashed lines. During my work on #330, I realized dashed lines are used a lot for shielding in the [...]

I understand, I will compile some examples with different line styles and post them here.

>I know it's simple to implement such a change, but can you first find references to how such jumpers are drawn in other diagrams, so we can compare this to any established practice?

I have done this at the beginning but also just now I did a short search again, and did not find any reference to Jumpers/Interconnects/Shorting pins in the IEEE/IEC/ISO standards, but it could be that I am just not searching with the right terms.

tobiasfalk, 2024-06-01:
# Dashed
[image]

# Dashed Bold
[image]

# Dotted
[image]

# Dotted Bold
[image]

# Solid
[image]

# Solid Bold
[image]

tobiasfalk, 2024-06-01:
Solid Bold seams quite good

kvid, 2024-06-01:
> Solid Bold seams quite good

I agree, and it's also a good fit to https://github.com/wireviz/WireViz/issues/48#issuecomment-656802355, but I hope we also can get opinions from several others before deciding. We are not in hurry.

kvid, 2024-06-02:
Some details that have not been discussed in this issue yet are:
- About the YAML input syntax: I know that `internal_shorts` and `external_shorts` were suggested in #48, and the latter ended up as `loops`. Should we still keep `internal_shorts` for this new feature? Or is e.g. `shorts` precise enough? If _internal short_ is an established term, then we should keep it, I guess.
- Should unconnected pins (no external wire connected) with an internal short count as `populated`? I guess so, but please argue for use cases when that could be wrong.

tobiasfalk, 2024-06-02:
> - About the YAML input syntax: I know that `internal_shorts` and `external_shorts` were suggested in #48, and the latter ended up as `loops`. Should we still keep `internal_shorts` for this new [...]

I think the "internal" emphasis that this short/connection is inside the Connector housing and not a wire loop that may go outside and than back in again.
One thing that could be done are synonyms. For the `internal_shorts`, `shorts` and for `loops`, `external_shorts`.

> - Should unconnected pins (no external wire connected) with an internal short count as `populated`? I guess so, but please argue for use cases when that could be wrong.

I agree.

kvid, 2024-06-05:
@tobiasfalk wrote:
> [...]
> I think the "internal" emphasis that this short/connection is inside the Connector housing and not a wire loop that may go outside and than back in again. One thing that could be done are synonyms. For the `internal_shorts`, `shorts` and for `loops`, `external_shorts`.

It makes sense, but such synonyms deserves a separate issue (there might be more synonym candidates, e.g. in #331).

Are internal shorts always short wires inside the connector housing, or does it also exist connectors with optional shorts integrated?

When the shorts need wires, they also should be counted in the BOM, but then we need more information than their colors. See also https://github.com/wireviz/WireViz/pull/288#issuecomment-1559722149 where I suggest the same for `loops`. Maybe we need a common solution for shorts and loops to enable their BOM entries?

> 
> > * Should unconnected pins (no external wire connected) with an internal short count as `populated`? I guess so, but please argue for use cases when that could be wrong.
> 
> I agree.

See also the latest update of my question with a follow-up that I wrote before my browser detected your answer.

tobiasfalk, 2024-06-05:
@kvid yes I think we should most likely decide for either loops or shorts, since they are practically the same.
Personally I like this representation more, since it does not interveres with any wires and for me connected pins are something that is part of the connector and not of wires.

As for representation of options/description for a short, I would just add a row directly under the jumpers and in the colom of the jumper, text is added that describes the jumper.
The color was more of a thought to make a differentiation on the drawing easier and less to describe the collor of the actual jumper(physically).

olmari, 2024-06-05:
AFAIK sometimes there _are_ literal loops for connectors that can't have literal jumpers, so as concepts they're different... Both have place and should exists as general thing..

> I think the "internal" emphasis that this short/connection is inside the Connector housing and not a wire loop that may go outside and than back in again.

Yep!

> Should unconnected pins (no external wire connected) with an internal short count as populated? I guess so, but please argue for use cases when that could be wrong.

I'd "vote" for _not_ populated, as "shortcut jumpers" are specific to.. hmm.. how to describe it best... kind of self-explanatory already here described as internal to connection blocks, so no _wires_ are attached to them at any point alone... AFAIK even with my old job as building these things, shortcut blocks were never treated as "external" not "populated" in context of wiring connections...

tobiasfalk, 2024-06-05:
The problem with shorts in the BOM is that it highly depends on the connector, one connector can only be shorted with a with short wires that goes directly from one pin to the next(maybe even blanknwires), like good old DB9 or DB25 connector(most of them at least).
Than there are those where the manufacturer provides way of shorting.
And lastly there are Crimp connectors, where with most of them you need to do a wire loop and can not realy short pins "inside" the connector housing.

One way could be to provide it as a option for the Connector to say that part XY is as often needed as shorts. 
I also have no Problem with a more complicated short syntax where one can define with the short what is on the BOM, if it is a peace of wire or something else. And than wire vice counts these.

```
X1:
  shorts:
    -Sh1:
      pins:[1,5,7]
      color: PK
      manufacturerPn: 42A5Z42
      description: 3 pin Short over 5 pins, name of the part
    -Sh2:
      pins:[2,8,10]
      color:GR
      wireLength: 10mm
      description: some additional information for the BOM
```

olmari, 2024-06-05:
These are generally what I think of shorts in this context https://www.starelec.fi/product_info.php?products_id=32067, these yellow things... While there can be all sorts of things in random connector to make similar thing happen, the gist is that if it is "internal" like in my example, it should not count as populated, if it does occupy an place for wire to go, it should be populated... aka as loop does at the very moment...

Rest is semantics and can be described already in wireviz "language"...

kvid, 2024-06-05:
Deciding whether shorts should count as `populated` or not is not a philosophical question. The `populated` multiplier is e.g. used as the number of crimps or pins (at the mating side) to include as additional components, but maybe also as the number of heat shrink pieces (at the wire terminal side). The `unpopulated` multiplier is e.g. used as the number of seals for a watertight connector (probably at the mating side, but maybe there exist connectors with separate seals at the wire side too?) Maybe we need separate multipliers for the wire terminal side and the mating side, like I suggested above in https://github.com/wireviz/WireViz/issues/350#issuecomment-2143813935, so the user can select which multiplier that fits best with each use case?

olmari, 2024-06-05:
@kvid mm... I see what you do say... All I can say for my "stance" is that what I linked in previous message is the thing... That thing does not happen on _either_ side of where wires or (physical) connector mates into, it is specific _within_ the connector or connector bloc, neither side of wiring nor possible mating is in consideration...

olmari, 2024-06-05:
...Again I'm not specifically against if user can describe the internal short any way possible, would always be more flexible that way, just describing the thoughtprocess of why we generally even consider separate things for "loops" and "shorts" and their core meaning...

kvid, 2024-06-05:
@olmari wrote:
> @kvid mm... I see what you do say... All I can say for my "stance" is that what I linked in previous message is the thing... That thing does not happen on _either_ side of where wires or (physical) [...]

Maybe the terms _mating side_ doesn't fit very well with your connection block, but I still believe the functionality I suggest will fit your use case as well:
- `populated` = count of pins with wire connection, loop, or short (count at mating side of mating connectors).
- `connected` = count of pins with wire connection or loop (count at wire terminal side).

Your connection block doesn't have a mating side, only a wire terminal side, so using the `connected` multiplier should fit your use case, but please prove me wrong. WireViz is used for a huge number of different use cases with different types of connectors, so I try to use a generic terminology, but please suggest alternative terms.

tobiasfalk, 2024-06-05:
@kvid now with your explanation 'populated' makes even more sense for shorts, since they require a Pin in the connector and not a plug or something else.

olmari, 2024-06-05:
Mm... I don't have exact suggestion to exact terms to use. Nor I kind of even care in the ultimate picture. All I really care is the thought process of what I had for "short" or "jumper", and how it was described in this issue originally (or as I understood it in the premise of what I personally need).

@kvid that explanation or difference of populated vs connected _does_ make sense when you put it like that... :)

kvid, 2024-06-05:
My suggestion of `populated` and `connected` counting doesn't fit well with the term used for the `hide_disconnected_pins` option, so we might need a general discussion about terminology and create definitions to use consistently in our application. Then we probably also should include the #331 suggestions.

tobiasfalk, 2024-06-05:
This gives me the question, do we change things so that the user needs to rewrite its yaml file with a version updated?
Since I would propose to remove the loop parameter and replace it with this short displaying.

tobiasfalk, 2024-06-05:
Did some work on  #369 , where I changed the syntax to alowe more options for the shorts

tobiasfalk, 2024-06-08:
@kvid I did some work on #369, like giving it a better syntax and having short types, like the "internal" that is proposed here and loop what practically does what already exists (it does not replace not the loops) but with "more" fetures

tobiasfalk, 2024-06-09:
@kvid 
>The way you want to combine internal shorts and loops, might also need a discussion in https://github.com/wireviz/WireViz/issues/350 because I fear it will be too complicated for the simplest use cases.

I do not think so, since this is an option that can be set by the user and is entirely optional. The standard behavior is how I proposed here and if the used sets it to 'type: loop' than the short is not drawn like it is here but the same way the loops are.