wireviz/WireViz #351: Plugging/Keying a Pin/Way on a Connector

tobiasfalk, 2024-05-19:
One thing that is commonly done with Connectors is to plug some Pins/Ways, this is done to key on connector and prevent it from being plugged in/connected to somewhere it should not.
Here are some Ideas how it could be represented:
[image]
The following was added in the row of a pin/way
1. `<td>P</td>` a very simple way, could be confused very easily
2. `<td><B>P</B></td>` practically the same as 1. but bold and still has the same problem
3. `<td BGCOLOR="#000000"></td>` very nice to see and understand that this means that this Pin/Way is plugged
4.
5. `<td BGCOLOR="#000000"><FONT COLOR="#ffffff">P</FONT></td>` The white P may makes it more understandable that this pin/way is plugged
6. 
7. `<td BGCOLOR="#000000"><FONT COLOR="#ffffff"><B>P</B></FONT></td>` making the Text bold makes it more readable, my opinion at least
8. 
9. `<td BGCOLOR="#000000"><FONT COLOR="#ffffff"><B>X</B></FONT></td>` This is for the showcase of a different Character

My favorite is 3. but with giving the user the option to put Text or Characters in the field could be a good solution, in this case I would make the text bold to make it more readable.

tobiasfalk, 2024-05-19:
This shows longer Text, with XYZ the first 3 are Bold the later are not
[image]

kvid, 2024-05-19:
Nice!
However, you should also think about how this might work in combination with todays feature `bgcolor` and the future feature `fontcolor` (currently on hold in PR #255).

In addition, this might trigger a need for an extra `plugged` option in `ConnectorMultiplier` to enable adding the correct number of plugs into the BOM.

kvid, 2024-05-19:
If I understand you correctly, such a plugged pin can never be connected to a wire. In that case, it might also be an option to use the pin cell for such a plug visualization - both to save space in the diagram and to indicate no connection is possible.

tobiasfalk, 2024-05-19:
> If I understand you correctly, such a plugged pin can never be connected to a wire. In that case, it might also be an option to use the pin cell for such a plug visualization - both to save space in the diagram and to indicate no connection is possible.

I may not know of a situation where one would like to connect a wire to a plugged pin, but I can imaging some use case where those plugged pins are connected to something(like gnd or so).

To #255 one way could be to just reverse the Text color and Background color.