wireviz/WireViz #352: Proposal for new way to draw Wires in the Wire box

tobiasfalk, 2024-05-20:
Currently, the Wires inside the Wire box are done by coloring table rows, I would propose to change this and do those wires to edges (actual wires) that start at one side of the box and go to the other side.
This would mainly bring the benefit of removing the slide but existing miss alignment between the wires outside and inside the box. But also bring everything that can be done to the wires (edges) to the inside of the wire box.

Here is a comparison, look at the border of the box and where the wire goes into the box:
Current:
[image]
Proposal:
[image]

To see it better, one can download them and switch between them with an image viewer.
# Implementation
This uses the same method of straitening edges as the dashed lines in #350.

## Table
First, the three rows (each with a height of 2) in the table are replaced with one row (with a height of 6) that is completely white (not colored)
Current:
```
    <tr>
    <td><!-- 1_in --></td>
    <td>
     1:RD
    </td>
    <td>X2:1:+28VDC</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w1" height="6">
     <table cellspacing="0" cellborder="0" border="0">
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#ff0000" border="0"></td></tr>
      <tr><td colspan="3" cellpadding="0" height="2" bgcolor="#000000" border="0"></td></tr>
     </table>
    </td>
   </tr>
```
Proposal:
```   
   <tr>
    <td><!-- 1_in --></td>
    <td>
     1:RD
    </td>
    <td>X2:1:+28VDC</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" port="w1"  cellpadding="0" height="6"></td>
   </tr>
```

## Wire

Then the actual wire is drawn from the right (east) to the left (west) side, this needs to be done with two wire because the straitening method causes problems (see later).
Here the "border" of the wire is drawn first, and the colored inside is drawn on top of it:
Proposal:
```   
	edge [color="#000000" label=" "  penwidth=6]
	W1:w1:e -- W1:w1:w[straight=straight]
	edge [color="#ff0000" label=" "  penwidth=2]
	W1:w1:e -- W1:w1:w[straight=straight]
```

## Problem with the straitening method

[image]

or:

[image]

martinrieder, 2024-05-20:
I thought about this as well, though I wonder whether it really brings advantages over the current implementation. Graphviz treats nodes (cables and wires) both the same way. They will be laid out based on the edges (connections) that connect them. Introducing these additional edges would have a major impact on the ranking algorithm.

tobiasfalk, 2024-05-20:
> I thought about this as well, though I wonder whether it really brings advantages over the current implementation. Graphviz treats nodes (cables and wires) both the same way. They will be laid out [...]

I think especially when someone wants to really have twisted wires(#353) it would really be the only option, additionally I just can not unsee the small misalignment between the out and inside wire.

With `constraint=false` one can take those line out of the ranking(https://graphviz.org/docs/attrs/constraint/).

martinrieder, 2024-05-20:
> I think especially when someone wants to really have twisted wires(#353) it would really be the only option,

I can confirm that. See my proposal here 

> In simple cases, it might be sufficient to show the twisting outside and just hide the cable and connector boxes using a tweak. 
[image]

> Note the difference between styles `none` and `plain`, whereof the latter supports HTML, just without the surrounding `box` style.
> The trick here is to set `fixedsize` to `"true"` (as a string), which apparently sets the `width` attribute to its minimum value of 0.01.

_@martinrieder in https://github.com/wireviz/WireViz/issues/3#issuecomment-2109969132_

Thanks for pointing out the following, because I noticed that my approach would not work in every case. The `fixedsize` tweak would not shrink the nodes when there is some other node in parallel on the same rank. 

> With `constraint=false` one can take those line out of the ranking(https://graphviz.org/docs/attrs/constraint/).

Setting this on the node might be required as well. Would the need to be merged into a subgraph? Does this need to be a cluster then?

kvid, 2024-05-20:
@tobiasfalk wrote:
> Then the actual wire is drawn from the right (east) to the left (west) side, this needs to be done with two wire because the straitening method causes problems (see later).
Here the "border" of the wire is drawn first, and the colored inside is drawn on top of it:

Is there a reason drawing from east to west and not the opposite?

Could you also show the edge codes generating the problem images?

tobiasfalk, 2024-05-20:
> Is there a reason drawing from east to west and not the opposite?
Not realy, just did it this way also works the other way around.(Maybe I have read too many mangas lately)

> Could you also show the edge codes generating the problem images?

1:
```
	edge [color="#000000:#ff0000:#000000" label=" "]
	W1:w1:e -- W1:w1:w[straight=straight]
```
2:
```
	edge [color="#000000:#ff0000:#000000" label=" "  penwidth=6]
	W1:w1:e -- W1:w1:w[straight=straight]
```

This has to do with the fact that graphviz is drawing this as three lines that are offset to each others and that the "filtering" with the gvpr removes may line points in between the start and end

martinrieder, 2024-05-20:
> > Is there a reason drawing from east to west and not the opposite?
> Not realy, just did it this way also works the other way around.

This would only make difference for arrowheads and the ranking in a `digraph`.

@tobiasfalk about replacing the ugly table tricks, note that the Graphviz docs mention using `<HR>` instead of the rows in a table.

https://graphviz.org/doc/info/shapes.html#hr

tobiasfalk, 2024-05-20:
> This would only make difference for arrowheads and the ranking in a `digraph`.
>
>
> @tobiasfalk about replacing the ugly table tricks, note that the Graphviz docs mention using `<HR>` instead of the [...]

Ok, thx did not know about this, would also work but this would limit to a solid line, as far as I can see

kvid, 2024-05-21:
@tobiasfalk wrote:
> Here the "border" of the wire is drawn first, and the colored inside is drawn on top of it:
Proposal:
```dot
	edge [color="#000000" label=" "  penwidth=6]
	W1:w1:e -- W1:w1:w[straight=straight]
	edge [color="#ff0000" label=" "  penwidth=2]
	W1:w1:e -- W1:w1:w[straight=straight]
```

How would your suggestion work when a wire has more than two colors? WireViz doesn't limit the number of colors specified for a wire. It does, however, treat a two-colored one as if it is three-colored with primary:secondary:primary colors as a special case for a prettier result.

Try making a 60-wire cable with `color_code: DIN` to see three-colored examples.

tobiasfalk, 2024-06-16:
Multicolor cables are now solved.
Thanks to the fokes at the forum, https://forum.graphviz.org/t/straitening-one-line-throu-a-table/2196/19
I think @ferdnyc. thx

tobiasfalk, 2024-06-16:
See latest comit to #369