wireviz/WireViz #353: Proposal for displaying of twisted Pairs

tobiasfalk, 2024-05-20:
Twisted pairs is a much talked request and even has the issue number #3.
My Proposal for displaying them for now look like this:
[image]

This may not have some fancy twisting of the wires inside the box, but it is clearly distinguishable from not twisted wires (in my opinion even better the really "twisting" the wires, since doing so makes it harder to see what text goes to what wire) and could be used as basis for other types of combining wires with each other.

If #352 is implemented, it would maybe even allow someone to find some kind of "graph viz magic" to twist the wires.

# Implementation
This uses the same method of straitening edges as the dashed lines in #350.

The wires are rearranged so that the wires are not spited by their corresponding text, and a Row is added above and below of the wire pair. The Row Above gets the text `Twisted Pair` or `Twisted Pair: <twist rate><twist rateunit>`. Then two edges are drawn from the right (east) to the left (west) side of the box to visual split the wire par of the rest, these are always attached to the top (north) of the added rows.

Table:
```
<tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t1t"  height="6">
    Twisted Pair
    </td>
    </tr>
   <tr>
    <td><!-- 1_in --></td>
    <td>
     1:RD
    </td>
    <td>X2:1:+28VDC</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" port="w1"  cellpadding="0" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6"></td>
   </tr>
   <tr>
    <td><!-- 2_in --></td>
    <td>
     2:BK
    </td>
    <td>X2:2:+28VDC RTN</td>
   </tr>
   <tr>
     <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t1b"  height="6"></td>
   </tr>
```
 Dashed lines:
```
    // Twisted pairs border
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t1t:ne -- W1:t1t:nw[straight=true]
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t1b:ne -- W1:t1b:nw[straight=true]
```

tobiasfalk, 2024-05-20:
The whole file:
```
graph {
// Graph generated by WireViz 0.3.2
// https://github.com/formatc1702/WireViz
	graph [bgcolor="#FFFFFF" fontname=arial nodesep=0.33 rankdir=LR ranksep=2  dpi="600"]
	node [fillcolor="#FFFFFF" fontname=arial height=0 margin=0 shape=none style=filled width=0]
	edge [fontname=arial style=bold label=" "]
	X2 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">X2</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">P1</td>
   <td balign="left">9-pin</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1">
   <tr>
    <td port="p1l">1</td>
    <td>+28VDC</td>
   </tr>
   <tr>
    <td port="p2l">2</td>
    <td>+28VDC RTN</td>
   </tr>
   <tr>
    <td port="p3l">3</td>
    <td>Token-In REF</td>
   </tr>
   <tr>
    <td port="p4l">4</td>
    <td>Token-In</td>
   </tr>
   <tr>
    <td port="p5l">5</td>
   </tr>
   <tr>
    <td port="p6l">6</td>
    <td>NC</td>
   </tr>
   <tr>
    <td port="p7l">7</td>
    <td>RS-485+</td>
   </tr>
   <tr>
    <td port="p8l">8</td>
    <td>RS-485-</td>
   </tr>
   <tr>
    <td port="p9l">9</td>
    <td>RS-485 Shield</td>
   </tr>
  </table>
 </td></tr>
</table>
> fillcolor="#FFFFFF" shape=box style=filled]
	edge [color="#000000:#ff0000:#000000" label=" "]
	W1:w1:e -- X2:p1l:w
	edge [color="#000000:#000000:#000000" label=" "]
	W1:w2:e -- X2:p2l:w
	edge [color="#000000:#f5f0d0:#000000" label=" "]
	W1:w3:e -- X2:p7l:w
	edge [color="#000000:#0066ff:#000000" label=" "]
	W1:w4:e -- X2:p8l:w
	edge [color="#000000:#00ff00:#000000" label=" "]
	W1:w5:e -- X2:p9l:w

    
	edge [color="#000000" label=" "  penwidth=6]
	W1:w1:e -- W1:w1:w[straight=true]
	edge [color="#ff0000" label=" "  penwidth=2]
	W1:w1:e -- W1:w1:w[straight=true]
    
	edge [color="#000000" label=" "  penwidth=6]
	W1:w2:e -- W1:w2:w[straight=true]
	edge [color="#000000" label=" "  penwidth=2]
	W1:w2:e -- W1:w2:w[straight=true]
    
	edge [color="#000000" label=" "  penwidth=6]
	W1:w3:e -- W1:w3:w[straight=true]
	edge [color="#f5f0d0" label=" "  penwidth=2]
	W1:w3:e -- W1:w3:w[straight=true]
    
	edge [color="#000000" label=" "  penwidth=6]
	W1:w4:e -- W1:w4:w[straight=true]
	edge [color="#0066ff" label=" "  penwidth=2]
	W1:w4:e -- W1:w4:w[straight=true]
    
	edge [color="#000000" label=" "  penwidth=6]
	W1:w5:e -- W1:w5:w[straight=true]
	edge [color="#00ff00" label=" "  penwidth=2]
	W1:w5:e -- W1:w5:w[straight=true]

    // Twisted pairs border
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t1t:ne -- W1:t1t:nw[straight=true]
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t1b:ne -- W1:t1b:nw[straight=true]
    
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t2t:ne -- W1:t2t:nw[straight=true]
    edge [color="#000000" headclip=false, tailclip=false, style="dashed", constraint=false]
    W1:t2b:ne -- W1:t2b:nw[straight=true]

	W1 [label=<
<table border="0" cellspacing="0" cellpadding="0">
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">W1</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellpadding="3" cellborder="1"><tr>
   <td balign="left">5x</td>
  </tr></table>
 </td></tr>
 <tr><td>
  <table border="0" cellspacing="0" cellborder="0">
   <tr><td>&nbsp;</td></tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t1t"  height="6">
    Twisted Pair
    </td>
    </tr>
   <tr>
    <td><!-- 1_in --></td>
    <td>
     1:RD
    </td>
    <td>X2:1:+28VDC</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" port="w1"  cellpadding="0" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w2" height="6"></td>
   </tr>
   <tr>
    <td><!-- 2_in --></td>
    <td>
     2:BK
    </td>
    <td>X2:2:+28VDC RTN</td>
   </tr>
   <tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t1b"  height="6"></td></tr>

   <tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t2t"  height="6">Twisted Pair: 5 1/m</td></tr>
   <tr>
    <td><!-- 3_in --></td>
    <td>
     3:IV
    </td>
    <td>X2:7:RS-485+</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w3" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" height="6"></td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w4" height="6"></td>
   </tr>
   <tr>
    <td><!-- 4_in --></td>
    <td>
     4:BU
    </td>
    <td>X2:8:RS-485-</td>
   </tr>
   <tr><td colspan="3" border="0" cellspacing="0" cellpadding="0" port="t2b"  height="6"></td></tr>
   <tr>
    <td><!-- 5_in --></td>
    <td>
     5:GN
    </td>
    <td>X2:9:RS-485 Shield</td>
   </tr>
   <tr>
    <td colspan="3" border="0" cellspacing="0" cellpadding="0" port="w5" height="6"></td>
   </tr>
   <tr><td>&nbsp;</td></tr>
  </table>
 </td></tr>
</table>
> fillcolor="#FFFFFF" shape=box style=filled]
}
```

martinrieder, 2024-05-20:
I agree to the point that WireViz should have an option to handle twisted pair wires I have to a specific way. As I had stated in an earlier comment:

> I think that putting a comment beneath the wire description is by far more explicit than any layout trickery:
[image]

_ @martinrieder in https://github.com/wireviz/WireViz/issues/3#issuecomment-2107665428_

The first thing to do discuss is how this would be implemented in the WireViz syntax. How about introducing a `category: twisted ` for cables?

tobiasfalk, 2024-05-20:
Yes, I agree this is more of a combination of what is/was already proposed, but I tried to compile it and give a way of how to implement it in graphviz.

As for syntax:
```
cables:
  W1:
    colors: [RD, BK, IV, BU, GN]
    twisted: [[1,2],[3,4]
```

kvid, 2024-05-20:
Thank's for this nicely presented contribution. It's very good that someone uses ideas from @SnowMB in #3 and taking them further.

Be aware that it might be a challenge to combine this with [Tutorial 08](https://github.com/wireviz/WireViz/tree/master/tutorial#08---part-numbers-and-additional-components), but such a combination is perhaps not much used in real life?

The dashed lines surrounding each pair might easily be interpreted as a shield around each of these pairs, which is another requested feature discussed in #330, but it should be possible to use each of these two features separately.

The suggested `category: twisted`  for cables makes it hard to mix twistes pairs and untwisted wires in the same cable. How about something like:
```yaml
cables:
  W1:
    colors: [RD, BK, IV, BU, GN]
    twisted:
      - [RD, BK]
      - [IV, BU]
```

kvid, 2024-05-20:
@tobiasfalk - my browser wasn't updated with your latest comment while I was writing mine, but I now see we have the same ideas about a possible syntax.

tobiasfalk, 2024-05-20:
> The dashed lines surrounding each pair might easily be interpreted as a shield around each of these pairs, which is another requested feature discussed in #330, but it should be possible to use each of these two features separately.

A small thin line or something else could also work, the dashed line was just my first idea.

## ´style="dotted"`:
[image]

## ´style="solid"`(could be thiner):
[image]

## ´style="invis"` aka no line (not very clear what is meant):
[image]

martinrieder, 2024-05-20:
Please also consider coaxial cables, for example inside a VGA monior cable. It may be better to simply call them pairs.

Thanks for pointing out Tutorial08. I have not practically used it, though I might give it a try. My common use case requires wires of different gauges combined. I prefer to split those up into separate bundles, in order to have some kind of grouping. This idea has also recently been suggested.

Maybe there is not so much difference in having separate groups for (twisted) pairs or any other number of wires that are somehow related.

tobiasfalk, 2024-05-20:
About Tutorial08, I think just writing the two text lines directly above each other and keeping the wires together could be a good solution.

> Please also consider coaxial cables, for example inside a VGA monior cable. It may be better to simply call them pairs.
> 
> Thanks for pointing out Tutorial08. I have not practically used it, though I might give it a try. My common use case requires wires of different gauges combined. I prefer to split those up into separate bundles, in order to have some kind of grouping. This idea has also recently been suggested in #268 and #325.
> 
> Maybe there is not so much difference in having separate groups for (twisted) pairs or any other number of wires that are somehow related.

I think of this more of a "simple" twist (like it is with Ethernet cables, yes they may have a shield per pair, or the old telephone cables), since what I can see the VGA cable for example has a bit more to it than just some twisted pairs, the same is with coax cable. 

So maybe a changing this to wire pairing of the type "simple" twist.

Syntax ?:
```
twisted:
      type: simple twist
      - [RD, BK]
      - [IV, BU]
```

martinrieder, 2024-05-20:
@tobiasfalk please also have a look at the many ideas suggested in #56.

tobiasfalk, 2024-05-20:
> @tobiasfalk please also have a look at the many ideas suggested in #56.

I have not read it completely and only view over it a bit, but this goes pretty fast in to "let's overhaul the whole Wire/Cable system" which could be a good idea or not but is a bit much for this.

I would say that the current system does not really Cable's as something that exists and is more of single wires connected to connector or each other. What I wanted to say earlier is that wires can be pared to each other (and then for example twisted or otherwise grouped) and this Issue only treats one special case of pairing wires, but based on this more cases could be added later.

And meant:
```
paired:
      type: simple twist
      - [RD, BK]
      - [IV, BU]
```

kvid, 2024-05-20:
@tobiasfalk wrote:
> About Tutorial08, I think just writing the two text lines directly above each other and keeping the wires together could be a good solution.

That's  worth trying.

> [...]
> So maybe a changing this to wire pairing of the type "simple" twist.
> 
> Syntax ?:
> 
> ```
> twisted:
>       type: simple twist
>       - [RD, BK]
>       - [IV, BU]
> ```

I'm not sure what such a `type` attribute is good for. What would be different with a `simple twist` value compared to not specifying it, and what other values might be possible? Or are you thinking this could be the title of each pair, like "Twisted Pair" in your examples above?

Another thing is that this is not valid YAML. The same attribute cannot have both `dict` entries and `list` entries at the same time.

tobiasfalk, 2024-05-20:
These pairs than can be part of Cable's, that can not only consist of Wires and Wire pairs but also pull cord and so on(maybe even air pipes).

tobiasfalk, 2024-05-20:
> I'm not sure what such a `type` attribute is good for. What would be different with a `simple twist` value compared to not specifying it, and what other values might be possible? Or are you thinking this could be the title of each pair, like "Twisted Pair" in your examples above?

Another type could be for example be a shielded twisted pair(CAT 5 or 6 and above) or one could consider a coax or triax cable as paired wires. All these would require a different representation

> Another thing is that this is not valid YAML. The same attribute cannot have both `dict` entries and `list` entries at the same time.

I do not realy know YAML, sorry