wireviz/WireViz #354: [Feature Request] Adding PDF as output

tobiasfalk, 2024-05-20:
We may already have SVG as a vector graphics output, but it still would be nice if PDF would also be available.

martinrieder, 2024-05-20:
This has been in development for some time. Check #32 and the associated feature branch.
The new CLI offers a switch to request PDF output format, but it is not implemented yet.

tobiasfalk, 2024-05-20:
I took a look at #32 and #304, and I am a bit confused, is it now possible to draw the wiring plan in to a drawing frame with title block or not, if yes how, since there is no mention I found in the doc or examples.
If the whole drawing frame with title block inclusion is not stated or though out I would recommend using an already existing template format, like the FreeCAD SVG, this would bring the benefit that there are already some templates existing and that in its base form it is just an SVG that can be rendered normally. 
In that the fields that are meant to be edited have the XML attribute `freecad:editable` with a variable name and some like the title or the author/creator have the additional attribute `freecad:autofill` with a specified value in it to identified what it is ([FreeCAD #13005](https://github.com/FreeCAD/FreeCAD/pull/13005) and [#13790](https://github.com/FreeCAD/FreeCAD/pull/13790)). 
With these ids the text that is in them can be changed before handing the template/svg over to the drawing engine to use.

When it comes to coding I would polity decline, since I not only do not really have too much experience with Python and especially not with graphvis (even when the issues I opened in the last view day would sugset otherwise I never used graphviz before that).
What I am happily do is help with question where I can and with ideas on improving things.

kvid, 2024-05-20:
Take a look at [demo02](https://github.com/wireviz/WireViz/blob/master/examples/demo02.yml)

tobiasfalk, 2024-05-21:
> @tobiasfalk there are many ways that the frame and title blocks could be drawn. What do you suggest? Only add it for "printable" formats like PDF?

I would add it to all Formats, since the user may want to use it in Programs like word where imprting a PDF as a page is not possible(theoretically yes but I have tried for weeks and was not able to do it).
So giving the user the ability to export this as a PNG with tilteblock is defenetly a good idea, has also the benefit of being able to use it in issues and not rely on the reader to open a link to a file or so.

I do not have the ability to do something for a while(not at home), but could one of you try to import one SVG from the FreeCAD templates as image in the background?

martinrieder, 2024-06-05:
> I do not have the ability to do something for a while(not at home), but could one of you try to import one SVG from the FreeCAD templates as image in the background?

Merging two SVG files seems like a simple task. Converting this into PDF can be solved with other tools.

WireViz is capable of drawing the BOM table into a dedicated node. It would thereby also assure that this does not overlap with the graph. Positioning and scaling might be an issue though. 

Do you know if it is possible to predefine a container or an anchor in some SVG template files that could be used for merging?

tobiasfalk, 2024-06-05:
https://forum.graphviz.org/t/background-image-frame-for-the-graph/2232

tpimh, 2025-07-09:
As a workaround, I tried to use weasyprint to generate PDFs from HTML.

Using "WireViz Demo 2" diagram, I generated HTML, and replaced `size: auto;` with `size: A3 landscape;` in it. The output was still not perfect, but good enough: [demo02.pdf](https://github.com/user-attachments/files/21137448/demo02.pdf).

After removing the tables at the bottom, it was ready for printing.

glt248, 2025-10-07:
Use Firefox export nice PDF from HTML

[Demo02.pdf](https://github.com/user-attachments/files/22737914/Demo02.pdf)