wireviz/WireViz #355: [bug?] ValueError: X is not in list

the-hess, 2024-05-21:
First of all thanks everyone who's worked on this, it's a great tool.

I'm having an issue when using arrows and simple connectors.

At the moment I'm trying to do a large wiring diagram for a full system and I'm getting an error when pointing a simple connector to a normal connector with an arrow. It works when I point the connector to pin 1, but not pin 2 or higher.

In my yaml file, the offending entry is on line 149 - if you change the [2] to a [1] the diagram renders as below, but I want it to point to contact 2 of connector 07-0613.

I'm currently using version the 0.4 and have (manually) implemented the basic fixes in #346 and #347.

Graphviz version 9.0.0 (20230911.1827)
Python 3.12.2

[image]

```
options:
  mini_bom_mode: true
  
connectors:
#Items
 07-0601:
    pinlabels: [J14,J15,J28,J4,J10,J12,J5,J7,J1,J6,J8,J2,J13,J16,J11,J9]
    show_pincount: false
    #bgcolor: OL
    
 07-0620:
    pinlabels: [9 Way, 25 Way, 37 Way, 4 Way XLR]
    show_pincount: false

 07-0518:
    pins: [L,N,E,+V,-V]
    show_pincount: false
 
 07-0624: 
    pinlabels: [1]
    show_pincount: false
    
 07-0613:
    pins: [1, 2, 3]
    pinlabels: [J1, J2, J3]
    show_pincount: false
#End of wire points 
 F:
    style: simple
    type: bootlace ferrule
    
 P:
    style: simple
    type: Molex
    
 E:
    style: simple
    type: E-Stop
    
 X:
    style: simple
    type: XLR
    
 D:
    style: simple
    type: D-SUB
    
cables:
 07-0608:
    show_wirecount: false
    colors: [BU,BUWH]
 
 07-0607:
    show_wirecount: false
    colors: [RD,WH,RDWH]
    
 07-0603:
    colors: [RD,BK]
    show_wirecount: false
    
 07-0606:
    colors: [VT,VTWH,VTWH,GN,GNWH,GNWH,OG,OGWH,OGWH]
    show_wirecount: false

 07-0604:
    colors: [GY]
    show_wirecount: false
    
 07-0609:
    colors: [GY]
    show_wirecount: false
    
 07-0610:
    colors: [GY]
    show_wirecount: false
    
 07-0611:
    colors: [GY]
    show_wirecount: false
    
 07-0612:
    colors: [GY]
    show_wirecount: false
    
 07-0614:
    colors: [VTWH]
    show_wirecount: false
    
connections:
 -
    - 07-0601: [2,3]
    - <--
    - P.
    - 07-0608: [1,2]
    - 07-0620: [9 Way,9 Way]
 -
    - 07-0601: [4-6]
    - <--
    - P.
    - 07-0607: [1-3]
    - 07-0620: [2,2,2]
    
 -
    - 07-0518: [+V,-V]
    - -->
    - F.
    - 07-0603: [1,2]
    - P
    - -->
    - 07-0601: [1,1]
    
 -
    - 07-0601: [7-15]
    - <--
    - P.
    - 07-0606: [1-9]
    - 07-0620: [3,3,3,3,3,3,3,3,3]
    
 -
    - 07-0601: [16]
    - <--
    - P.
    - 07-0604: [1]
    - 07-0620: [4]
    
 -
    - 07-0620: [4]
    - <--
    - X.
    - 07-0609: [1]
    - E.
    
 -
    - 07-0620: [2]
    - <--
    - D.
    - 07-0610: [1]
    - D.
    - -->
    - 07-0613: [1]
    
 -
    - 07-0620: [3]
    - <--
    - D.
    - 07-0611: [1]
    - D.
    - -->
    - 07-0613: [2]

 -
    - 07-0620: [1]
    - <--
    - D.
    - 07-0612: [1]
    - D.
    - -->
    - 07-0624: [1]
    
 -
    - 07-0613: [3]
    - <--
    - P.
    - 07-0614: [1]
    - P.
```

martinrieder, 2024-05-21:
@the-hess Check your connector definitions. You mixed up pins and pinlabels. The first defines the designators that are used for the connection. Specifying pinlabels will cause numeric pins to be generated.

the-hess, 2024-05-21:
Yeah I had this issue a couple of times, hence the switching. The issue is I can't refer to the pinlabels either. I then just get J2 is not in list.

martinrieder, 2024-05-21:
Sounds a bit like #336, but this one seems to be caused by the autogeneration. By coincidence, also #356 reports a similar issue. They might be related.

I noticed another issue with the P connector beneath 07-0603, which is missing the "."

the-hess, 2024-05-21:
> I noticed another issue with the P connector beneath 07-0603, which is missing the "."

That was deliberate as I want both those wires into 1 connector.

I've tried various combinations of pins, pinlabels, just using numbers (without any names for pins). I can't get the issue to go away fully though it does shift around exactly what connector/signal it flags for.

kvid, 2024-06-01:
@the-hess - You should now try my PR #358 intended to fix this issue, and please report back here what you find. I fixed the initial issue pretty quickly, but found also another bug in the same code section, and now that also should be fixed.

kvid, 2024-07-06:
The relevant fixes are now merged into the `release/v0.4.1-rc` branch of PR #365 that is not yet released, but also describes how you can install the latest collection of fixes before the release.

callumroest, 2024-07-09:
The fix in PR #365 resolved this issue for me.

kvid, 2024-07-13:
Closed after releasing v0.4.1.

closed 2024-07-13 by kvid