wireviz/WireViz #366: Not installing

RobinMaub, 2024-05-28:
Hello all,

I have installed Graphviz and now try to install wireviz but when I type "pip3 install wireviz" it opens the script but does not launch it on Windows 10. Is there any issue in the current pypi package ?

Thanks in advance

kvid, 2024-05-28:
@RobinMaub - Thank's for reporting this unexpected behaviour. Can you provide a dump from your command shell screen, showing commands you try and their output? Have your tried installing it in a virtual environment? Please try something like this:
```bat
C:\>mkdir \Repos\test2

C:\>cd \Repos\test2

C:\Repos\test2>python3 -m venv .venv

C:\Repos\test2>call .venv/Scripts/activate.bat

(.venv) C:\Repos\test2>pip install wireviz
Collecting wireviz
  Using cached wireviz-0.4-py3-none-any.whl (48 kB)
Collecting pillow
  Using cached pillow-10.3.0-cp310-cp310-win_amd64.whl (2.5 MB)
Collecting click
  Using cached click-8.1.7-py3-none-any.whl (97 kB)
Collecting pyyaml
  Using cached PyYAML-6.0.1-cp310-cp310-win_amd64.whl (145 kB)
Collecting graphviz
  Using cached graphviz-0.20.3-py3-none-any.whl (47 kB)
Collecting colorama
  Using cached colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Installing collected packages: pyyaml, pillow, graphviz, colorama, click, wireviz
Successfully installed click-8.1.7 colorama-0.4.6 graphviz-0.20.3 pillow-10.3.0 pyyaml-6.0.1 wireviz-0.4

[notice] A new release of pip is available: 23.0.1 -> 24.0
[notice] To update, run: python.exe -m pip install --upgrade pip

(.venv) C:\Repos\test2>wireviz -V

WireViz 0.4

(.venv) C:\Repos\test2>where wireviz
C:\Repos\test2\.venv\Scripts\wireviz.exe

(.venv) C:\Repos\test2>dot -V
dot - graphviz version 2.38.0 (20140413.2041)

(.venv) C:\Repos\test2>python -V
Python 3.10.11

(.venv) C:\Repos\test2>pip freeze
click==8.1.7
colorama==0.4.6
graphviz==0.20.3
pillow==10.3.0
PyYAML==6.0.1
wireviz==0.4

(.venv) C:\Repos\test2>pip -V
pip 23.0.1 from C:\Repos\test2\.venv\lib\site-packages\pip (python 3.10)

(.venv) C:\Repos\test2>pip3 -V
pip 23.0.1 from C:\Repos\test2\.venv\lib\site-packages\pip (python 3.10)

(.venv) C:\Repos\test2>ver

Microsoft Windows [Version 10.0.19045.3803]

(.venv) C:\Repos\test2>
```

amotl, 2024-06-04:
Dear Robin,

thank you for writing in.

> when I type "pip3 install wireviz" it opens the script but does not launch it on Windows 10

Can you further clarify what this means? For example, can you share any error message with us, or any command line invocations which lead to this problem?

With kind regards,
Andreas.

ThomasDus, 2024-12-01:
any chance to install as a beginner?

kvid, 2024-12-22:
@ThomasDus wrote:
> any chance to install as a beginner?

Have you tried the recommended instructions? https://github.com/wireviz/WireViz/blob/master/docs/README.md#installation