wireviz/WireViz #370: Splice Naming + Cable Hump Query

akikinho, 2024-05-30:
Hello Wireviz team,
I am using Wireviz in a project and I am facing two issues. 
- How do I display the splice name?
- For electrical cables crossing, I would like to show the hump to avoid confusion between the cables. How do I do that?

Any help will be greatly appreciated!

Thank you and Best Regards!

martinrieder, 2024-05-31:
@akikinho please provide a few more details, so we can clearly understand your request, i.e. which version you are using. Also make sure to check [CONTRIBUTING. md](https://github.com/wireviz/WireViz/blob/master/docs%2FCONTRIBUTING.md)

- #270 has a discussion about splices
- #174 might help with wire crossings

I'm afraid that displaying a "hump" won't be easy to implement due to the nature of the underlying Graphviz tool.

kvid, 2024-05-31:
@akikinho wrote:
> [...]
> * How do I display the splice name?

In addition to the details @martinrieder asks you about, it will also help providing an example YAML input to better understanding in what context you want to "display the splice name". However, maybe what you seek can be obtained with `show_name: true` for the splice connectors.

> * For electrical cables crossing, I would like to show the hump to avoid confusion between the cables. How do I do that?

If I understand this request correctly, I agree with the previous answer, that it might not be easy to implement. Embedding or linking to an image showing the visual appearance you request will always help - I assume you request what is named "Older convention" at [this page](https://www.allaboutcircuits.com/textbook/reference/chpt-9/wires-and-connections/).

akikinho, 2024-05-31:
Hello @kvid @martinrieder, thank you so much for the help. @kvid yes for the hump, it is exactly what I want for the electrical connectors. Is there a possibility to do it?

martinrieder, 2024-05-31:
@akikinho Anything might be possible, so it is merely a question of how much effort it takes. The "new convention" is much easier to achieve using "virtual/hidden" splices. Since WireViz leaves the layouting to the Graphviz "dot" engine, it cannot predict which wires would cross.
How many crossings does your schematic have? Can you tell in advance which wires should cross? Would it be acceptable to do some manual postprocessing?

akikinho, 2024-05-31:
@martinrieder , here is a screenshot of the crossings. 
[image]

martinrieder, 2024-05-31:
Okay that is quite a challenge. It arises from the fact that we use splines instead of straight lines for the wires. Now you actually have 3 wires crossing and a 4th one nearby. What do you expect that the output should look like?
By manually postprocessing, I meant editing the SVG output in Inkscape.

akikinho, 2024-05-31:
I understand @martinrieder. Thank you so much !

closed 2024-07-09 by kvid