wireviz/WireViz #373: [meta] Logo Suggestions

martinrieder, 2024-06-02:
Any software deserves to have logo, so I am starting this issue for sharing and discussing any suggestions. With Daniel Rojas @formatc1702 being the originator of this great tool, he deserves the right to make the final choice. We could still have a community vote (before the next major release?) to come up with the best candidates.

kvid, 2024-06-02:
I suggest using e.g. SVG to enable high-quality scaling when needed. Maybe see how these projects store logos:
- https://github.com/simple-icons/simple-icons
- https://github.com/gilbarbara/logos

martinrieder, 2024-06-02:
I agree that there are some good reasons for choosing SVG, though I do not see this as a limiting factor by itself. It only implies some limitations regarding simplicity and coloring though... I feel confident that I can vectorize them using Inkscape. Contributions in raster images are also welcome, yet even pen-and-paper format if anyone here feels creative! 

I must admit that I left the creativity part to an AI, which came up with some interesting suggestions. It seems to be legasthenic though, so some spelling correction must be done... 

AFAIK there is no LLM that was trained on vector images, so I used DALL-E3. Let me know if you want to see the prompts for these. Some are minimalist and easy to vectorize, but others a bit more fancy and colorful. Please cast your vote and I will try my best to vectorize.

martinrieder, 2024-06-02:
One
[image]

martinrieder, 2024-06-02:
Two
[image]

martinrieder, 2024-06-02:
Three
[image]

martinrieder, 2024-06-02:
Four
[image]

martinrieder, 2024-06-02:
Five
[image]

martinrieder, 2024-06-02:
Six
[image]

martinrieder, 2024-06-02:
Seven
[image]

martinrieder, 2024-06-02:
Eight
[image]

martinrieder, 2024-06-02:
Nine
[image]

martinrieder, 2024-06-02:
Eleven
[image]

martinrieder, 2024-06-02:
Twelve
[image]

tobiasfalk, 2024-06-02:
They all look good  but I think that most of them are tomuch, crowded, complicated.
Something simpler would be much better, nine, goes in to the right direction.

Consider how they work in different sizes, like a github profile picture or a desktop icon.

kvid, 2024-06-02:
I agree with most arguments by @tobiasfalk that a simple logo is better. It should IMHO be easily recognizable also when shown as small icons, e.g. in Windows. Of the 12 suggestions, I've seen so far, nine seems better, but it might be too similar to the USB logo.

@tobiasfalk  wrote:
> [...] Should the logo have a comandline/terminal representation and so on.

Are you thinking about an ASCII art version of a graphical logo?

tobiasfalk, 2024-06-02:
> @tobiasfalk  wrote:
> > [...] Should the logo have a comandline/terminal representation and so on.
> 
> Are you thinking about an ASCII art version of a graphical logo?
> 

Yes

martinrieder, 2024-06-02:
Thanks for the feedback. I understand your concerns. Please see the images above as an inspiration. None of them are meant to be final. We could also merge certain aspects into a combined version.

There's another one I left out because I thought that it might be suited as an icon for the YAML files. 
[image]

That brings up the question: does the logo need to include the WireViz text?

kvid, 2024-06-02:
@martinrieder wrote:
> [...] 
> That brings up the question: does the logo need to include the WireViz text?

IMHO - no, because that is unrealistic to recognize for small icon sizes. The name is too long.

tobiasfalk, 2024-06-02:
@martinrieder this also looks like a good start, as said a more simpler Version could be more effective in recognition and so, especially when small.

As for the the WireViz in the logo, I would say maybe not  the whole name but a W and a V would be a good idea.

These two arrangements come to my minde:

V
W

And

W
V

tobiasfalk, 2024-06-02:
[image]

[image]

tobiasfalk, 2024-06-02:
https://youtu.be/YPoLD_5FQ38?si=7bf4MCppmUOhpJVl
Sorry😁🤦‍♂️

kvid, 2024-06-02:
If you search for WV logo, you'll find West Virginia and a few others that we should be careful about getting too similar to.

tobiasfalk, 2024-06-02:
WZ could also work, since the Z is more prominent when saying Viz

martinrieder, 2024-06-02:
@kvid wrote:
> I suggest using e.g. SVG to enable high-quality scaling when needed. Maybe see how these projects store logos:
> - https://github.com/simple-icons/simple-icons
> - https://github.com/gilbarbara/logos

I am well aware of the trend to have minimalist logos, optionally spelling out the name separately. Knowing what stands behind them is often hard and comes close to learning Egyptian hieroglyphics. In that respect, I would say that we should not incorporate any letters into the logos. It needs to show some connectors and wires though.

tobiasfalk, 2024-06-02:
What about a Plan where a wire comes out of?

martinrieder, 2024-06-02:
> What about a Plan where a wire comes out of?

[image]

martinrieder, 2024-06-02:
> > What about a Plan where a wire comes out of?
> 
> [image]
> 
> 

This one I had already saved. Any ideas for simplification?

tobiasfalk, 2024-06-02:
Maybe only one paiper(or half of it) and one wire with connector.
[image]

martinrieder, 2024-06-02:
> Maybe only one paiper(or half of it) and one wire with connector.

IMHO a wiring harness would always contain multiple wires and connectors... 

Thinking opposite, we could also show a connector _going into_ the schematics sheet.

tobiasfalk, 2024-06-02:
> Thinking opposite, we could also show a connector _going into_ the schematics sheet. 

Yes sounds good, but isn't WireViz used to design harneses before creating them, and this may look the other way around.

Since designing is often connected with pens, why not a pen or pencil and wires somewhere/how.

martinrieder, 2024-06-02:
> Yes sounds good, but isn't WireViz used to design harneses before creating them, and this may look the other way around.
>
> Since designing is often connected with pens, why not a pen or pencil and [...]

Funny, because that is one thing that I also got as a suggestion. The prompt was: "Create a logo with a minimalistic style in rectangular shape, featuring connectors in a wiring harness surrounded by a frame."

[image]

martinrieder, 2024-06-02:
Here's a connector going outwards:
[image]

tobiasfalk, 2024-06-02:
> Here's a connector going outwards:
> [image]
> 
> 

Maybe just the circle

martinrieder, 2024-06-02:
[image]

Is this one is still too crowded?

tobiasfalk, 2024-06-02:
And more wires than traces

martinrieder, 2024-06-02:
[image]

My last one for today. Just some wires and really simple connectors.

martinrieder, 2024-06-02:
> And more wires than traces 

I can remove the circles resembling vias in a PCB trace of course.

obelisk79, 2024-06-02:
Greetings, I recently led the effort to replace the logo for FreeCAD. As part of that I spend some time studying logo/design theory. I would note that a logo should be visually simple, unique and easily identifiable from other products. Using letters such as WV aren't really that important as long as the visuals are clean, scale properly and don't rely on color to be effective.

It seems like there's a lot of AI generated images in this discussion. I would caution against this approach. However, of the images shown, No. 9 seems to be the closest to a decent concept.

martinrieder, 2024-06-02:
@obelisk79 thanks for your feedback. I noticed that www.freecad.org shows a different logo than wiki.freecad.org. Is there a reason for this?  Which one do you prefer?

The AI content is only intended as a source of inspiration. None of them are meant to be the final logo. Some editing will be required anyways.

tobiasfalk, 2024-06-02:
> I noticed that www.freecad.org shows a different logo than wiki.freecad.org. Is there a reason for this?
This has most likely to do with the fact that the new Logo was finalised just some weeks ago and that not all is moved.

obelisk79, 2024-06-02:
This is the correct new logo for FreeCAD. The migration has not been instant but most instances have been updated:
[image]

obelisk79, 2024-06-03:
I did a little doodling based on image #9 
![image](https://github.com/wireviz/WireViz/assets/17993710/c97b6fd4-c79f-4c0c
[image]
-a2ea-7418114f0b63)

martinrieder, 2024-06-03:
@obelisk79 wrote:
> I did a little doodling based on image #9 

Thanks for that idea. It looks nice, though I think that it is missing some important characteristics of a wiring harness:

- Diversity of connectors
- Spliced connections
- Bundling of wires

Therefore, I have created something similar, though not as sleek as yours:
[image]

obelisk79, 2024-06-03:
You could add those elements in, I actually had splice dots and removed them, a full representation isn't necessary. It's abstract and simple. Either way, I attached the vector file above. I created it in inkscape.

martinrieder, 2024-06-05:
> [.. .] I thought it should be left-right oriented like a real WireViz diagram. Spinning a bit further on that thought, made me try out how I could use WireViz to create a logo suggestion. [...] My goal is to capture the essence of WireViz in a simple logo that is easy to recognize and easy to distinguish from other logos.

I love the idea. It provides a proof of the capabilities that WireViz has. Thanks for sharing the YAML code on this piece of "artwork".

> The small 16x16px version is very blurry - how can that be avoided? Is it better to reduce the number of connectors and wires from 3 to 2 and scale up pen thickness, or does it have to be hand drawn?

Why does it need to be that small if the trend is moving in another direction? Could we have different levels of detail for different resolutions? 

My humble idea that I have not tried yet:
- 4 connectors to the left, labeled with the letters W, I, R, E
- 2 wires only to have them crossed and spliced
- 3 connectors to the right, containing the letters V, I, Z

Nice Idea. There is an online tool called [SketchViz](https://www.sketchviz.com) that will generate a hand-drawn version of your graph.

My personal favorite is actually number two, of which I made a simplified version. It still needs some polishing though:

https://github.com/wireviz/WireViz/issues/373#issuecomment-2143866846
> EDIT two simplified
[image]

kvid, 2024-06-06:
@martinrieder wrote:
> [...]
> > The small 16x16px version is very blurry - how can that be avoided?
> 
> Why does it need to be that small if the trend is moving in another direction? Could we have different levels of detail for different resolutions?

16x16px seems to be the smallest icon size used at the relevant OS platforms:
https://blog.icons8.com/articles/choosing-the-right-size-and-format-for-icons/#h-desktop

tobiasfalk, 2024-07-03:
@martinrieder yes looks good, could you put it on a white background?

obelisk79, 2024-07-03:
Looks decent. As far as the properties stated in #373, I'd disregard them. In the world of graphic/logo design we aren't necessarily conveying purpose as much as we want a unique and easily identifiable design loosely associated with the purpose of the brand/product. This checks those boxes fairly well.

formatc1702, 2024-07-05:
Hi,
thanks for getting this discussion going. It's true that it would be nice to give the project a bit of a visual identity.

Some rough points from my side. Very personal opinions obviously, and up for discussion:

- All the AI-generated logos are very cluttered and lack some soul (i.e. they clearly look AI made), so I like that the conversation is going towards simplification
- Having letters/text in the logo can "cheapen" it if not done carefully
- Having the logo resemble WireViz output could be clever, but might risk making it too bland. Depends on the implementation.

So far, I see the most potential in:

- [image] by @obelisk79
  With a bit of imagination, one can almost see a W and a V. It's a VW more than a WV but subtle and far enough from Volkswagen. The connectors look a bit like Lego figures wearing tophats 🎩 which is not necessarily bad 😄 Regarding the splice dots, to me that is more reminiscent of a PCB-level schematic, and (currently), dots are not used as splices in WireViz, so I don't miss them.
- [image] by @martinrieder 
  More reminiscent of WireViz output. It evokes a letter E; perhaps rotating it can help evoke a W. Generally, if done well, a W without V could be iconic enough.

---
  
For what it's worth, here is my own attempt, quickly sketched out and terribly colored in.

Full version:
<img width="300" 
src="https://github.com/wireviz/WireViz/assets/2447251/df592b44-c7cd-4428-80fe-3fd176e48d1b"> <img width="300" 
src="https://github.com/wireviz/WireViz/assets/2447251/b6fd0cbb-44d3-4021-b1d3-62844d94f3db">

Simplified icon version (64x64, 32x32, 16x16, downscaled in browser, looks good on retina)
<img width="64" 
src="https://github.com/wireviz/WireViz/assets/2447251/88afd629-101d-42b4-accc-4f9171b783af"> <img width="64" 
src="https://github.com/wireviz/WireViz/assets/2447251/136ef538-5665-4ac2-a4ac-63c34d85499a"> <img width="32" 
src="https://github.com/wireviz/WireViz/assets/2447251/88afd629-101d-42b4-accc-4f9171b783af"> <img width="32" 
src="https://github.com/wireviz/WireViz/assets/2447251/136ef538-5665-4ac2-a4ac-63c34d85499a"> <img width="16" 
src="https://github.com/wireviz/WireViz/assets/2447251/88afd629-101d-42b4-accc-4f9171b783af"> <img width="16" 
src="https://github.com/wireviz/WireViz/assets/2447251/136ef538-5665-4ac2-a4ac-63c34d85499a">

Icon version (true 16x16 px, retina screens might apply some blurry upscaling)
<img width="16" 
src="https://github.com/wireviz/WireViz/assets/2447251/bff39f63-43ef-4a39-b4ae-548f235f10aa"> <img width="16" 
src="https://github.com/wireviz/WireViz/assets/2447251/c567239f-3603-43fd-959f-0cb5f6331d02">

- Simple basic structure: One connector on one side, a curl, a splice, two connectors/wires on the other end
- The W is quite obvious, the V is a bit more subtle
- Should work in color and B/W
- Colors were chosen quite arbitrarily
- The line thickness on the large version could probably be optimized
- 16x16 retina version is readable, true 16x16 would need some pixel tuning
- I like the proportions of the icon version better, the V on the full version is quite slanted.
- Would need to be paired with an appropriate font (also curly for cohesion, or more angular as a contrast?) when combining logo with "WireViz" spelled out next to it.

If anybody wants to try their hand at digitizing/fine-tuning it, go ahead 🚀  If I find the time, I might give it a go as well.