wireviz/WireViz #376: [feature] Single wire cable with `style: simple`

martinrieder, 2024-06-05:
The following example brings up an interesting use case for cables containing only a single wire. Similar to the way that connectors can be defined with `style: simple`, I suggest to implement cables with this attribute. It would not be displayed as a box and should optionally show the wire name/label. 

_ Credits to @EloiVilalta in https://github.com/wireviz/WireViz/issues/293#issuecomment-2150401138_

```yaml
connectors:
  C1:
    style: simple
    type: Connector
  C2:
    style: simple
    type: Connector      
cables:  
  W1:
    wirecount: 1
    show_name: false
    show_wirecount: false
    colors: [RD]

connections:
  -
    - C1: [1]
    - W1: [1]
    - C2: [1]
``` 

The issue with the current implementation is that it requires a lot of space in the graph due to the HTML table. This is not needed for only displaying a the wire label/name as a text. 

[image]

martinrieder, 2024-06-06:
@kvid I noticed that you discussed some similar idea with @formatc1702 a few years back in #31. Do you have any updates on the topic? 

> Another thing that might help, is to combine all invisible nodes on the same rank inside a sheathing into a common invisible node to avoid that Graphviz swap their order.

Taking the picture by @SnowMB as an example from this:
> You can use `rank=same` to clean up node placement. 
> 
> [image]
>

kvid, 2024-06-07:
See also https://github.com/wireviz/WireViz/issues/225#issuecomment-951164193 where @formatc1702 strongly considers consolidating the optional `style` attribute into `category` that already exists for both connectors and cables.

Maybe this issue is an extra argument for such a consolidation?

martinrieder, 2024-06-07:
If connector `type:` would not already be in use, I would suggest to use this instead of `category:` for both cables and connectors.

kvid, 2024-06-07:
@martinrieder wrote:
> I like the idea of consolidating. If connector `type:` would not already be in use, I would suggest to use this instead of `category:` for both cables and connectors.

When consolidating, we might end up with allowing more than one value for the same connector/cable, and if so, then `flags` or `class` might be alternative names? I suggest we try collecting different values that might be useful in a consolidated attribute, and see if they are mutually exclusive.

martinrieder, 2024-06-07:
I went through the list of Graphviz attributes and found that `class` is used in the [context of stylesheets](https://graphviz.org/docs/attrs/class/), so it might be reserved for also implementing these in WireViz. 

I cannot think of an example for `flags`, which implies multiple settings being combined...

What about `model` or `type`?

kvid, 2024-06-07:
Maybe just stick with `category` for now? It's perfectly legal to be member of more than one category if needed.

martinrieder, 2024-06-08:
Okay, what is the default category called then? Would it be defined as `sheathed` then?

1. Single wire
  - solid
  - stranded
2. Paired wire
  - twisted 
  - coaxial
3. Multi-core
  - sheathed
  - bundle

Any of these could have either sheathing and/or shielding. Even single wires are sometimes shielded, though this comes close to the definition of a coaxial cable then. 

Please also see https://github.com/wireviz/WireViz/issues/353#issuecomment-2121173915 where I suggested a `category` for twisted pairs.

martinrieder, 2024-06-08:
For this kind of cable, wire count and number are implicitly hidden. In the same way, I would omit the color label by default. Having it explicitly displayed would require adding `show_color` to the syntax. If there is no `wirelabel` defined, then nothing would be displayed at all.

In https://github.com/wireviz/WireViz/issues/155#issuecomment-2155874534, I suggested adding `connectorlabel`, but I realize that maybe `cablelabel` might be needed as well. In case of a single wire, this would be equal to the `wirelabel`, of course.