wireviz/WireViz #377: [feature] Unfortunate HTML output template placeholder

kvid, 2024-06-05:
HTML output templates with placeholders for generated text is a new feature introduced in v0.4. All placeholders are named ``, except one placeholder called `"sheetsize_default"` only.
1. Why is it called `sheetsize_default` and not `sheetsize`? The replacement text is fetched from `metadata.template.sheetsize`.
2. Why is it not called `` to be consistent with the other placeholders?
3. I guess the quote characters was needed to include in the search term to avoid false hits, and therefore re-inserted around the fetched value to keep the correct HTML syntax. This seems like a hack for this placeholder only, and would not be needed if using my suggestion in point 2 above. Then `` could be placed within the quote characters in the template, but the quote characters need not be replaced, and special treatment of this placeholder is no longer needed. The code would be cleaner, and a documentation description would be easier to understand because all placeholders are treated equally.

I suggest replacing https://github.com/wireviz/WireViz/blob/v0.4/src/wireviz/wv_html.py#L104-L106 with a simple entry
```py
        "<!-- %sheetsize% -->": metadata.get("template", {}).get("sheetsize", ""),
```
inserted toether with the others at https://github.com/wireviz/WireViz/blob/v0.4/src/wireviz/wv_html.py#L88
and then update the placeholder to `` in `templates/din-6771.html`

martinrieder, 2024-06-06:
I propose to use the default placeholder pattern that is used in Jinja. It would allow using this as a templating engine that is more powerful than simple replacements:

- block_start_string
  The string marking the beginning of a block. Defaults to `'{%'`.
- block_end_string
  The string marking the end of a block. **Defaults to `'%}'`.**
- variable_start_string
  The string marking the beginning of a print statement. **Defaults to `'{{'`.**
- variable_end_string
  The string marking the end of a print statement. **Defaults to `'}}'`.**
- comment_start_string
  The string marking the beginning of a comment. Defaults to `'{#'`.
- comment_end_string
  The string marking the end of a comment. **Defaults to `'#}'`.**

[Jinja Template Designer Documentation](https://jinja.palletsprojects.com/en/3.0.x/templates/)

kvid, 2024-06-07:
@formatc1702 - Is my original issue here trivial, or might there be bad side-effects?

The suggestion by @martinrieder is also valid, but that must be handled later.

closed 2024-07-13 by formatc1702

kvid, 2024-07-13:
Closed after releasing v0.4.1.

commit d2dd6015 (2024-06-11): Rename "sheetsize_default" to  (#380)

Fixes #377
commit a4c48720 (2024-06-14): Rename "sheetsize_default" to  (#380)

Fixes #377 (makes HTML output template placeholders more consistent)
commit c8307cc9 (2024-06-17): Rename "sheetsize_default" to  (#380)

Fixes #377 (makes HTML output template placeholders more consistent)
commit e55a2a48 (2024-06-17): Rename "sheetsize_default" to  (#380)

Fixes #377 (makes HTML output template placeholders more consistent)