wireviz/WireViz #378: Wiring Information Query for 70 pin connector

akikinho, 2024-06-07:
Hello Wireviz Team,
There is a 70 pin connector for which the wiring information table is huge and it is difficult to read. Is there a way to format the wire information such that the wire data is condensed for such large connectors?

kvid, 2024-06-07:
Please provide a YAML input example that demonstrates your issue. One commonly used function to make connectors more compact is to add `hide_disconnected_pins: true`.

akikinho, 2024-06-07:
Hello @kvid , here is the demo file which produces errors I don't understand as I don't know software. 
connectors:
     
  C1:
     pincount: 70
     type: Plug Housing
     subtype: male
     pins: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49,50,
     51,52,53,54,55,56,57,58,59,60,
     61,62,63,64,65,66,67,68,69,70]
        
   C2:
     pincount: 60
     type: Plug Housing
     subtype: female
     pins: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49,50,
     51,52,53,54,55,56,57,58,59,60]
     
cables:
    
  W1:
    gauge: [-,0.8,0.5,2,2,2,0.5,1,0.5,0.5,0.5,0.5,0.5,1,-,0.5,0.5,1,0.5,0.5,1,0.8,0.8,0.8,0.8,0.5,-,-,0.5,0.5,-,-,0.8,2,-,-,-,-,0.5,0.5,-,-,0.5,0.5,0.5,0.5,-,0.8,0.5,2,2,2,0.5,1,0.5,0.5,0.5,0.5,0.5,1,-,0.5,0.5,1,0.5,0.5,1,0.8]
    #length: 0.2
    type: [-,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,-,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,-,-,TXL,TXL,-,-,TXL,TXL,-,-,-,-,TXL,TXL,-,-,TXL,BC,XLPE,XLPE,-,-,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL,-,TXL,TXL,TXL,TXL,TXL,TXL,TXL,TXL]
    # color_code: IEC
    wirecount: 70
    show_equiv: true # auto-calculate AWG equivalent
    colors: [-,RD,OG,BN,GY,BN,OG,BN,YE,GY,YE,BU,YE,GY,-,YE,RD,BN,GN,YE,GY,BN,BK,BN,BK,BU,-,-,YE,GN,-,-,BK,GY,-,-,-,-,PK,YE,-,-,RD,SH,GN,YE,-,-,RD,OG,BN,GY,BN,OG,BN,YE,GY,YE,BU,YE,GY,-,YE,RD,BN,GN,YE,GY,BN,BK]
    shield: false # add cable shielding
    category: bundle
    wirelabels: [-,A,B,C,D,E,F,G,H,I,J,K,L,M,
    -,N,O,P,Q,R,S,T,U,V,W,X,-,-,Y,Z,-,-,A1,B1,
    -,-,-,-,C1,D1,-,-,E1,F1,G1,H1,-,-,I1,J1,K1,L1,M1,N1,O1,P1,Q1,R1,S1,T1,U1,
    -,V1,W1,X1,Y1,Z1,A2,B2,C2]

connections:
-
    - C1: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49,50,
     51,52,53,54,55,56,57,58,59,60,
     61,62,63,64,65,66,67,68,69,70]
    - W1: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49,50,
     51,52,53,54,55,56,57,58,59,60,
     61,62,63,64,65,66,67,68,69,70]
    - C2: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49,50,
     51,52,53,54,55,56,57,58,59,60]

kvid, 2024-06-07:
I took the liberty to edit you message, adding YAML code tags around your YAML to make it easier to read. Please verify that I didn't destroy your YAML input code in the process.

I had to make a few modifications to make WireViz accept your YAML input:
- Reduced C2 indentation with 1 space to make it equal C1 indentation
- Replaced your lists in `gauge` and `type` with simple strings to avoid ugly output (different values per wire not supported)
- Removed pins 61-70 in C1 connections to equalize the number of connections in the same connection set

Non-critical issues:
- The warning `Unknown color name: SH` can be ignored (default color white will be used)
- You don't need to list all pins individually when you use `pincount: 70` and `pincount: 60` in your connectors.
- I recommend you use `[1-60]` in your connection set instead of listing all number individually.
```yaml
connectors:
  C1:
     pincount: 70
     type: Plug Housing
     subtype: male
  C2: # Removed 1 space indent
     pincount: 60
     type: Plug Housing
     subtype: female
cables:
  W1:
    gauge: 1 per_wire_not_supported! # [1, 0.7, 0.7, 0.7, 0.7, 0.9, 0.5, 0.7, 0.5, 0.9, 0.7, 1, 1, 0.9, 0.5, 1, 0.9, 1, 0.6, 0.6, 0.7, 1, 0.7, 0.9, 0.5, 1, 0.8, 0.7, 0.9, 0.8, 0.5, 0.6, 0.6, 0.5, 0.7, 1, 0.5, 1, 1, 0.7, 0.8, 0.9, 0.7, 0.8, 0.5, 0.9, 0.9, 0.9, 1, 0.7, 0.9, 0.6, 0.5, 0.6, 0.6, 0.5, 0.7, 0.5, 0.9, 1, 0.8, 1, 0.6, 0.9, 0.7, 0.9, 0.7, 1, 0.7, 0.6, 0.6, 0.7, 0.5, 0.9, 0.5, 1, 1, 0.5, 1, 0.8, 1, 0.5, 0.9, 0.8, 0.9, 0.9, 0.9, 0.5, 0.6, 0.9, 0.9, 0.8, 0.8, 1, 0.8]
    length: 0.2
    type: TXL
    # color_code: IEC
    wirecount: 70
    show_equiv: true # auto-calculate AWG equivalent
    colors: [BN, YE, BN, BU, RD, BN, GN, YE, YE, YE, BK, GN, GY, RD, YE, YE, BK, GY, YE, GY, GY, OG, BN, BN, YE, BK, YE, BN, BN, RD, BN, YE, BU, RD, YE, YE, BN, BN, GY, BK, OG, BK, BN, OG, RD, YE, GY, BN, YE, BK, YE, YE, GY, RD, YE, RD, YE, GY, YE, BN, YE, YE, BN, YE, YE, BN, BN, GY, SH, OG, YE, GY, BN, YE, OG, GN, GY, BN, YE, GY, BU, RD, YE, YE, BN, YE, YE, GN, GY, BN, YE, YE, YE, YE, RD, YE, RD, BN, BN, GY, OG, BK, YE, YE]
    shield: true # add cable shielding
    category: bundle
    wirelabels: [-,A,B,C,D,E,F,G,H,I,J,K,L,M,
      -,N,O,P,Q,R,S,T,U,V,W,X,-,-,Y,Z,-,-,A1,B1,
      -,-,-,-,C1,D1,-,-,E1,F1,G1,H1,-,-,I1,J1,K1,L1,M1,N1,O1,P1,Q1,R1,S1,T1,U1,
      -,V1,W1,X1,Y1,Z1,A2,B2,C2]
connections:
  -
    - C1: [1-60] # Removed pins 61-70
    - W1: [1-60]
    - C2: [1-60]
```

akikinho, 2024-06-10:
Thank you @kvid! I have a question. How do you display the gauge, type information beside the wire information? Is it something like show_type: true and show_gauge: true?

kvid, 2024-06-10:
WireViz v0.4 only supports a common `type` and `gauge` value for all wires in the same cable/bundle - see [syntax description](https://github.com/wireviz/WireViz/blob/v0.4/docs/syntax.md#cable-attributes), but #268 suggests supporting different gauges for each wire as a new future feature. However, it has not (yet?) been implemented.

akikinho, 2024-06-10:
Thank you @kvid. I really hope in the future update the team implements it as it will be really helpful.

akikinho, 2024-06-10:
In my opinion, both (length: <int/float>[ <unit>])and (gauge: <int/float/str>) should be of type List.

kvid, 2024-06-10:
Please join the #268 discussion and present your use case and view to contribute to the on-going suggestions there.

kvid, 2024-07-06:
This issue seems to be resolved. I copied your two latest comments into #268 as they seems to belong there.

closed 2024-07-06 by kvid