wireviz/WireViz #384: Ferrule dot notation does not work

chironjit, 2024-06-13:
Using the example from this page: https://github.com/wireviz/WireViz/blob/master/tutorial/readme.md#06---custom-ferrules

Auto-generating ferrules using this example provided:

```
connectors:
  X1:
    pinlabels: [+12V, GND, GND, +5V]
    type: Molex 8981
    subtype: female
  F_10: # this is a unique ferrule
    style: simple
    type: Crimp ferrule
    subtype: 1.0 mm²
    color: YE # optional color
  F_05: # this is a ferrule that will be auto-generated on demand
    style: simple
    type: Crimp ferrule
    subtype: 0.5 mm²
    color: OG

cables:
  W1:
    category: bundle # bundle
    length: 0.3
    gauge: 0.5 mm2
    colors: [YE, BK, BK, RD] # custom colors, wirecount is implicit

connections:
    -
      - [F_05., F_10.F1, F_10.F1, F_05.]
      - W1: [1-4]
      - X1: [1-4]
```

Gives this error:

```
Traceback (most recent call last):
  File "/home/user/.miniconda3/version/bin/wireviz", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/.miniconda3/version/lib/python3.11/site-packages/wireviz/wireviz.py", line 266, in main
    parse(yaml_input, file_out=file_out)
  File "/home/user/.miniconda3/version/lib/python3.11/site-packages/wireviz/wireviz.py", line 103, in parse
    raise Exception('First item not found anywhere.')
Exception: First item not found anywhere.
```

I've tested all the examples of dot notations for auto-generating ferrules and none of them work. There must have been a change at some point that broke this.

kvid, 2024-06-13:
@chironjit - Thank you for reporting this error, but please provide your version numbers as described in the contributing guidelines.

chironjit, 2024-06-13:
Hey, sorry, missed that. version I am using is 0.3.2. 

Versions:
`wireviz 0.3.2`
`graphviz version 9.0.0 (20230911.1827)`
`Python 3.11.5`

I see that there is a release version 4.0. Not sure why mine is on 0.3.2 I will try the latest version tomorrow and report back if the problem persists.

kvid, 2024-06-13:
The syntax you're trying to use is new with v0.4: https://github.com/wireviz/WireViz/blob/master/docs/CHANGELOG.md

chironjit, 2024-06-14:
Cool, thank you!

closed 2024-06-14 by chironjit