wireviz/WireViz #389: [query] Non-connector

akikinho, 2024-06-17:
Hello Wireviz Team,
Is there any 'non connectors' object in Wireviz instead of the syntax
connectors:
cables:
connections:

Thank you!

kvid, 2024-06-17:
Currently, WireViz only supports https://github.com/wireviz/WireViz/blob/master/docs/syntax.md

However, maybe the suggestions in #142 might be what you seek? If so, then please contribute there and close this issue. Otherwise, please explain what kind of "_non connectors_" objects you seek.

kvid, 2024-07-06:
@akikinho - Please describe your use case and expectations a bit more, so it'll be easier for others to help with suggestions.

akikinho, 2024-07-11:
Hello @kvid, I was wondering if there is an object for representing relays?
[image]

kvid, 2024-07-14:
Similar to the answers you got in #397 and #402, using a connector is currently the best we can offer to represent such components.

You have now convinced me that your request is a duplicate of #142, so please contribute with questions and suggestions there so we can close this issue.

akikinho, 2024-07-16:
ok @kvid