wireviz/WireViz #390: [query] Wireviz file not working

akikinho, 2024-06-18:
Hello Wireviz Team,
I am faced with an issue which I am unable to diagnose and understand the reason. The error Exception: First item not found anywhere.

Thank you in advance!

connectors:

CX:
     pincount: 49
     type: Plug Housing
     subtype: male
     pins: [1,2,3,4,5,6,7,8,9,10,
     11,12,13,14,15,16,17,18,19,20,
     21,22,23,24,25,26,27,28,29,30,
     31,32,33,34,35,36,37,38,39,40,
     41,42,43,44,45,46,47,48,49]

S: &splice  # generic autogenerated splice
     style: simple
     autogenerate: true
     type: splice
     show_name: true
     
SPLICE_CX:
     <<: *splice          # copy attributes from 'S'
     autogenerate: false  # but disable autogeneration
     show_name: true     # this is currently necessary when autogenerate == false

cables:

WIRE SPLICE_CX:
    gauge: 1
    type: TXL 
    wirecount: 3
    show_equiv: true # auto-calculate AWG equivalent
    colors: [RD,GN,SH]
    shield: false # add cable shielding
    category: bundle
    wirelabels: [A,B,C]
    
connections:
-
    - SPLICE_CX
    - WIRE SPLICE_CX: [2]
    - CX: [45]

kvid, 2024-06-18:
@akikinho - Thank's for reporting this error, but please provide your version numbers as described in the Contribution Guidelines.

See also #207 that seems to report a similar error message.

akikinho, 2024-06-18:
Thank you @kvid. Here is the version number: wireviz 0.3.2

akikinho, 2024-06-19:
Thank you! It works now.

kvid, 2024-06-19:
@akikinho - I'm glad your problem got solved, but please explain what you changed to solve the problem, so others with similar problems might learn from your use case.

akikinho, 2024-06-20:
Hello @kvid, the indentation was not correct as per @martinrieder, so I corrected it in the YAML file.

closed 2024-06-22 by kvid