wireviz/WireViz #391: Help with OS Error

ElPi23, 2024-06-19:
Hello everyone,
I have received the following error and can't figure out what the problem is.

Error:
`ERROR: Traceback (most recent call last):
File "", line 198, in _run_module_as_main
File "", line 88, in _run_code
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Scripts\wireviz.exe\__main__.py", line 7, in
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1157, in __call__
return self.main(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1078, in main
rv = self.invoke(ctx)
^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 1434, in invoke
return ctx.invoke(self.callback, **ctx.params)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\click\core.py", line 783, in invoke
return __callback(*args, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\wireviz\wv_cli.py", line 141, in wireviz
wv.parse(
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\wireviz\wireviz.py", line 88, in parse
yaml_data, yaml_file = _get_yaml_data_and_path(inp)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\wireviz\wireviz.py", line 416, in _get_yaml_data_and_path
raise e
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\site-packages\wireviz\wireviz.py", line 406, in _get_yaml_data_and_path
yaml_path = Path(inp).expanduser().resolve(strict=True)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\elias.pircher\AppData\Local\Programs\Python\Python312\Lib\pathlib.py", line 1240, in resolve
s = self._flavour.realpath(self, strict=strict)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "", line 730, in realpath
OSError: _getfinalpathname: path too long for Windows`

The YML file:  [YMLFILE.txt](https://github.com/user-attachments/files/15901304/YMLFILE.txt)

I have already uninstalled and reinstalled wireviz without success
i have also tried to save the file to a location with a shorter path without success

Wireviz version: 4.0
Graphviz version 11.0.0 (20240428.1522)
Python 3.12.3

ElPi23, 2024-06-20:
i have installed the development version and when i use prependet file i still get the same error message.

it seems to have something to do with the length of the content of the .yml file because if i shorten it it works (even if i only delete comments)

kvid, 2024-06-20:
@ElPi23 - Can you confirm that PR #392 solves your issue, or do you still get an error message?

ElPi23, 2024-06-21:
with the vesion wireviz-0.4.1.dev0 it does not work for me 
(i hope i did everything right i have never worked with git before)

I have done everything once again as described in the first comment of @martinrieder

ElPi23, 2024-06-21:
Thank you 
can confirm that it is now working 

many thanks for the fast and awesome support

kvid, 2024-07-06:
The relevant fixes are now merged into the `release/v0.4.1-rc` branch of PR #365 that is not yet released, but also describes how you can install the latest collection of fixes before the release.

closed 2024-07-13 by formatc1702

kvid, 2024-07-13:
Closed after releasing v0.4.1.

commit 646f66b1 (2024-07-04): Exception on e.errno is None. Fixes #391.