wireviz/WireViz #396: [syntax error]: Splice error in Wireviz

akikinho, 2024-06-26:
Hello Wireviz team,

I am getting an error in using splice 
(ValueError: 'autogenerate' in Connector 'SPLICE': 'autogenerate' is replaced with new syntax in v0.4) in the new version. Can you please help ?

Package         Version
--------------- -----------
click           8.1.7
colorama        0.4.6
graphviz        0.20.3
numpy           1.26.4
pandas          2.2.2
pillow          10.3.0
pip             24.0
python-dateutil 2.9.0.post0
pytz            2024.1
PyYAML          6.0.1
setuptools      70.1.0
six             1.16.0
tzdata          2024.1
wireviz         0.4.1.dev0

kvid, 2024-06-26:
@akikinho - you have not provided your YAML input (that normally makes it easier to help you), but the error message describes the problem pretty well:
- _ValueError: 'autogenerate' in Connector 'SPLICE':_  - You have used the `autogenerate` attribute in your Connector named 'SPLICE'
- _'autogenerate' is replaced with new syntax in v0.4_ - The `autogenerate` attribute is no longer supported by v0.4 because the same functionality is now available with the new syntax using the dot notation as you already have used in your previous issue #393.

Please read the documents I linked to in https://github.com/wireviz/WireViz/issues/393#issuecomment-2181089582 and https://github.com/wireviz/WireViz/issues/393#issuecomment-2187461366

akikinho, 2024-06-26:
Thank you @kvid. I did not understand how to use the dot notation. Here is the yaml file

```
connectors:           
    CX:
         pincount: 5
         type: Plug Housing
         subtype: female
         pins: [1,2,3,4,5]
            
    S: &splice  # generic autogenerated splice
         style: simple
         autogenerate: true
         type: splice
         show_name: true
     
    SPLICE G:
         <<: *splice          # copy attributes from 'S'
         autogenerate: false  # but disable autogeneration
         show_name: true     # this is currently necessary when autogenerate == false
         
cables:
    WIRE G:
        gauge: 1 
        #length: 0.2
        type: TXL
        # color_code: IEC
        wirecount: 5
        show_equiv: true # auto-calculate AWG equivalent
        colors: [GY,OG,GN,RD,BN]
        # shield: true # add cable shielding
        category: bundle
        wirelabels: [A,B,C,D,E]

        
connections:        
-
    - CX: [B]
    - WIRE G: [2]
    - SPLICE G
   
```

akikinho, 2024-06-26:
@kvid it works now
```
connectors:           
    CX:
         pincount: 5
         type: Plug Housing
         subtype: female
         pins: [1,2,3,4,5]
            
    S: &splice  # generic autogenerated splice
         style: simple
         type: splice
         show_name: true
         
cables:
    WIRE G:
        gauge: 1 
        #length: 0.2
        type: TXL
        # color_code: IEC
        wirecount: 5
        show_equiv: true # auto-calculate AWG equivalent
        colors: [GY,OG,GN,RD,BN]
        # shield: true # add cable shielding
        category: bundle
        wirelabels: [A,B,C,D,E]

        
connections:        
-
    - CX: [2]
    - WIRE G: [2]
    - S.SPLICE G
   
```

kvid, 2024-06-26:
Remove all `autogenerate` attributes! When you want unnamed splices/connectors/cables to be autogenerated, then append a dot to their names in the connection sets.

Please read the comment and the document I linked to in https://github.com/wireviz/WireViz/issues/393#issuecomment-2187461366 and explain what parts you don't understand.

kvid, 2024-06-26:
I'm sorry, but my browser didn't show your latest comment while I wrote my answer, so saw it too late...

closed 2024-07-06 by kvid