wireviz/WireViz #397: [query] CHASSIS GROUND (MAIN)

akikinho, 2024-06-28:
Hello,
Is there a way to show the chassis main (ground) in Wireviz ? Can it be shown using splice?

Thank you and Best Regards,

kvid, 2024-06-28:
> Try creating a connector with `style: simple` and `type: GROUND` attributes.

That will work, and a "Connector, GROUND" entry will be added to the BOM unless you also add connector attribute `ignore_in_bom: true`.

> It is a bit counter-intuitive that type describes the label that is displayed. Effectively, this kind of connector is also what can be used as a splice.

The `type` describes the type of connector (not label) and is also used in the BOM entry. The name of a `simple` connector is hidden by default unless you also add connector attribute `show_name: true`.

It's also possible not providing a `type`, but with an empty `type`, the connector node becomes very tiny unless also adding the name or e.g. an image like this:
[image]

kvid, 2024-06-29:
Chassis ground is not necessarily the same as earth ground, and I didn't find the chassis ground symbol as a unicode character.

akikinho, 2024-07-09:
Thank you @kvid @martinrieder !

closed 2024-07-09 by kvid