wireviz/WireViz #399: [query] yaml input file throwing error

akikinho, 2024-07-01:
Hello Wireviz Team,
I am facing an issue with the Wireviz software recently after the update. Sometimes the yaml input file runs and sometimes it doesn't. Next when I switch to another new file and start again it works and then it fails again. I am wondering if there is some random bug in the software which breaks the code? I get the following error again and again until I switch to a new file and start again from the beginning.  Could you please guide regarding this issue and how to best go about resolving it?

File "<frozen ntpath>", line 730, in realpath
OSError: _getfinalpathname: path too long for Windows

Package         Version
--------------- -----------
click           8.1.7
colorama        0.4.6
graphviz        0.20.3
numpy           1.26.4
pandas          2.2.2
pillow          10.3.0
pip             24.0
python-dateutil 2.9.0.post0
pytz            2024.1
PyYAML          6.0.1
setuptools      70.1.0
six             1.16.0
tzdata          2024.1
wireviz         0.4.1.dev0

Microsoft Windows [Version 10.0.19045.4529]

Thank you and Best Regards,

kvid, 2024-07-01:
It seems you already are using the branch of PR #365 that should include a fix for a similar error, but I also wait for PR #392 to be ready for merging into #365.

In the meantime you could try the current branch of #392 to see if it already contains the fix you need. If not, I suggest you add a comment to #392 with a reference to this issue and ask the author if the issues might be related.

Please provide your Python version and a YAML input that triggers the error.

akikinho, 2024-07-04:
Hello @kvid,
The Python version is Python 3.12.3 and the YAML file is too big.

akikinho, 2024-07-07:
Hello @kvid, I am wondering how to call the #392 ? Is it using this command gh pr checkout 392?

kvid, 2024-07-07:
@akikinho wrote:
> Hello @kvid, I am wondering how to call the #392 ? Is it using this command gh pr checkout 392?

That command might still work, so please try that. However, as #392 now is already merged into PR #365, you might also try that PR. See my previous comment above.

akikinho, 2024-07-08:
Hello @kvid, I downloaded the #392 wireviz.py file into the local system and ran the yaml input file and it showed the same error.

kvid, 2024-07-08:
@akikinho wrote:
> Hello @kvid, I downloaded the #392 wireviz.py file into the local system and ran the yaml input file and it showed the same error.

Thank's for reporting this. Please provide a full copy of text output in your terminal window of your latest test, starting with the command you execute, all output from WireViz, including stack trace and exception output.

Then I want you to provide a full copy of text output in your terminal window of you testing #365, starting with the installation commands described in that PR description (`pip uninstall wireviz` + `pip install git+https...` , including executing wireviz on your YAML input, all output from WireViz, including stack trace and exception output.

When you paste in such a large text dump that I ask for in your comment, I recommend you select all lines of the text dump and press the `<> Code` button to format the text dump as raw text.

kvid, 2024-07-14:
@akikinho - Now it's too late to try those PRs I requested you to try. The fixes was released as v0.4.1 yesterday.

If this issue is still a problem with the latest release, you must provide a full copy of all text output in your terminal window while you run the latest release (as I described with the PR testing above). Otherwise, please close this issue as completed.

kvid, 2024-07-16:
@akikinho - I'm glad that this issue has been resolved with the v0.4.1 release (confirmed by you closing the issue as completed). 🎉