wireviz/WireViz #401: TypeError from pinlabels: [NO, VCC]

quozl, 2024-07-08:
When a connector has a pinlabels containing NO, for a normally open switch, WireViz does fail with TypeError;

```log
Traceback (most recent call last):
  File "wireviz", line 8, in <module>
    sys.exit(wireviz())
  File "click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "wireviz/wv_cli.py", line 141, in wireviz
    wv.parse(
  File "wireviz/wireviz.py", line 382, in parse
    harness.output(filename=output_file, fmt=output_formats, view=False)
  File "wireviz/Harness.py", line 679, in output
    graph = self.graph
  File "wireviz/Harness.py", line 653, in graph
    self._graph = self.create_graph()
  File "wireviz/Harness.py", line 489, in create_graph
    from_string = ":".join(from_info)
TypeError: sequence item 2: expected str instance, bool found
```

Workaround is to enclose NO with double quotes.

kvid, 2024-07-09:
This seems to be a duplicate of #305

quozl, 2024-07-09:
Agreed.  Have now read #305.  Wow.  Sorry!

closed 2024-07-09 by quozl