wireviz/WireViz #403: I feel dumb for asking this: is there a way to elegantly use this to display complete cables instead of individual wires?

Shadowphoenix, 2024-07-09:
[image]

I did it by just assuming each cable to be one wire and each connector to be a "pin" but i think this is not the way.

Also, for my purposes i technically dont need to know the specs of each individual cable, in theory it would work if it was just a line from DMX4X M2 to MOV3-I without having the Chart specifying W2 in the middle - is there a way to not display cable-specs? Technically i could create such a graph in another tool but I decided to give this one a try to learn how to use this later if i ever actually need the entire functionality of specifying connectors wire by wire.

kvid, 2024-07-10:
Please read usage examples in https://github.com/wireviz/WireViz/tree/master/docs#examples and alternative cable visualization suggestions in #212 and #376.

Shadowphoenix, 2024-07-10:
> Please read usage examples in https://github.com/wireviz/WireViz/tree/master/docs#examples and alternative cable visualization suggestions in #212 and #376.

thanks :D