wireviz/WireViz #404: [discussion] make position of pin numbers within connectors consistent?

formatc1702, 2024-07-12:
Split of from https://github.com/wireviz/WireViz/issues/286#issuecomment-2225714107 , image taken from there

[image]

Currently, pin numbers appear on the left or the right of the connector's pin table depending on which side the wires come out.

This was implemented since the very first draft of WireViz and has remained unchanged. As more and more fields can/will be defined on a pin level (color, notes, shorts/bridges, ...), it can cause confusion.

Should pin numbers always appear on the left, to keep the order of the columns consistent across all connectors?
Personally, I am leaning more and more in favor of this solution.

AlanLivingston, 2024-07-12:
I use these wireviz drawings to document cabling and to assist technicians with construction cables.  Our techs like the pin numbers next to the wires, as that's the information they use most when making the cable.  Many times, they don't care about or refer to the signal name. I'd prefer to move the signal names into the rightmost column after the shorts, so that it's a mirror image of the left connector.

formatc1702, 2024-07-12:
> Our techs like the pin numbers next to the wires, as that's the information they use most when making the cable.

That was the thinking behind the initial behavior :)

tobiasfalk, 2024-07-14:
First, I would not talk about left and right but rather see it like an onion, with a core, outside, and inside (currently empty) layers. Currently, the outside layer is always th Nr. and then the core goes lable and jumpers

Second, I think that this outside and inside layers are mirrored if wiers are connected to both sides of the connector.

It would be reasonable to put the lable into the first inside layer.

It would also be good if it would be selectable if these layers are always left, right, both sides or auto(similar to current).

It would also be reasonable if it would be possible to select if the lable is part of the core or layer.

I think that things like the jumpers should remain in the core.

kvid, 2024-09-12:
@martinrieder wrote:
> I wrote [#404 (comment)](https://github.com/wireviz/WireViz/issues/404#issuecomment-2225983880) under the assumption that `hide_disconnected_pins` is active. IMO this should treat both sides independently to produce a more compact output. Any opinions on this?

- When hiding disconnected pins, it makes sense to hide the whole row because no information about those pins are needed.
- Pins that are connected at one side and disconnected at the other side in the diagram, _are_ indeed connected in real life, and all information about them are needed. Do you want to replace the pin name at the disconnected side with an empty cell? That will not be more compact. I don't really see the benefit. Please elaborate.

kvid, 2024-10-22:
- Personally, I like the pin names close to the wire ends (at both sides when needed) for the same reason as described by @AlanLivingston  above.
- Technically, the wire splines need named ports as anchors to align the wire ends with the correct pin rows. These ports must be the leftmost cells for left-side wires and rightmost cells for right-side wires. Together with the point above, the easy choice is to keep the pin name cells as such ports, as is done today.