wireviz/WireViz #407: Several pictures for one connector ?

bigboss59400, 2024-07-14:
Hi,
Is there a way to have several pictures for one connector ? (of course a workaround is to create group everything in one ) ?
THanks

kvid, 2024-07-14:
Except for the work-around you already mention to create a combined image from several others, there is currently not possible to include more than one per connector/cable.

There might be several ways to implement such a feature. Would you suggest a list of images that are rendered below each other in the same node, or did you have something else in mind?

bigboss59400, 2024-07-14:
Yes load multiple images for the same connector,
Yes pictures below each other would be fine
I feel it's nice to have the pinout of the connector and at least one real picture of the connector 

I will go with the workaround for the moment