wireviz/WireViz #409: [bug] Unable to install package on Windows 11

superlou, 2024-07-15:
Installation via pip fails on Windows 11, with the following error:

```
pip install wireviz
...
Installing collected packages: pyyaml, graphviz, click, wireviz
  WARNING: Failed to write executable - trying to use .deleteme logic
ERROR: Could not install packages due to an OSError: [WinError 2] The system cannot find the file specified: 'C:\\Python312\\Scripts\\wireviz.exe' -> 'C:\\Python312\\Scripts\\wireviz.exe.deleteme'
```

Interestingly, if I run `pip install wireviz` again, pip thinks it is installed (requirement already satisfied: wireviz), but running `wireviz` gives an executable not found error.

Environment:

* Wireviz version: 0.4.1
* Graphviz: 11.0.0 (20240428.1522)
* Python: 3.12.3 (64-bit)
* OS: Windows 11 Business, 22H2

kvid, 2024-07-15:
@superlou wrote:
> Installation via pip fails on Windows 11, with the following error:
>
> ```
> pip install wireviz
> ...
> Installing collected packages: pyyaml, graphviz, click, wireviz
>   WARNING: Failed to write executable [...]

Thank's for reporting this issue.

Maybe your current user doesn't have the write privileges in the destination folder?

Please try first creating a virtual environment in a folder where you have write privileges (e.g. your `Documents` folder), and activating it, before trying installing again (now into the virtual environment):
```cmd
cd %USERPROFILE%\Documents
python3 -m venv .venv
call .venv/Scripts/activate.bat
pip install wireviz
```

superlou, 2024-07-15:
Installing via venv (used powershell rather than bat, but similar) worked. I was surprised about privileges, even though I don't have admin, since my Python installation is for my user. After digging into it more, it looks like I had a weird combination of Path variables that were using C:\Python311 which was the problem. Even without a venv, it works fine on a user-only Python installation. Thanks!

closed 2024-07-15 by superlou