wireviz/WireViz #411: [query] Shielding 2 separate wires

akikinho, 2024-07-17:
Hello,
I would like to know how to shield 2 separate wires connected to 2 pins into a cable?

[image]

Thank you and Best Regards,

kvid, 2024-07-17:
This seems to be a duplicate of #330. If you agree, please join that discussion, and close this issue as not planned. If not, then explain here what makes this issue different.

akikinho, 2024-07-17:
Hello @kvid, here is the yaml input file. The shield has to be around 10A and 11A
`connectors:
    CX:
        pincount: 23
        type: Plug Housing
        subtype: female
        pins: [1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23]
        hide_disconnected_pins: true
    
    S: &splice  # generic autogenerated splice
        style: simple
        type: splice
        show_name: true
        
cables:
    WIRE SPLICE 10_A:
        gauge: 1 
        #length: 0.2
        type: TXL
        # color_code: IEC
        wirecount: 3
        show_equiv: true # auto-calculate AWG equivalent
        colors: [GN,GN,GN]
        shield: true # add cable shielding
        category: bundle
        wirelabels: [10B,10H,10A]
        
    WIRE SPLICE 11_A:
        gauge: 1 
        #length: 0.2
        type: TXL
        # color_code: IEC
        wirecount: 3
        show_equiv: true # auto-calculate AWG equivalent
        colors: [YE,YE,YE]
        shield: true # add cable shielding
        category: bundle
        wirelabels: [11B,11H,11A]
        
connections:
-
    - CX: [5]
    - WIRE SPLICE 10_A: [3]   #10A
    - S.SPLICE 10_A
-
    - CX: [6]
    - WIRE SPLICE 11_A: [3]   #11A
    - S.SPLICE 11_A`

kvid, 2024-07-19:
@akikinho wrote:
> Hello @kvid, here is the yaml input file. The shield has to be around 10A and 11A
> [...]

The YAML you provided is perfectly valid in v0.4.1 and produces a harnes with two shielded bundles of 3 wires each where the shield normally is interpreted to contain all 3 wires, but it's possible to add notes that explains such details differently.

If you expect a way to specify which wire(s) are shielded individually or together with some other wire(s), and visualize this somehow, then your request is a duplicate of #330, and my recommendation above still holds. If this is not what you expect, then you need to describe in detail the differences between your expected output and the output using the currently latest version. A sketch where you hand draw on top of the generated output might be useful.

PS: I still don't understand what is shown in the image you included in the original message above?