wireviz/WireViz #413: How to deal with a lot of connection ?

bigboss59400, 2024-07-20:
hello,
I have a big connector (100+ pins), I can't find a way to make it more viewable, I thought about increasing the space between the connector and the connections however I don't know if it's psosible ? if so how to do it ? is it possible to apply this kind of settings only for a particular connector ?

I'm open to suggestions too

[image]

kvid, 2024-07-20:
This looks very similar to #174, so I recommend a similar work-around:
```yaml
tweak:
  override:
    graph:
      ranksep: '6'  # Larger horizontal separation between all nodes
```
Default `ranksep` value is 2. You should try out how much you need to increase it in your use case.

Unless you argue why your use case is not covered by #174, I'm tempted to close this issue as a duplicate.

kvid, 2024-07-21:
My suggested work-around above affects the global spacing. I don't know any way to specify a larger spacing for only one node (connector in your case) - except maybe to insert an invisible point node before and/or after your node and connect them with invisible edges using `tweak.append`, but I haven't tried it (yet).