wireviz/WireViz #414: How to draw small width wires ?

bigboss59400, 2024-07-20:
Hello,

Is there a way to modify the aesthetics of the connections between the connectors? I would prefer something simpler, such as no borders, just a line.

Thanks!

tobiasfalk, 2024-07-20:
Do you mean something like this?
[image]

bigboss59400, 2024-07-20:
Exactly ! could you share how did you do it please ?

tobiasfalk, 2024-07-20:
I edited the Python code that generates the wire edges.

bigboss59400, 2024-07-20:
> I edited the Python code that generates the wire edges.

Do you mind share it ?

tobiasfalk, 2024-07-20:
In my fork #412 (https://github.com/tobiasfalk/WireViz/tree/Jumpers_reimp2) I changed line 455 and 492 of `src/wireviz/wv_graphviz.py` there I removed the `#000000`(HEX for Black) and the `:`