wireviz/WireViz #415: Period in Connector Name not working in Connections:

eovnu87435ds, 2024-07-24:
Hello! I recently found a bug with Wireviz 0.4.1 on Linux. 

I can create a connector with a decimal point in it, and wireviz renders everything correctly.  Example:

```
  Connector Molex Picoblade 1.25 4 Position:
    mpn: 510210400
    manufacturer: Molex
    pn: E3-00052
    pins: [1,2,3,4]
    pinlabels:
      - ETH_RX-
      - ETH_RX+
      - ETH_TX-
      - ETH_TX+
```

Wireviz will continue to render properly until I go to use this connector in a Connection, such as this example:

```
connections:
  -
   - Connector JST ZPD 26 Position: [17,15,11,9]
   - Ethernet Cable: [1,2,3,4]
   - Connector Molex Picoblade 1․25 4 Position: [1,2,3,4]
```

When attempting to render with this connector, I get this error:

```
WireViz 0.4.1
Input file:   harness1.yaml
Output file:  harness1.[html|png|svg|tsv]
Traceback (most recent call last):
  File "/usr/bin/wireviz", line 33, in <module>
    sys.exit(load_entry_point('wireviz==0.4.1', 'console_scripts', 'wireviz')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/wireviz/wv_cli.py", line 141, in wireviz
    wv.parse(
  File "/usr/lib/python3.12/site-packages/wireviz/wireviz.py", line 306, in parse
    raise Exception(
Exception: Connector Molex Picoblade 1 is an unknown template/designator/arrow.
```

As a workaround, if I replace the period / decimal point with "One Dot Leader" (U+2024), everything renders correctly.  Is there a way to escape the decimal so that I don't have to use a homoglyph? 

Let me know if I can provide any further information!

kvid, 2024-07-24:
Please try setting a custom `template_separator` value, e.g.:
```yaml
options:
  template_separator: '!'
```
See doc:
- https://github.com/wireviz/WireViz/blob/master/docs/syntax.md#options
- https://github.com/wireviz/WireViz/blob/master/docs/syntax.md#autogeneration-of-items

eovnu87435ds, 2024-07-24:
D'oh! Always have to RTFM!

Tested with the template separator, everything works as intended! Closing issue.

closed 2024-07-24 by eovnu87435ds