wireviz/WireViz #417: How to draw wire studs?

0x0fe, 2024-07-27:
I am struggling to reproduce something like this, no problem with the conenctors, however i am really not sure how to generate the two wire studs (red and green) i tried to add two ferrules but i always get errors since the synthax of using ferrules is unclear.

[image]

0x0fe, 2024-07-27:
So this is the best i can do, i wish we could make X3 view less anbiguous, for example if it could be a little higher the wiers bend would be more clear.

```
connectors:
  X1:
    type: VH3.96
    subtype: female
    pinlabels: [12V, R1CO, R2CO, R1NO, R2NO, GND]
    image:
      src: VHR-6N.png
      height: 70  
      caption: 

  X2: 
    type: TOYOTA
    subtype: female
    pinlabels: [R2CO, GND, R1CO, W4, R1NO, W6, R2NO]
    image:
      src: toyota7p_f.png
      height: 70  
      caption: 
  X3:
    type: TOYOTA
    subtype: male
    pinlabels: [R2CO, GND, R1CO, W4, R1NO, W6, R2NO]
    image:
      src: toyota7p_m.png
      height: 70  
      caption: 

cables:
  W1: 
    gauge: 20 awg
    length: 0.1
    colors: [GN, VT, BN, VT, RD]
    category: bundle
  W2: 
    gauge: 20 awg
    length: 0.1
    colors: [VT, RD, GN, BK, BN, OG, VT]
    category: bundle
  W3: 
    gauge: 20 awg
    length: 0.1
    colors: [RD, GN]
    category: bundle

connections:
  -
    - X1: [2-6]
    - W1: [1,2,3,4,5]
    - X3: [3,1,5,7,2]
  -  
    - X2: [1-7]
    - W2: [1-7]
    - X3: [1-7]
  -
    - X1: [1-2]
    - W3: [1-2]
```

[image]

kvid, 2024-07-28:
@0x0fe wrote:
> I am struggling to reproduce something like this, no problem with the conenctors, however i am really not sure how to generate the two wire studs (red and green) i tried to add two ferrules but i always get errors since the synthax of using ferrules is unclear.

Please help us understand what parts (of docs/examples/tutorial) are unclear, so we can try improving them. I assume you want two equal ferrules in your use case. Then you only need to define it once, e.g. like this:
```yaml
connectors:
  F1:
    style: simple  # Single-pin connector with simplified rendering
    type: Ferrule  # Type of connector
    color: OG  # Optional color
  # together with the other connectors
```
Then you add `F1.` to your last connection set where `F1` is the designator of the ferrule definition in `connectors` and the trailing dot indicates you want separate anonymous instances created for each wire using `F1` as a template, like this:
```yaml
connections:
  -  # 1st connection set
  -  # 2nd connection set
  -
    - X1: [1-2]
    - W3: [1-2]
    - F1.
```
In rare cases you might want to specify designators to each ferrule instance, and then you can replace the simple `F1.` syntax with `[F1.F11, F1.F12]` to use `F1` as a template for both `F11` and `F12`. To show these latter designators in the diagram, you also must add `show_name: true` (default false for `simple` connectors) to the `F1` definition.

kvid, 2024-07-28:
@0x0fe wrote:
> So this is the best i can do, i wish we could make X3 view less anbiguous, for example if it could be a little higher the wiers bend would be more clear.

I can see the wire splines between W1 and X3 are a bit hard to separate. This layout is made by the Graphwiz back-end that probably optimize on shortest total spline lengths, and places X3 low because there are more splines between W2 and X3 than between W1 and X3. There are a couple of things you might try:
- Increase the global horizontal spacing to make more space for the splines.
- Add an invisible edge/spline between W3 and X3 to encourage Graphviz to place X3 higher.

This is a combination of the two techniques, and you can try different `ranksep` and `weight` values to suit your needs:
```yaml
tweak:
  override:
    graph:
      ranksep: '3'  # Global horizontal spacing (default 2)
  append: |
    "W3" -- "X3" [style=invis weight=2]  // Invisible edge to bring X3 closer to W3
```