wireviz/WireViz #418: Alignment of the connector images

martonmiklos, 2024-08-17:
I have some connectors like this:
[image]

However when positioned below the connector pin table they waste  a lot of vertical space.

If they could be aligned next to the pin table the diagrams generated would be more compact.

martonmiklos, 2024-08-25:
I managed to patch the code, the results are quite satisfying for me:
[image]

I have added an alignment attribute (with possible values 'under', 'above', 'left', 'right') to the images for connectors only. 

Questions:
As the image attributes are documented for the cables and connectors together, would it make sense to have it implemented for cables too?

Code lives here in the case if someone want to peek into it: 
https://github.com/martonmiklos/WireViz/tree/add_connector_image_alignment

kvid, 2024-09-07:
@martonmiklos - Thank you very much for this feature suggestion, and I encourage you to create a draft PR with your code suggestions. I have tried your code and and have some comments. If you create a draft PR, then I can place such comments directly at the relevant code changes. My preliminary comments:
- The optional `left`/`right` seems very useful in some cases to improve the diagram readability.
- Maybe `below` could be an alternative to `under`?
- The `above` option is currently placing the image above the connector title. IMHO, just above the connector pins is better, but please argue against me, or suggest different alternatives.
- A property named `align` or `alignment` could also be misinterpreted as alignment within the cell when the image is smaller than the available cell space. Is a property named e.g. `position` perhaps better?
- For connectors with `style: simple`, it seems any `left` or `right` image is not shown.
- Ideally, I would like to have the similar options for cables, but it can wait until the code changes for connectors has settled and also check for possible conflicts with other PRs to be merged in soon.

martonmiklos, 2024-09-07:
Hi @kvid 

Many thanks for your feedback, I almost thought that the project development stalled totally ;)

Draft PR created see: #420 

> Maybe below could be an alternative to under?

I am not a native English speaker and my English is not particularly good, so I changed the under to below if it makes more sense to you.

> The above option is currently placing the image above the connector title. IMHO, just above the connector pins is better, but please argue against me, or suggest different alternatives.

Yes I agree, fixed it.

> A property named align or alignment could also be misinterpreted as alignment within the cell when the image is smaller than the available cell space. Is a property named e.g. position perhaps better?

Position is better I think too, changed it. 

> For connectors with style: simple, it seems any left or right image is not shown.

Uhm yes, I will fix this. 

> Ideally, I would like to have the similar options for cables, but it can wait until the code changes for connectors has settled and also checked for possible conflicts with other PRs to be merged in soon

Agreed!