wireviz/WireViz #422: Define a conductor in a wire bundle in the BOM

DaControl, 2024-09-18:
Hi,

I don't hope my google skills were just not sufficient, but I have a pretty basic question.

When I have defined a cable bundle with multiple wires I would like to give each wire his own P/N, since it contains multiple different wires. It is not the whole bundle who has a part number, the cable in it has a unique one. It is not as important on the graphic, the most import would be in the BOM.

The cables get already defined as a single wire, I just don't get the P/N in there.

Is that possible?

kvid, 2024-09-18:
See bundle `W2` in [tutorial 08](https://github.com/wireviz/WireViz/tree/master/tutorial#08---part-numbers-and-additional-components). Unfortunately, it seems this option is not yet described in the syntax documentation.

DaControl, 2024-09-18:
That was, what I was looking for. Great, thank you so much!

closed 2024-09-18 by DaControl