wireviz/WireViz #424: template placeholders example needed

ask6483, 2024-10-03:
I don't quite understand how to use custom placeholders in template, is there any examples how to use this?

[image]

What I try to achieve is accessing drawing.number and drawing.file:
```
metadata:
  title: 'bla bla'
  description: 'bla bla'
  pn: '1005004'
  date: '08.07.2024'
  template:
    name: '../templates/simple'
  drawing:
    number: 'ASE 100 E 254 - C'
    file: '../drawings/file.pdf'
```

Tried ``, ``, ``?

kvid, 2024-10-03:
@ask6483 - Thank you for describing your use case. Unfortunately, what you ask for is not (yet?) supported.

Custom metadata supported in v0.4.1:
- Simple {item} str/int/float value as ``
- Dict {item} of dicts {category} of {key} str/int/float values as `` and `` where `{i}` is 1 for the first {category}, 2 for the second, etc.

The metadata below demonstrates the latter, and can e.g. be referenced as `<p> drawing: <a href=""></a></p>` in the template.
```
metadata:
  drawings:
    Initial:
      number: 'ASE 100 E 254 - C'
      file: '../drawings/file.pdf'
```

The TODO at https://github.com/wireviz/WireViz/blob/v0.4.1/src/wireviz/wv_html.py#L118 might be a solution to what you expect to support ``, but it's not thoroughly tested.

ask6483, 2024-10-04:
Thank you for clarifying this.

closed 2024-10-04 by ask6483