wireviz/WireViz #425: [feature] Add support for fetching images from the Web

kacper-ka, 2024-10-04:
## Problem description
We are using our own private instance of *Gitlab* where we hold wiring documentation, rendered by `wireviz`. In each wiring diagram we use `image` directive, whose field `src` points to a file inside the repository. All is good as long as the diagrams are generated locally, when all the image files are accessible. Recently we have instantiated our own self-hosted `kroki` service, which was intended to be used as the diagram generation endpoint for documentation rendered by *Gitlab* (instead of embedding generated diagrams in the text documents we were intending to embed just the `wireviz` source code and have *Gitlab* take care of the rendering using appropriate `kroki` endpoint). We have stumbled upon an issue that the `wireviz` component of the `kroki` service does not have access to the static files inside the repository (which is pretty obvious). We have thought about possible solutions and the most straightforward one which is to have the repository cloned and mounted in the server the `kroki` service runs doesn't seem very elegant, apart from the questions how often and what branches to synchronize.

## Proposed solution
It was the first thing we tried but unfortunately that didn't work: specify a URI location of the image source. If URI started with a `file://` protocol (the default) then it would mean the image should be accessed on a local file system. If the URI started with a `http://` or `https://` protocol then the file would be accessed via the Internet. That would resolve our use case as we simply could place all assets in a common location accessibly from all the machines and the diagram generation process would succeed.

kvid, 2024-10-04:
The Graphviz utility we use doesn't support reading URIs. It only support reading the local file system. See https://gitlab.com/graphviz/graphviz/-/issues/1664

kacper-ka, 2024-10-08:
Moreover, a preprocessor for that purpose has been added to the Graphviz repository over a year ago: https://gitlab.com/graphviz/graphviz/-/commit/af8a77c49b2aef720344f0a1b90a429ca1da5931