wireviz/WireViz #426: AttributeError: 'NoneType' object has no attribute 'get'

Monniasza, 2024-10-12:
I'm unable to process my WireViz file. It's correctly formatted, but it fails with a strange error. To reproduce, save the harness.yaml file in the same directory that command will be run from, and then run `wireviz harness.yaml`
```
oskar@oskar-Inspiron-5577:~/SynologyDrive/3D/filamaker$ wireviz harness.yaml

WireViz 0.4.1
Input file:   harness.yaml
Output file:  harness.[html|png|svg|tsv]
Traceback (most recent call last):
  File "/home/oskar/.local/bin/wireviz", line 8, in <module>
    sys.exit(wireviz())
             ^^^^^^^^^
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/wireviz/wv_cli.py", line 141, in wireviz
    wv.parse(
  File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/wireviz/wireviz.py", line 140, in parse
    image = attribs.get("image")
            ^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'get'

```

And here's the harness.yaml file:
```
templates:
   - &CableMains
     category: bundle
     gauge: 0.5 mm2
     length: 1
     colors: [BN, BU, GN]
     show_equiv: true
   - &StepperCon
     type: Molex KK 254
     subtype: female
     pinlabels: [12V, 5V, EN, STEP, DIR, GND]
   - &I2C
     type: DuPont
     subtype: female
     pinlabels: [GND, VCC, SDA, SCL]
   - &CableI2C
     category: bundle
     gauge: 0.08 mm2
     length: 0.5
     colors: [RD, BK, GN, WH]
     show_equiv: true
   - &Encoder
     type: DuPont
     subtype: female
     pinlabels: [CLK, DT, SW, VCC, GND]
   - &Mains
     type: Ferrules
     subtype: male
     pinlabels: [L, N, PE]
   
  
connectors:
  BoardMains:
    type: Molex KK 254
    subtype: female
    pinlabels: [L, N, PE]
  BoardMains2:
    
  BoardPower:
    type: Molex KK 254
    subtype: female
    pinlabels: [12V, 5V, 3V3, GND]
  BoardThermistor:
    type: JST XH
    subtype: male
    pinlabels: [TH1, TH2]
  BoardStepper:
    <<: *StepperCon
  BoardLCD:
    <<: *I2C
  BoardCaliper:
    <<: *I2C
  BoardEncoder:
    <<: *Encoder
  MotorPower:
    type: Spade 6.3
    subtype: female
    pinlabels: [12V, GND]
  ToPSU:
    type: Soldered
    pinlabels: [L, N, PE, 12V, 5V, 3V3, GND]
  Heater:
    type: Spade
    pinlabels: [L, N, PE, TH1, TH2]
  Stepper:
    <<: *StepperCon
  LCD: 
    <<: *I2C
  Caliper:
    <<: *I2C
  Encoder:
    <<: *Encoder
     
cables:
  CableMotor:
    category: bundle
    gauge: 2.5 mm2
    length: 1
    colors: [RD, BK]
    show_equiv: true
  CableBoard:
    category: bundle
    gauge: 0.25 mm2
    length: 1
    colors: [YE, RD, OG, BK]
    show_equiv: true
  CableMainsIn:
    <<: *CableMains
  CableMainsHeater:
    <<: *CableMains
    length: 0.6
    colors: [BN, BU, GN, WH, WH]
  CableStepper:
    category: bundle
    gauge: 0.25 mm2
    length: 0.6
    colors: [YE, RD, GN, BU, WH, BK]
    show_equiv: true
  CableLCD:
    <<: *CableI2C
  CableCaliper:
    <<: *CableI2C
  CableEncoder:
    category: bundle
    gauge: 0.08 mm2
    length: 0.15
    colors: [WH, GN, BU, RD, BK]
    show_equiv: true
connections:
  -
    - ToPSU: [1-3]
    - CableMainsIn: [1-3]
    - BoardMains: [1-3]
  -
    - BoardMains2: [1-3]
    - CableMainsHeater: [1-3]
    - Heater: [1-3]
  -
    - BoardThermistor: [1, 2]
    - CableMainsHeater: [4, 5]
    - Heater: [4, 5]
  -
    - ToPSU: [4, 7]
    - CableMotor: [1, 2]
    - MotorPower: [1, 2]
  - 
    - ToPSU: [4-7]
    - CableBoard: [1-4]
    - BoardPower: [1-4]
  -
    - BoardStepper: [1-6]
    - CableStepper: [1-6]
    - Stepper: [1-4, 6, 5]
  - 
    - BoardLCD: [1-4]
    - CableLCD: [1-4]
    - LCD: [1-4]
  - 
    - BoardCaliper: [1-4]
    - CableCaliper: [1-4]
    - Caliper: [1-4]
  - 
    - BoardEncoder: [1-5]
    - CableEncoder: [1-5]
    - Encoder: [1-5]
  
    
```

kvid, 2024-10-14:
@Monniasza wrote:
> I'm unable to process my WireViz file. It's correctly formatted,

I'm afraid, your YAML input is not completely correct.

> but it fails with a strange error. To reproduce, save the harness.yaml file in the same directory that command will be run from, and then run `wireviz harness.yaml`
> 
> ```
> oskar@oskar-Inspiron-5577:~/SynologyDrive/3D/filamaker$ wireviz harness.yaml
> 
> WireViz 0.4.1
> Input file:   harness.yaml
> Output file:  harness.[html|png|svg|tsv]
> Traceback (most recent call last):
>   [...]
>   File "/home/oskar/.local/share/pipx/venvs/wireviz/lib/python3.12/site-packages/wireviz/wireviz.py", line 140, in parse
>     image = attribs.get("image")
>             ^^^^^^^^^^^
> AttributeError: 'NoneType' object has no attribute 'get'
> ```

This exception tells me that one of your entries where an `image` attribute is allowed (connector or cable entry) is `None` instead of the expected `dict` type.

I admit that we should improve the explanation of such unexpected YAML input, and we discuss such improvements in #207.

> 
> And here's the harness.yaml file:
> 
> ```yaml
> templates:
>    - [...]
> connectors:
>   BoardMains:
>     type: Molex KK 254
>     subtype: female
>     pinlabels: [L, N, PE]
>   BoardMains2:
>     
>   BoardPower:
>     [...]
> ```

Your `BoardMains2` entry has no attributes and is therefore interpreted as `None`. If I add a `pincount: 3` attribute to this entry, the YAML input is processed without any error.

> ```yaml
> connections:
>   -
>     - ToPSU: [1-3]
>     - CableMainsIn: [1-3]
>     - BoardMains: [1-3]
>   -
>     - BoardMains2: [1-3]
> ```

The `BoardMains2` needs at least 3 pins for this connection to be legal.

Monniasza, 2024-10-14:
@kvid The message should say that entry is empty, like "SomeConnector is empty"