wireviz/WireViz #427: [query] Is it possible to order the pins in descending order?

AdriZ, 2024-10-23:
I have a cable that is straightforward but the pin numbering is in opposite way on each connector.

Is it possible to order pins in descending order or another solution to arrange that?
[image]

Here is my yml file renamed with .txt to be accepted by Github.
[example.yml.txt](https://github.com/user-attachments/files/17495780/example.yml.txt)

kvid, 2024-10-24:
It's possible to specify custom pin names in any order you want:
```yaml
  pins: [12,11,10,9,8,7,6,5,4,3,2,1]
  pinlabels: [...]  # NB: The pin labels are expected to be in the same order.
```
See also #279 for a suggested future alternative that could be useful in your use case.

closed 2024-10-24 by AdriZ

kvid, 2024-12-11:
> What does the comment "The pin labels are expected to be in the same order." mean? Assuming P1 is Pin #1, Pin 12 is Pin #2, etc., which of these is correct?
> 
> ```yaml
> # Option 1
> pins: [12,11,10,9,8,7,6,5,4,3,2,1]
> pinlabels: [P12, P11, P10, P9, P8, P7, P6, P5, P4, P3, P2, P1]  # NB: The pin labels are expected to be in the same order. (??)
> 
> # Option 2
> pins: [12,11,10,9,8,7,6,5,4,3,2,1]
> pinlabels: [P1, P2, P3, P4, P5, P6, P7, P8, P9, P10, P11, P12]  # NB: The pin labels are expected to be in the same order. (??)
> ```

- In option 1: pin 12 is labelled P12, pin 11 is labelled P11, pin 10 is labelled P10, etc.
- In option 2: pin 12 is labelled P1, pin 11 is labelled P2, pin 10 is labelled P3, etc.