wireviz/WireViz #428: [doc] Clarification of ability to use titles and other notes

zedfactorial, 2024-10-27:
For the past few hours I have been trying to understand if adding a title to the output image is possible.

I have gone through all documentation and the issues and found lots of discussion related to this, but it was in context of implementing .pdf output which really quickly goes in technical levels above my ability.

I dont need fancy PDF output, I'm just looking for a way to put a title (or any other text) anywhere on the output image.  The examples in #32 look perfect, something like that would be great but Id be happy with a lot less.

I have gone through all the examples and the tutorials and found nothing where a title would be present. So Im assuming it's not there, at the same time some of the examples in the .pdf discussions include a lot more than what Im asking for.

So in short - can someone who is only capable of editing the .yml files get text on a image or no?

As a workaround I used a unconnected connector, which works pretty well. But annoyingly I found that I need to set it to have at least 1 pin which clutters up the title and makes it confusing to people I send the image to.

In any case an update of the documentation to clarify this for monkeys like me would be very nice.

Your help and patience is greatly appreciated.

Attached is the example of the workaround Im using now.
[image]

kvid, 2024-10-27:
All our examples create a title (from metadata or filename) and a BOM in the HTML output that also includes the diagram. [Demo02](https://github.com/wireviz/WireViz/blob/master/examples/demo02.yml) demonstrates how this can be customized. See https://github.com/wireviz/WireViz/issues/227#issuecomment-805399108 for a suggested work-around to test including such things directly in the diagram. A similar issue is also discussed in #367.

zedfactorial, 2024-10-27:
> All our examples create a title (from metadata or filename) and a BOM in the HTML output that also includes the diagram. [Demo02](https://github.com/wireviz/WireViz/blob/master/examples/demo02.yml) demonstrates how this can be customized. See [#227 (comment)](https://github.com/wireviz/WireViz/issues/227#issuecomment-805399108) for a suggested work-around to test including such things directly in the diagram. A similar issue is also discussed in #367.

Hi, kvid. Thanks for the quick reply.

Thanks for the clarification about the HTML file. I was not aware of that, while it doesn't solve my problem its certainly useful to know.

The comment from 227 seems to be exactly what I was looking for, I tried it and it works, brilliant!

Before this I was not even aware that tweak existed. Maybe that should be in one of the examples? Seems like a great feature to be aware of.

Apologies in advance if it is and I missed it - I ctrl-F -ed the tutorial page and found no mention of it.

For me the situation is solved, thanks again!

kvid, 2024-10-27:
The tweak feature is still an experimental post-processing of the .gv output (input to Graphviz) and is mainly designed for testing stuff not yet implemented or bug work-arounds. It requires detailed knowledge of the Graphviz syntax. The goal is to include useful functionality as ordinary features that doesn't require Graphviz knowledge. However, it has been useful in many issues, and might deserve an example some day, but describing the Graphviz syntax is clearly out of scope within this project.

In addition to examples and tutorials, I recommend reading the [syntax doc](https://github.com/wireviz/WireViz/blob/master/docs/syntax.mdl).

bradstaff, 2025-02-12:
I took the HTML and used wkhtmltoimage to convert it to a png. I made a script that runs wireviz and then converts the HTML to a PNG so I get a PNG with a title at the top. Is there a way for metadata beyond the title and notes to print without using the HTML templates?

kvid, 2025-02-15:
> I took the HTML and used wkhtmltoimage to convert it to a png. I made a script that runs wireviz and then converts the HTML to a PNG so I get a PNG with a title at the top.

Good idea! If you want to, feel free to share your script here to help others with similar needs.

> Is there a way for metadata beyond the title and notes to print without using the HTML templates?

The current WireViz code that creates the HTML output always use a template for this. Unless specified otherwise, this default template is used:  [simple.html](https://github.com/wireviz/WireViz/blob/master/src/wireviz/templates/simple.html)

This default simple template uses these metadata:
- title
- description
- notes

and these options:
- fontname
- bgcolor

I you open the template file, you'll find their placeholders between a leading ``. If you need other metadata values in the HTML output, then I suggest you create and use a custom template file. Just copy one of the standard templates and add placeholders according to:  [the template documentation](https://github.com/wireviz/WireViz/blob/master/src/wireviz/templates/README.md)