wireviz/WireViz #431: Distinction between bundle and wire?

lutorm, 2024-11-24:
Hi,
I find myself in need of designing a pretty complicated harness (for a small aircraft) so I'm looking around for options and came across Wireviz. I couldn't figure out whether there's a distinction between a wire and a bundle in Wireviz:

In my mind, a harness typically has an acyclical graph structure, with copper wires going between the various nodes (generally leaf nodes, but not necessarily in the case of in-line splices) in the graph. A "bundle? to me denotes an edge in this graph. So, a wire from X1 to X2 will traverse several different bundles of different lengths, bundled together with different sets of wires in any of them, on its way between the end points.

To me, the real utility of defining the harness would be to be able to input the harness physical layout, ie the bundle graph with edge lengths, separately define the electrical connections, i.e. the wires, connecting the nodes, along with their color/gauge, and then have the software calculate where the wire goes and how long the wire needs to be. It looks like this is not the problem Wireviz is attempting to solve?

kvid, 2024-11-24:
> Hi, I find myself in need of designing a pretty complicated harness (for a small aircraft) so I'm looking around for options and came across Wireviz. I couldn't figure out whether there's a [...]

- Your definition of a bundle seems to be equal to how this term us used in WireViz.
- A cable or bundle in WireViz currently consist of one or more wires with a common length.
- Variations of extra wire lengths at both ends is suggested in https://github.com/wireviz/WireViz/issues/268#issuecomment-1024807332
- How to make a wire continue through several bundles is suggested in https://github.com/wireviz/WireViz/issues/270#issuecomment-1029535877

> To me, the real utility of defining the harness would be to be able to input the harness physical layout, ie the bundle graph with edge lengths, **separately** define the electrical connections, [...]

In WireViz the electrical connections are specified for each connection set - which is one or more connections sharing the same sequence of alternating connectors and cables/bundles, i.e. a botton-up approach connecting each segment instead of your top-down request.

However, WireViz calculates the length sum of wires with equal properties for the BOM.

lutorm, 2024-11-24:
Yeah, I'm pondering how I would describe the necessary input data now, so I'll post back if I come up with anything.

Did I understand correctly that it's now possible to put different gauge wires in a bundle? I haven't seen an example of the syntax to do so, though.

kvid, 2024-11-24:
> [...]
> Did I understand correctly that it's now possible to put different gauge wires in a bundle? I haven't seen an example of the syntax to do so, though.

No, the current version doesn't support this yet, but this feature is suggested in #268. See my comment in that issue linked in my comment earlier today for how such a feature might look like.

lutorm, 2024-11-26:
Ok, I have started working on a routing processor per what I outlined above. In order for the result to be renderable in Wireviz, though, it would be a requirement to support:
 - Different gauge wires in a bundle
 - Several shielded wires in a bundle (along with other conductors)
 
If I get this to work, the resulting harness will likely have of order 30 nodes and 500 conductors, so I suspect that the result will be too complicated for graphviz to render in comprehensible fashion. I saw a discussion of an "outline mode" where, rather than rendering all the wires, it just renders a fat line for the bundle along with a table of the wires in the bundle. This would be fully sufficient for my purpose, rendering all the wires will just make a mess, I think.

lutorm, 2024-11-28:
So the generation code appears to be working on a small test file I made up. However, the result is not pleasant. Even for such a small case, the result is pretty much unintelligible. It's possible the generated file is incorrect, but I figured I'd throw it out here and see if anyone has suggestions for improvements in the generated file to make it better.
[wireviz.txt](https://github.com/user-attachments/files/17944946/wireviz.txt)
[harness.txt](https://github.com/user-attachments/files/17944949/harness.txt)
[image]

Harness.txt is the source file for the generation, in case anyone's interested. The basic topology of the harness is given by the "edges" dict, where the edge is named for the two nodes it connects. The nodes named J* are all harness branch points and not connectors (although the generated wireviz output generates them as a simple connector.)

The connections are in the netlist dict, which specifies the connectors the net connect to, the lable of the cable connecting it, and the type of conductor. The conductors can be hierarchical, shielded, multiconductors, but since wireviz can't handle that anyway, I just used the first conductor in the multiconductor cable and dropped the others in the generated output. Since they all are routed together and just connect to different pins, they're not super interesting anyway.

When the harness passes a physical connector, the conductors in the harness that don't connect to any pins on that connector are connected to instances of the autogenerated simple "X" connector. I was hoping that would keep them just flowing along with the rest of the conductors in the wire.

One particular problem evident in the generated image is that some lines take the wrong direction through the wire boxes. I have assumed that a connection entry is nondirected, ie a connection P1->W1->P2 is equivalent to P2->W1->P1, since obviously electrically there's no difference. Does the wireviz rendering somehow rely on the direction of the entries?

Also, when a "connector" is just a branch point in the harness, like J1 and J2, I've generated those as real connectors which the wires connect to. Is this the best way to do it? I could also generate individual splices for every conductor, but I thought using a single "connector" would keep everything together. Maybe it just makes it worse since it apparently can't decide which end to make the wires come out?

kvid, 2024-11-29:
Thank you for sharing your work. I've not yet looked into the details, but I must answer your question "Does the wireviz rendering somehow rely on the direction of the entries?"

Yes, all connection sets are assumed by WireViz to be from left to right to enable the YAML author to influence the diagram layout. See also my explanation in https://github.com/wireviz/WireViz/issues/293#issuecomment-1249843359

lutorm, 2024-11-29:
Yeah, I saw that discussion after I had posted my question. Sorting the connections consistently improved the look a lot. I also reworked the whole-harness diagram to generate a separate wireviz cable for every cable in a bundle, since that's the only way I could figure out to get the BOM to generate correctly. Currently it looks like this:
[image]

I think the full diagram will never be useful, but the BOM will be. An "overview" mode that doesn't render the individual wires would still be very nice, though.

Instead I focused on getting views of individual connectors to generate. These look like this, which is pretty nice and certainly will be good enough to manufacture the individual connector endpoints.
[image]

kvid, 2024-11-29:
- If you consider splitting your project into several harnesses, but still need an overall BOM, then I recommend #267.
- See #330 for several shielded wire groups.
- See #268 for different gauges and lengths, and this concept might also work for other attributed

KevinPaul1995, 2026-08-18:
I’m interested in working on this feature from the physical harness/topology side.

What I have in mind is not just changing GraphViz ordering, but adding a higher-level harness model that describes the physical trunk, branch points, branch lengths, and connector locations separately from the electrical netlist.

For example, something along these lines:

harness:
nodes:
ENTRY:
type: endpoint
J1:
type: branch
J2:
type: branch
EV1:
type: connector
EV2:
type: connector

segments:
MAIN_1:
from: ENTRY
to: J1
length: 300 mm
MAIN_2:
from: J1
to: J2
length: 400 mm
BRANCH_EV1:
from: J1
to: EV1
length: 150 mm
BRANCH_EV2:
from: J2
to: EV2
length: 180 mm

The idea would be that the physical harness topology describes the trunk, branch nodes, and segment lengths, while the electrical connectivity remains separate using the existing WireViz connection/netlist data. A conductor could traverse multiple physical harness segments.

A 2D bundle/outline renderer could then show the actual harness structure instead of drawing every individual conductor. Later, the same topology model could potentially feed a 3D renderer without changing the electrical model.

I would prefer to start with the 2D physical topology first and treat 3D as a later renderer rather than mixing 3D concerns into the data model.

Would this kind of model fit the intended direction of WireViz? If so, I’d be interested in implementing an initial minimal version and would appreciate guidance on where you would prefer this to live, for example in core WireViz or as a preprocessor, and what naming/syntax would be preferred.