wireviz/WireViz #432: Loops specification can't use pin labels?

lutorm, 2024-11-29:
I have a wire that should connect two pins on the same connector, and I discovered the "loops" feature. However, when I attempted to use it, I got this error:

> exception: Unknown loop pin "6" for connector "P4601"!

From some experimentation, it appears that the loops lists only understands pin numbers, not pin labels? Is this intentional? It seems to go against the pattern used everywhere else where you can use numbers and labels interchangeably.

(This feature also seems a bit half-baked, honestly. I would need to specify the color and gauge of the wire used, so I'd need to make up a cable section for it anyway so these can be specified.)

kvid, 2025-01-06:
I agree the current `loops` feature has limitations. Possible improvements are discussed in several issues, e.g. #350 and #439.

kvid, 2025-07-27:
It seems the [pin mapping code](https://github.com/wireviz/WireViz/blob/v0.4.1/src/wireviz/Harness.py#L112-L129) in `Harness.connect()` needs to be moved into a new `Connector.pin_index()` method to be available from both `Harness.connect()` and from `Connector.__post_init__()` (the latter is where loop pins are processed).